corneel27/day-ahead #19: Error connecting to the database

EJhagenaar, 2023-06-21:
[image]

ik heb bovenstaande error. in de options.json & secrets.json de in mijn ogen goede gegevens in gevuld. (welke database is het?)

corneel27, 2023-06-21:
Hij zou de "day ahead" database moeten openen.
Bij mij heeft die de volgende instellingen:
```
  "database da": {
    "server": "192.168.178.36",
    "database": "day_ahead",
    "port": 3306,
    "username": "day_ahead",
    "password": "!secret db_da_password"
  },
```
Kun jij met jouw instellingen wel contact maken met die database met een ander programma zoals phpMyAdmin of HeidiSQL?

corneel27, 2023-06-21:
Standaard is poort 3306 (en niet 3308) de poort waarover mysql/mariadb communiceert.
Of heb je dat aangepast?

corneel27, 2023-06-21:
De huidige versie probeert beide databases (day_ahead en homeassistant) te openen.
Als ik naar jouw errormelding kijkt zie  ik dat hij wel contact krijgt met de day_ahead database maar dat het fout gaat met de ha-database. Ik zal die foutmelding duidelijker en explicieter maken.
Dat verandert allemaal in de versie die er binnenkort aankomt.
Dan wordt alleen bij een bepaalde actie de daarvoor benodigde database(s) geopend.
Je krijgt wel een melding dat data aanwezig zijn tot 20 juni 00:00 .
Kun je met HeideSQL of phpmyadmin checken of er data in de values tabel staan?

corneel27, 2023-06-21:
Ik heb nog even in de code gekeken.
ik zie nu dat die niet goed omgaat met een andere port.
Ik zal vanavond de code op github updaten.

corneel27, 2023-06-21:
Ik heb een nieuwe versie geupload op github.
Omdat ik met port 3306 werk heb ik niet kunnen testen of het ook werkt met port 3308 (of een willekeurige andere poort).
Zou jij willen testen of deze versie werkt met 3308?

EJhagenaar, 2023-06-21:
[image]

corneel27, 2023-06-21:
Dank je voor het testen.

corneel27, 2023-06-22:
Sorry de documentatie loopt iets achter op de ontwikkeling van de software.
Om fouten tijdens het ophalen van tibber data te voorkomen moeten nog deze twee regels worden toegevoegd aan de tabel "variabel":
```
INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (7, 'cost', 'cost', 'euro');
INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (8, 'profit', 'profit', 'euro');

```

ruudbikeright, 2023-07-22:
> ```
> INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (7, 'cost', 'cost', 'euro');
> INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (8, 'profit', 'profit', 'euro');
> ```

@corneel27 Deze twee regels heb je nog niet in de README bijgewerkt.

tonvanboven, 2023-07-22:
> INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (7, 'cost', 'cost', 'euro');
> INSERT INTO `variabel` (`id`, `code`, `name`, `dim`) VALUES (8, 'profit', 'profit', 'euro');

@corneel27 Deze twee regels staan nog niet in de README.md

tonvanboven, 2023-07-31:
De regels zijn doorgezet in Develop.

closed 2023-07-31 by tonvanboven