corneel27/day-ahead #21: Foutmelding day_adead.py

tonvanboven, 2023-07-31:
@corneel27:

Resultaat van `python3 day_ahead.py calc`

All taxes refund (alles wordt gesaldeerd)
Traceback (most recent call last):
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 1366, in <module>
    main()
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 1347, in main
    day_ah.calc_optimum(show_graph=True)
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 379, in calc_optimum
    max_charge_power.append(self.battery_options[b]["charge stages"][-1]["power"]/1000)
KeyError: 'charge stages'

tonvanboven, 2023-07-31:
Ah, lezen zei mijn vader vroeger ;-) Opgelost. Maar dan komt de volgende:

Doorzetten van alle settings naar HA
Traceback (most recent call last):
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 1366, in <module>
    main()
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 1347, in main
    day_ah.calc_optimum(show_graph=True)
  File "/home/ton/develop/day-ahead/prog/day_ahead.py", line 997, in calc_optimum
    entity_pv_switch = self.config.get(["entity pv switch"], self.solar[s])
  File "/home/ton/develop/day-ahead/prog/da_config.py", line 17, in get
    result = options[keys[0]]
KeyError: 'entity pv switch'
ton@mip:~/develop/day-ahead/prog$ 

En dit maal wél gezocht maar niet gevonden in het `options_vb.json` bestand.

tonvanboven, 2023-07-31:
@corneel27 README moet nog aangepast worden met uitleg / instructie:
- entity pv switch
- entity balance switch

Zoals ik het begrijp kun je per solar een on/off switch hebben om bij negatieve stroomprijzen de PV uit te schakelen. Heb je dat niet dan kun je volstaan met slechts één helper in HA.

Wat de balance switch doet zou ik graag willen weten.

closed 2023-07-31 by corneel27

corneel27, 2023-07-31:
entity balance switch is beter uitgelegd in README.md

commit 874db77b (2023-07-31): Foutmelding day_adead.py
Fixes #21