corneel27/day-ahead #29: structuur database homeassistant ontbreekt

hvdort, 2023-09-29:
Goedenavond,

Een heel interessant project! Ik probeer e.e.a. aan de praat te krijgen op een NUC (dietpi distributie).

Bij het opstarten van de webserver loop ik tegen dit probleem aan:

mysql.connector.errors.ProgrammingError: 1146 (42S02): Table 'homeassistant.statistics' doesn't exist

Waar kan ik de structuur van de database 'homeassistant' vinden?

corneel27, 2023-09-29:
Beste Hugo,
Om jouw situatie te begrijpen en te voorkomen dat ik het verkeerd interpreteer eerst een paar vragen:
Heb je de addon MariaDB geinstalleerd en staat daar ook de database van Home Assistant op?
Heb je de addon phpMyAdmin geinstalleerd?
Als die beide er zijn dan zou je via phpMyAdmin de structuur van de HA database moeten kunnen zien:
[image]
Lukt dat?
Ik hoor het graag.

hvdort, 2023-09-30:
Beste Cees,

Dank voor de snelle reactie!

Het zal voornamelijk liggen aan mijn onbekendheid met HA. Dat gebruikt kennelijk in mijn geval (nog) geen MySQL/MariaDB maar een andere database.
Ik heb eerst een poging gedaan met de OS installer van HA zelf (HassOS), maar dan blijkt dat het filesystem zo goed beveiligd is dat elke wijziging die ik wil doen (incl. jullie python code) wordt verwijderd.  Daarnaast lukte het niet om inotifywait te gebruiken.

Daarom ben ik opnieuw begonnen, nu met DietPI als basis (lichtgewicht). Die heeft standaard ondersteuning voor de installatie van HA, maar ook daarbij wordt kennelijk standaard geen MySQL/MariaDB gebruikt. (Ook hier moest ik trouwens nog een package extra installeren om inotifywait (gebruikt in de watchdog) te kunnen gebruiken).

Om aan jullie requirements te kunnen voldoen heb ik (in dit geval buiten HA om) MariaDB en phpmyadmin geinstalleerd, maar dan is de database "homeassistant" natuurlijk niet vanzelf aanwezig. DietPI is in dit geval wel heel handig. Ik red me handmatig ook wel, maar het is wel praktisch om in de package manager gewoon een vinkje te kunnen zetten bij MariaDB, phpmyadmin én HA.

Ik begrijp dat ik dus moet kijken naar HA hoe ik die aan de MariaDB kan koppelen?

corneel27, 2023-09-30:
Beste Hugo,

Dat gaat heel eenvoudig.
Als je in HA zit dan heb je links in de verticale menubalk, de optie "Instellingen"
Als je die selecteert, dan krijg je een groot aantal keuzemogelijkheden o.a. "Addons".
Als je die selecteert dan zoek je MariaDB en installeer je die.
Je kunt dan ook gelijk je hudige db van ha (sqllite) overzetten naar MariaDB. 
Ik zou daarna dan ook gelijk de addon "phpMyAdmin" installeren zodat je toegang krijgt tot mariadb.

Succes!

hvdort, 2023-09-30:
Beste Cees, zo simpel is het helaas niet. de Addons optie heb ik niet omdat ik een installatievariant van HA heb gekozen (Core) die dat niet ondersteunt.

Ik heb een andere weg bewandeld: https://community.home-assistant.io/t/migrating-home-assistant-database-from-sqlite-to-mariadb-only-if-you-are-very-familiar-with-database-administration/96895/81

Tot zover lijkt alles goedgegaan. De webserver wil nu inderdaad starten:

root@DietPi:/usr/share/day-ahead/webserver# python3 da_server.py
MySQL database connection successful. Default database: day_ahead
MySQL database connection successful. Default database: homeassistant
 * Serving Flask app 'webserver.app'
 * Debug mode: off

Maar als ik dan http://homeassistant:5000 aanroep krijg ik een error 500 en de volgende foutmelding:

[2023-09-30 12:47:24,360] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/share/day-ahead/webserver/../webserver/app/routes.py", line 114, in optimalisering
    active_time = str(flist[index]["time"])
                      ~~~~~^^^^^^^
IndexError: list index out of range

De mogelijkheid dat er helemaal nog geen data is wordt kennelijk niet afgevangen. Ik mis dus nog wat data. Dat zou kunnen aangezien de koppeling met entsoe nog niet actief is vanwege afwachting van goedkeuring. Klopt mijn gedachtengang?

corneel27, 2023-09-30:
Beste Hugo,
Je bent al een flink stuk op weg, goed bezig!!
Het ontbreken van data wordt inderdaad niet afgevangen, omdat iedereen - tot nu toe - begint met de rekenmodule aan de praat te krijgen.
Zodra die rekent kun je met de webserver data en plaatjes ophalen en laten zien.
Ik ga jouw bevinding wel verwerken in een van de volgende updates zodat je gewoon een melding krijgt in de trand van: "Er zijn nog geen data, svp eerst de rekenmodule (bijv "python3 \prog\day_ahead.py calc") activeren".

Als die nog niet werkt omdat je nog geen entsoe key hebt kunt de nordpool-optie gebruiken om day-ahead data op te halen. Die werkt zonder key en daarvan zijn de data vaak al vanaf 12:55 uur beschikbaar voor de volgende dag.

hvdort, 2023-09-30:
We komen steeds een stukje verder :-)

De berekening komt een eind, maar dan stopt het bij regel 1524:

Doorzetten van alle settings naar HA
Traceback (most recent call last):
  File "/usr/share/day-ahead/prog/day_ahead.py", line 1764, in <module>
    main()
  File "/usr/share/day-ahead/prog/day_ahead.py", line 1737, in main
    day_ah.calc_optimum(show_graph=True)
  File "/usr/share/day-ahead/prog/day_ahead.py", line 1524, in calc_optimum
    lns = ln1[0]
          ~~~^^^
IndexError: list index out of range

Ongetwijfeld ontbreekt er nog input data, maar ik kan er even niet achterkomen wat dat dan is. Ik heb in de config de verwijzingen naar boiler, heating, battery, solar, EV en tibber leeg gemaakt.

corneel27, 2023-10-01:
Beste Hugo,
Als je de berekening laat lopen, maar je hebt alles uit staan, dan kan het programma niet schuiven met het verbruik en dan komt er nooit een "optimale" verbruikssituatie.
Ook deze situatie heb ik nooit getest om te zien hoe het programma daar mee omgaat en de gebruiker te voorzien van adekwate feedback (nog wat te doen ;) )
Dus je zult voorlopig minimaal een van de zaken moeten invullen, zodat het programma kan schuiven met het verbruik.

hvdort, 2023-10-07:
Beste Cees,
Ik begrijp het probleem. Echter, het is even de vraag in welke vorm ik mijn verbruik dan moet aanbieden. Ik lees momenteel de P1 poort uit en stop die data in een InfluxDB database. Het is niet niet zo simpel om de data ook in HA te krijgen, nog los van de dubbeling die dat kost.

Eigenlijk ben ik (voorlopig) op zoek naar een wat eenvoudiger scenario: Het vinden van een zo efficient mogelijk profiel voor het laden en ontladen van een thuisbatterij op basis van de uurtarieven via entsoe/nordpool en daarbij het eigen verbruik voorlopig buiten beschouwing te laten in de berekening.

Veel mensen in mijn omgeving zijn best geinteresseerd in het kopen van een thuisaccu, maar dan zonder al te veel rompslomp. Denk aan de Sessy, een mooi product, goede marketing, maar wel te duur gezien de opslagcapaciteit en vooral de capaciteit van de omvormer. Het alternatief zou een Bliq kunnen zijn, maar dan zit je weer aan abonnementskosten vast.

Ik denk op dit moment dus aan een soort groeimodel, dus starten met een lowcost dhz oplossing met hetzelfde gemak als een Sessy, dus uitpakken, stekker in het stopcontact, wifi/ethernet koppelen en klaar. En daarna de (groei)mogelijkheid om stap voor stap de regeling "slimmer" te maken.

Denk je dat jullie code bruikbaar kan zijn in dit (beperkte) scenario?

corneel27, 2023-10-08:
Beste Hugo,
Je kunt het met ons script zo makkelijk/moeilijk maken als je zelf wilt.
Maar een low-cost dhz oplossing mbt een thuisaccu bestaat alleen als je je verdiept in de hardware.
Die heb je niet in een kant en klare doos, maar zul je zelf moeten bouwen en aansluiten.
Dat heb ik zelf ook gedaan: https://gathering.tweakers.net/forum/list_message/73080530#73080530
Helaas is het nog steeds niet mogelijk om altijd voor een dubbeltje op de eerste rij te zitten.
Misschien worden ze goedkoper of komt er subsidie. 
Ik verdien vanaf september 2022 iedere maand met mijn accu ca 75 euro (in combinatie met een dynamisch contract).
Dat weegt straks wellicht op tegen lagere prijzen en/of subsidie.

hvdort, 2023-10-09:
Beste Cees,
Geheel mee eens! Mooi om te lezen dat de opbrengst in de praktijk ook inderdaad in de buurt komt van een schatting die mijn zoon voor me had gemaakt. Ik ga er tenminste nu vanuit dat je 10kWh aan accucapaciteit hebt staan?

Nu is er een bedrijfje bij mij in de buurt die recent is gestart. Zij willen zich vooral toeleggen op de thuisaccu en gaan die rechtstreeks importeren vanuit China. Dat betekent dat de prijs heel concurrerend gaat worden tov het huidige aanbod. Daarmee wordt de terugverdientijd uiteraard korter. Er is een proefexemplaar onderweg. Zodra die binnen is mag ik er ook "een blik" op werpen. Ik ken de details van de aansturing nog niet, maar weet intussen wel dat het in iedergeval (ook?) een Modbus koppeling is via RS485. Dat lijkt me geen onoverkomelijke moeilijkheid. Ik heb eerder gekoppeld met Modbus. Ik heb tot nu toe geen ervaring met Python en moet er nog wel aan wennen. Daarnaast is het doorgronden van de code van iemand anders niet zo eenvoudig, zeker als je de hoofdstructuur niet kent ;-)

Ik wilde dus in de tussentijd vast kijken of ik iets kan maken wat uiteindelijk leidt tot een lijstje met laad- en ontlaadmomenten voor de komende 24h. De enige variabelen die ik dan nog moet aanleveren is de capaciteit van de accu en de tijd die nodig is voor (maximaal) laden cq. ontladen, (hoewel misschien mooier is om gewoon het vermogen van de omvormer op te kunnen geven.)
Dat is dan de input voor een nog te schrijven module die specifiek hoort bij een accu. Daar zie ik zelf wel kans toe, maar ik hou er niet van om dingen na te doen die een ander al (ongetwijfeld beter!) heeft gedaan ;-)

closed 2024-02-02 by tonvanboven