corneel27/day-ahead #34: Prijs berekening incorrect. BTW ook gerekend over de Energie belasting.

hfvdv, 2023-12-13:
I tripple checked the price with tibber and it was consistently off which made me wonder if the VAT calculations were okay.  For now i am also assuming that ol_l and ol_t are without VAT like da_price.

When changing the code to the code below for now the prices appear to be correct to two decimal places ( which is what i get from Tibber )

price_l = round((row.da_price + ol_l)
                            * (1 + btw / 100) + taxes_l, 5)
            price_t = round((row.da_price  + ol_t)
                            * (1 + btw / 100) + taxes_t, 5)

corneel27, 2023-12-13:
Sorry, but I don't agree.
You only are right when the energy taxes which are filled in options.json are inclusive btw (vat). 
But the energy taxes which you  fill in in options.json are exclusive btw (vat): 
```
    "energy taxes delivery": {
      "2022-01-01": 0.06729,
      "2023-01-01" : 0.12599
    },
    "energy taxes redelivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599
    },
```
And that is also stated in the README.md:
```
- energy taxes delivery: energiebelasting op verbruik excl. btw, euro per kWh
    2022-01-01 : 0.06729,
    2023-01-01 : 0.12599
- energy taxes redelivery: energiebelasting op teruglevering excl. btw, euro per kWh
    2022-01-01: 0.06729,
    2023-01-01: 0.12599
```

hfvdv, 2023-12-14:
Fair enough.  We an always agree to disagree :-)Assuming based on the code and the ol_t from the sample options file my assumption is that tibber is the energy provider. What could be a possible explanation as to the observed price difference? My other explanation would be that the tibber pricing information in the app is incorrect. Sent from my iPhone

hfvdv, 2023-12-14:
Did some further checking and it appears to me the app ( and the information exposed in the api which HA picks up for me to use ) is incorrect. Am contacting tibber. Will share if I learn something usefull.Screenshot attached showing a 2 cent price difference for current hour between tibber app/api and tibbers price information on their website. Sent from my iPhone

corneel27, 2023-12-14:
I checked this morning the price in the app.
It says from 9:00 till 10:00: 32 cent.
My own calculation says: 0,32148 euro/kWh
I see no difference.

hfvdv, 2023-12-14:
Just received some feedback from tibber. See below but long story short i
am apparently in the 10.000 - 50.000 bracket which means my
milieu belasting is lower.

I need to have a think about how i can adjust the rate based on the current
yearly consumption.

and thanks for double checking on your side as well.

will let you know how i will solve this topic if you find this useful.

H

Goedemorgen Hendrick,

Bedankt voor je bericht. Het verschil zit hem in in de milieubelasting. De
prijzen die we op de website tonen zijn gebaseerd op een standaard verbruik
onder de 10.000 kWh per jaar. Deze is dus niet aangepast aan jouw
persoonlijke situatie. De prijs in de app is echter wel gebaseerd op jouw
persoonlijke verbruik en jij valt onder de 10.000 tot 50.000 kWh categorie.

Belasting op elektriciteit 2023:
Verbruik tot 10.000 kWh per jaar: € 0,12599
Verbruik van 10.000 tot 50.000 kWh per jaar: € 0,10046
Verbruik van 50.000 tot 10.000.000 kWh per jaar: € 0,03942

Wanneer je de prijzen zonder belastingen en extra kosten bekijkt zul je
zien dat deze wel gelijk zijn.

Mocht je verder nog vragen hebben, dan verneem ik het graag. Anders wens ik
je een fijne dag! ✨

Met vriendelijke groet,
Rena

corneel27, 2023-12-14:
Are you using my software for day ahead optimalisation?
In that case I will adept the calculation and take at least the lower energytax for the bracket 10.000 - 50.000 kWh into account.

hfvdv, 2023-12-14:
yes i am and so far really happy with it but in the initial testing phases.

H

closed 2024-01-05 by corneel27