corneel27/day-ahead #70: Error melding

pvhpsb, 2024-02-08:
Ik zal wel een fout hebben in de config, want ik zie in het logboek *** _mysql_connector.MySQLInterfaceError: Unknown column 't2.start_ts' in 'field list'** kan iemand mij op weg helpen?
Mvg, Patrick

corneel27, 2024-02-08:
Dag Patrick,
Kun je de hele foutmelding uit de logging inclusief trace  hier posten?
Groeten van Cees

pvhpsb, 2024-02-08:
Bij dezeThe above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/webserver/app/routes.py", line 339, in api_report
    result = report.get_api_data(fld, periode, cumulate=cumulate)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_report.py", line 692, in get_api_data
    df_grid = self.get_grid_data(periode)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_report.py", line 475, in get_grid_data
    df_ha = self.get_sensor_data(sensor, last_moment, tot, "consumption")
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_report.py", line 183, in get_sensor_data
    df = self.db_ha.run_select_query(sql)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/db_manager.py", line 417, in run_select_query
    self._c.execute(sql)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/mysql/connector/cursor_cext.py", line 330, in execute
    result = self._cnx.cmd_query(
             ^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/mysql/connector/opentelemetry/context_propagation.py", line 77, in wrapper
    return method(cnx, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/mysql/connector/connection_cext.py", line 641, in cmd_query
    raise get_mysql_exception(
mysql.connector.errors.ProgrammingError: 1054 (42S22): Unknown column 't2.start_ts' in 'field list'Is dit voldoende? Mvg, PatrickVerstuurd vanaf mijn iPad

corneel27, 2024-02-08:
Ja dit is voldoende.
Welk commando gaf je op het dashboard?
Heeft het dashboard daarvoor wel gewerkt?
Staat de configuratie naar de home Assistant database goed?

corneel27, 2024-02-08:
Ik heb nog eens goed naar de foutmelding gekeken en hier lokaal nagegaan wat er fout kan zijn.
Bovenstaande genoemde mogelijke fouten geven een andere foutmelding. 
Ik vermoed nu dat je een oude versie van Home Assistant draait, waarin  het veld **start_ts** nog niet in de database is opgenomen (deze is later toegevoegd).
Welke versie draai je? 
Dat kun je achterhalen op je eigen lokale pagina van Home Assistent bij \Instellingen\Over.
En dan gaat het over het versienummer achter **Core**
[image]
Het programma draait alleen als je een recente versie van HA minimaal 2023.12.0 of later draait.

pvhpsb, 2024-02-11:
Hoi,
Alles verwijderd en opnieuw geinstaleerd.
Programma loopt en elk uur een update.
Grafisch ook te zien maar Report geeft Internal Server Error

http://192.168.178.133:5000/api/report/da/vandaag
Werkt

http://192.168.178.133:5000/api/report/cost/vandaag
http://192.168.178.133:5000/api/report/production/vandaag
http://192.168.178.133:5000/api/report/consumption/vandaag
http://192.168.178.133:5000/api/report/profit/vandaag
Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Maar in in logboek blijft de melding.

Unknown column 't2.start_ts' in 'field list

Mvg,Patrick

pvhpsb, 2024-02-12:
> Dag Patrick,
> Het programma rekent dus goed?
> Maar de "report" functie werkt niet goed.
> Die foutmelding  geeft aan dat er iets mist of iets fout gaat richting de database van Home Assistant.
> Kun je mij de volgende info sturen, dan kijk ik met je mee:
> Het configuratiescherm van de addon MariaDB:
> Schermafdruk.van.2024-02-11.23-50-39.png (view on web)
> Als je webpagina van  de addon phpMyAdmin opent en dan van de database homeassistant de structuur en de eerste records van de tabel statistics:
> afbeelding.png (view on web)

Hallo  Cees,Tabel lijkt leegIs dit het probleem? Zoja hoe op te lossen.Mvg PatrickVerstuurd vanaf mijn iPad

pvhpsb, 2024-02-14:
Ik ben er nog niet aan uit.

  "tibber": {
    "api url": ["https://api.tibber.com/v1-beta/gql"](https://api.tibber.com/v1-beta/gql),
    "api_token": "!secret tibber_api_token"
  },

  "report": {
    "entities grid consumption": [
      "sensor.electricity_meter_energy_consumption_tarif_1",
      "sensor.electricity_meter_energy_consumption_tarif_2"
    ],
    "entities grid production": [
      "sensor.electricity_meter_energy_production_tarif_1",
      "sensor.electricity_meter_energy_production_tarif_2"
    ]
  },
  "scheduler": {
    "0355": "get_meteo_data",
    "0955": "get_meteo_data",

Dit is het zelfde als de setting in het Energie Bord van HA

Hierbij de uitdraai van de optimalisatie.

Day Ahead Optimalisatie gestart: 13-02-2024 17:00:00 :  calc_optimum
Locatie:  51.59768094857052:4.725751876831056

Prognose data:
                  tijd        time  temp  glob_rad    pv_rad  da_price
0  2024-02-14 00:00:00  1707865200   8.0       0.0   0.00000   0.06063
1  2024-02-14 01:00:00  1707868800   8.0       0.0   0.00000   0.05934
2  2024-02-14 02:00:00  1707872400   8.0       0.0   0.00000   0.05576
3  2024-02-14 03:00:00  1707876000   8.0       0.0   0.00000   0.05479
4  2024-02-14 04:00:00  1707879600   8.0       0.0   0.00000   0.05506
5  2024-02-14 05:00:00  1707883200   9.0       0.0   0.00000   0.05850
6  2024-02-14 06:00:00  1707886800   9.0       0.0   0.00000   0.06352
7  2024-02-14 07:00:00  1707890400   9.0       0.0   0.00000   0.07298
8  2024-02-14 08:00:00  1707894000   9.0       0.0   0.00000   0.07989
9  2024-02-14 09:00:00  1707897600   9.0       5.0   5.00000   0.06965
10 2024-02-14 10:00:00  1707901200   9.0      11.0  12.84650   0.06386
11 2024-02-14 11:00:00  1707904800   9.0       9.0  10.51080   0.06329
12 2024-02-14 12:00:00  1707908400  10.0      16.0  18.68580   0.06216
13 2024-02-14 13:00:00  1707912000  11.0      21.0  24.52520   0.06191
14 2024-02-14 14:00:00  1707915600  11.0       7.0   8.17505   0.06729
15 2024-02-14 15:00:00  1707919200  11.0       9.0  10.51080   0.07069
16 2024-02-14 16:00:00  1707922800  11.0       2.0   2.00000   0.07599
17 2024-02-14 17:00:00  1707926400  11.0       0.0   0.00000   0.08165
18 2024-02-14 18:00:00  1707930000  11.0       1.0   1.00000   0.08767
19 2024-02-14 19:00:00  1707933600  11.0       0.0   0.00000   0.08458
20 2024-02-14 20:00:00  1707937200  11.0       0.0   0.00000   0.07660
21 2024-02-14 21:00:00  1707940800  11.0       0.0   0.00000   0.06971
22 2024-02-14 22:00:00  1707944400  11.0       0.0   0.00000   0.06820
23 2024-02-14 23:00:00  1707948000  11.0       0.0   0.00000   0.06380

Prijs levering:
[0.23507,
 0.23351,
 0.22917,
 0.228,
 0.22833,
 0.23249,
 0.23856,
 0.25001,
 0.25837,
 0.24598,
 0.23897,
 0.23829,
 0.23692,
 0.23662,
[... 151 lines cut ...]

Zie je nog andere fouten?

Mvg,Patrick

Op 12-2-2024 om 11:23 schreef Cees van Beek:
>
> Ja hij zou niet leeg mogen/moeten zijn.
> Heb jij de "history" in home assistant uitstaan?
> Default zou die aan moeten staan.
> Lees anders dit even door:
> https://www.home-assistant.io/integrations/history/
> En heb jij het energie dashboard van Home Assistant geactiveerd.
> Dat dashboard werkt alleen als "history" aan staat.
> Bovendien moet je in het options.json bestand de de volgende report entiteiten invullen:
>   "report": {
>     "entities grid consumption": [
>       "sensor.grid_consumption_low......",
>       "sensor.grid_consumption_high....."
>     ],
>     "entities grid production": [
>       "sensor.grid_production_low.......",
>       "sensor.grid_production_high...."
>     ]
>

corneel27, 2024-02-14:
Het ziet er goed uit!
consumption none en production none worden veroorzaakt door het ontbreken van deze data in de database. Omdat er geen data zijn opgehaald bij Tibber. 
Heb jij een andere leverancier of heb je het ophalen uitstaan in de scheduler. Als het wel aanstaat wat staat er in de logging na het ophalen? (Je kunt met een explorer via samba bij de map config_addons).
Die koppeling met Tibber vervalt in de volgende versie.

corneel27, 2024-02-14:
In de volgende versie van de dao-addon zal de verplichte band met Tibber zijn doorgeknipt (nog ca 1 week wachten) en dan kun je (als alles goed is ingesteld) volledig gebruik maken  van de data van de slimme meter die in HA worden gelogd.

Bovenstaande foutmelding ontstaat  omdat de Tibber-api geen data terugstuurt.
Ik zal zelf wat meer werk gaan besteden aan adequate foutmeldingen in dit soort situaties.

Kun jij inloggen op de developer-webpagina van Tibber met je email-adres en wachtwoord:
https://developer.tibber.com/ ?
Kies om in te loggen op de pagina rechtsboven "**sign in**" en op de volgende pagina kies je voor "**Api Explorer**"
Op de daarna volgende pagina  staat het access.token op de regel onder Api Explorer.
Klopt dat access token met wat je hebt ingevuld bij de instellingen in het secret-bestand bij **tibber_api_token**?

Als dat allemaal goed is kies in het drop-down menu rechts voor **Home/Price/Conusmption**
Klik dan op de pijl -knop naast GraphiQL en dan zou je jouw recentse gegevens bij Tibber te zien moeten krijgen.
Lukt dat allemaal?

pvhpsb, 2024-02-22:
Sorry, het heeft even geduurd.
Ik gebruik geen Tibber maar ga binnenkort naar ANWB.
Ik heb het nu draaiend met 2 Sessy's en vanmiddag is er ontladen, nu kijken of het vannacht goed gaat met laden.
Het rapport werkt nog niet, nog steeds  ****Unknown column 't2.start_ts' ***
Ik wacht wel op de aanpassing voor de HA energie data.
Voor nu hartelijk dank voor get geweldige programma.
Mvg, Patrick

storeman, 2026-11-11:
Probleem lijkt te zijn opgelost of niet meer relevant. Sluit dit issue.

closed 2026-11-11 by storeman