corneel27/day-ahead #73: EV Maximale Amp

tonvanboven, 2024-02-18:
Gezien de prijzen wil DAO volgens haar eigen berekeningen om 9 uur de auto een klein stukje laden om vervolgens om 10 en 11u volle bak te gaan.

[image]

Echter DAO zet om 9 uur de stroom niet omlaag:

[image]

In options is dit de entity die de stroom zou moeten aansturen:

[image]

Klopt het dat DAO de stroom (nog?) niet goed aanstuurt?

corneel27, 2024-02-18:
Ik heb dit wel op mijn lijstje staan, maar ik heb het nog niet geprogrammeerd.
Dus het werkt nog niet!
Mijn idee is dat je bij de instellingen een aantal "stappen" kunt definieren met per stap het max aantal ampere (per fase) en het rendement (mag gelijk zijn). Ik kies voor die stappen omdat er bij mij maar 5 stappen zijn (0, 6, 10, 13  en 16 A).
Bij jou zijn dat dan 17 stappen (0..16)
Verder moet er een entity bijkomen, waarin dao wegschrijft met hoeveel ampere er daadwerkelijk geladen moet gaan worden. 
Denk je dat dat zo gaat werken?

tonvanboven, 2024-02-18:
De huidige entity 'entity_max_amperage' kan volgens mij worden 'entity_amperage' waarmee DAO de auto aan gaat sturen. Het maximale kan dan toch gehaald worden uit instelling waarin de stappen staan? In mijn geval zou ik toch voor 16 stappen gaan, want 0A komt overeen met de status 'niet laden'. Als er geladen wordt is het 1, 2 tot en met 16 A.
En dan graag éérst de laadstroom instellen vanuit DAO en dan vervolens auto starten met laden. Tijdens het laden kan de laadstroom aangepast worden.

dewi-ny-je, 2026-09-15:
@tonvanboven is het probleem opgelost? kann dit ticket worden "opgelost"?

dewi-ny-je, 2026-09-15:
en ook: ga je de verbeteringen van jouw fork hier delen? via PRs

storeman, 2026-11-12:
Laten we dit ticket sluiten aangezien er veel is gewijzigd sinds februari 2024.