corneel27/day-ahead #74: "notification entity" stopt

tonvanboven, 2024-02-19:
Ik heb in HA een watchdog geknoopt aan de "notification entity" omdat de "last activity entity" het al enige tijd niet meer doet, en om een onverklaarbare wijze is om vijf uur vanmorgen de melding 'calc gestart' niet doorgekomen

[image]

Vervolgens gaat om 5:01 uur in HA het watchdog alarm af en heb met de hand om 5:02 de berekening gestart.

Echter blijkt nu dat er wél om vijf uur een berekening plaats gevonden heeft omdat er een logbestand is gemaakt

[image]

Door een herstart van de Add-on is het 'gerepareerd'. Dit is de tweede keer dat het mij is overkomen. Kun je eens kijken Cees of er in de code iets mis is ? Ik het eventueel wel vanuit HA oplossen door bij een watchdog alarm de Add-on te herstarten maar dat is naar mijn mening niet de manier om mogelijke knelpunten in software op te lossen.

corneel27, 2024-02-19:
Het zou zelfs zo kunnen zijn dat dan de verbinding met het logbestand wordt verbroken (de log- en grafiekbestanden staan net als de instellingen buiten de addon zodat bij een nieuwe versie deze intakt blijven) .
Dat kan ik testen door een logbestand lokaal aan te maken en na het genereren van de grafiek deze te verplaatsen naar de "permanente" locatie.

corneel27, 2024-02-19:
En er is ook een "png" bestand (optimum0500.png)?

corneel27, 2024-02-19:
Dat is interessante info. 
Dat zou erop kunnen duiden dat het programma uit zijn grafische resources loopt.
De log-bestanden gaan wel ieder uur door?

tonvanboven, 2024-02-19:
> Dat is interessante info. Dat zou erop kunnen duiden dat het programma uit zijn grafische resources loopt. De log-bestanden gaan wel ieder uur door?

Ik heb zojuist verkeerde info gegeven over het logbestand (de post hierboven aangepast). Vanaf 5:00 zijn er geen logbestanden én png bestanden meer gegenereerd. Totdat DAO Add-On is herstart.
My two cents: schedular loopt niet meer, DAO reageert echter wél op een API call om te herberekenen.

corneel27, 2024-02-19:
Het programma met scheduler is een eigen proces dat ook de berekening uitvoert als de scheduler zegt dat zijn tijd gekomen is.
Als je een berekening start vanaf het dashboard of via de api dan wordt daarvoor een apart proces gestart dat stopt als ie klaar.
Daarom zou het interessant zijn om te zien of en zo ja wat er in de logging van de addon zelf staat. Waarschijnlijk is door een fout het schedulerproces gestopt.

tonvanboven, 2024-02-19:
Ik heb in [HA](https://www.home-assistant.io/integrations/system_log/) het aantal logregels van standaard 50 naar 500 verzet en hoop daarmee te vangen wat er in de addon gebeurd.

storeman, 2026-11-12:
Ik sluit deze ook als stale. Er is inmiddels veel gewijzigd in de watchdog en de rest van de code.

closed 2026-11-12 by storeman