corneel27/day-ahead #79: Weergave laatste berekening

tonvanboven, 2024-04-01:
De laatste weergave komt niet in beeld, zelfs niet als je op de knop drukt

[image]

In de map staan deze bestanden:

[image]

De bestanden worden geschoond, die van 24 maart staat niet meer in de map. Het herstarten van de Add-on geeft geen verbetering.
 
Kijk ik met Studio Code naar het bestand van 1 april 18:00u dat ziet dit er normaal uit. 
[image]

Ik heb dit al een keer eerder gehad, en het is geen 1 april grap.

corneel27, 2024-04-01:
Het lijkt wel of ie naar een andere (oude) folder zit te kijken.
Volgens mij is het ook niet toevallig dat je een week geleden een nieuwe versie hebt geinstalleerd.
Misschien gaat er iets fout tijdens de installatie?
Of heb je misschiende de develop versie (addon branche) naast de productie versie (main branche) staan? 
Zou je eens willen kijken of er misschien meer dao directories zijn in addon_configs?
Je kunt op twee manieren naar de webgui gaan:
- via de zijbalk
- via de informatiepagina van de addon en onderaan op die pagina klikken op webgui

Geven beide hetzelfde resultaat?
Ik hoor graag, want ik ben benieuwd en ik wil het graag opgelost hebben.

corneel27, 2024-04-02:
Goede suggestie, neem ik mee naar de volgende (test)versie!

closed 2024-04-02 by corneel27

corneel27, 2024-04-02:
Ik heb de fix zowel naar de addon als naar de main-branche gepushed.