corneel27/day-ahead #82: Update WEbGui

pvhpsb, 2024-04-02:
Ook na de update naar 3.1 wordt de WebGui bij mijn niet ververst. Mij vraag is bij anderen wel?

corneel27, 2024-04-02:
Ik heb zelf het proleem helemaal niet.
Ton van Boven denkt dat het aan de browser Firefox ligt.
Om het probleem meer inzichtelijk te krijgen: graag  onderstaande vragen beantwoorden:
- Met welke browser werk jij?
- Treedt het probleem ook op een andere pc/laptop/mobiel  en/of een andere browser op?
- Als je via "RUN" een berekening uitvoert en je drukt daarna op "Home" welke grafiek heb je dan?

pvhpsb, 2024-04-02:
Ik gebruik op de PC Edge maar ook met Ipad of Iphone (Safari) gaat het fout.
Via "Run" komt het juiste scherm, maar na de "calc optimum" blijft hij hangen op het scherm/tijd van de laatste "Run"

corneel27, 2024-04-02:
Ik begrijp niet zo goed wat je bedoelt met "blijft hij hangen".
Dus je voert via "Run" een optimaliseringsberekening uit.
Je klikt als hij daar mee klars is (duurt enkele tot ca 15 seconden) op "Home".
Wat krijg je dan te zien?
Kun je een screenshot mee sturen?

pvhpsb, 2024-04-02:
[image]
[image]
Om  21:34 heb ik een “run” gedaan daarna is om 10:00 automatisch een “calc optimum” gedaan.
Daarna blijft de Webgui op 21:34 staan

corneel27, 2024-04-02:
Heb je samba geinstalleerd?
Kun je met je file explorer op de HA machine naar de volgende directory gaan:
/addon_configs/94e386ba_day_ahead_opt/dao_data/images
Als je daar de bestanden sorteert op datum/tijd
Kun je daar dan een screenshot van maken?

pvhpsb, 2024-04-02:
Screenshot van midnigt commander
[image]

pvhpsb, 2024-04-02:
Screenshot van midnigtcommander
[image]

corneel27, 2024-04-02:
Ik denk dat er iets fout zit in je configuratie.
Zou de de addon opnieuw willen opstarten: op de addon informatie pagina links onder "herstarten".
En dan kijken wat het logboek zegt.
En dan kijken wat er op het volgende hele uur gebeurt (evt ook in logboek en de logfile van de berekening).

pvhpsb, 2024-04-03:
[image]
[image]
[image]
[image]

corneel27, 2024-04-03:
Ziet er allemaal goed uit!
Ben wel benieuwd wat er in de logfile van vandaag 10 uur staat: cal_optimun1000.log.

pvhpsb, 2024-04-03:
[calc_optimum1000.log](https://github.com/corneel27/day-ahead/files/14849694/calc_optimum1000.log)
[calc_optimum1100.log](https://github.com/corneel27/day-ahead/files/14849695/calc_optimum1100.log)

pvhpsb, 2024-04-03:
Grafiek blijft staan op 2 4   21.34

corneel27, 2024-04-03:
De berekeningen zijn volgens de log goed gegaan. 
Zijn er geen grafieken aangemaakt?
Staan er png-bestanden van 10 en 11 uur in de directory "images"?

corneel27, 2024-04-03:
Ik heb nog even goed naar de loggings gekeken.
Hij stopt na de melding "PV garage aangezet".
Wat heb je bij de instellingen ingevuld bij de pv van de garage  bij "entity pv switch"?
Check svp of die naam exact overeen komt met een bestaande input_boolean (of andere switch).

pvhpsb, 2024-04-03:
Sorry stomme fout van mij in "electric vehicle" had ik foute  "entity ready datetime"  staan. Daardoor stopte hij.
Loopt nu prima. nogmaals bedankt

corneel27, 2024-04-03:
Fijn dat het weer werkt.
Misschien een tip voor de volgende keer: verander steeds maar een ding (instelling of een nieuwe versie), maar nooit tegelijk. Na iedere verandering even testen, zodat je weet of het nog werkt.
Als er dan iets niet werkt weet je ook sneller waar je/wij de fout moeten zoeken.

closed 2024-04-03 by corneel27