corneel27/day-ahead #87: Ontbrekende gegevens

robbak2, 2024-04-10:
Sinds een paar dagen gaat het vanaf een uurtje of 16h mis. Als ik kies er staan dan ook maar een paar Nordpool uren afgedrukt en “er ontbreken enkele gegevens”
Ik heb de nordpool integratie ook los draaien en daarin is alles zichtbaar. Kies ik voor ophalen gegevens, dan komt er ook 24h over, maar gaat waarschijnlijk niet goed met wegschrijven ervan(?)

Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 1911, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 1872, in main
    day_ah.calc_optimum()
  File "/root/dao/webserver/../prog/day_ahead.py", line 1275, in calc_optimum
    df_accu[b].loc["Total"] = df_accu[b].sum(axis=0, numeric_only=True)
    ~~~~~~~~~~~~~~^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/indexing.py", line 911, in __setitem__
    iloc._setitem_with_indexer(indexer, value, self.name)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/indexing.py", line 1932, in _setitem_with_indexer
    self._setitem_with_indexer_missing(indexer, value)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/indexing.py", line 2295, in _setitem_with_indexer_missing
    value = value.reindex(index=self.obj.columns, copy=True)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/series.py", line 5144, in reindex
    return super().reindex(
           ^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/generic.py", line 5607, in reindex
    return self._reindex_axes(
           ^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/generic.py", line 5630, in _reindex_axes
    new_index, indexer = ax.reindex(
                         ^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/core/indexes/base.py", line 4426, in reindex
    raise ValueError("cannot handle a non-unique multi-index!")
ValueError: cannot handle a non-unique multi-index!

robbak2, 2024-04-10:
Zojuist via run-prijzen ophalen
Day Ahead Optimalisering versie: 2024.3.4
Day Ahead Optimalisering gestart op: 10-04-2024 18:38:46
[{'start': datetime.datetime(2024, 4, 9, 22, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 9, 23, 0, tzinfo=<UTC>), 'value': 12.9}, {'start': datetime.datetime(2024, 4, 9, 23, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 0, 0, tzinfo=<UTC>), 'value': 5.1}, {'start': datetime.datetime(2024, 4, 10, 0, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 1, 0, tzinfo=<UTC>), 'value': 5.0}, {'start': datetime.datetime(2024, 4, 10, 1, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 2, 0, tzinfo=<UTC>), 'value': 8.99}, {'start': datetime.datetime(2024, 4, 10, 2, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 3, 0, tzinfo=<UTC>), 'value': 24.7}, {'start': datetime.datetime(2024, 4, 10, 3, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 4, 0, tzinfo=<UTC>), 'value': 48.8}, {'start': datetime.datetime(2024, 4, 10, 4, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 5, 0, tzinfo=<UTC>), 'value': 74.75}, {'start': datetime.datetime(2024, 4, 10, 5, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 6, 0, tzinfo=<UTC>), 'value': 99.63}, {'start': datetime.datetime(2024, 4, 10, 6, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 7, 0, tzinfo=<UTC>), 'value': 106.7}, {'start': datetime.datetime(2024, 4, 10, 7, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 8, 0, tzinfo=<UTC>), 'value': 80.46}, {'start': datetime.datetime(2024, 4, 10, 8, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 9, 0, tzinfo=<UTC>), 'value': 67.64}, {'start': datetime.datetime(2024, 4, 10, 9, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 10, 0, tzinfo=<UTC>), 'value': 52.99}, {'start': datetime.datetime(2024, 4, 10, 10, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 11, 0, tzinfo=<UTC>), 'value': 19.21}, {'start': datetime.datetime(2024, 4, 10, 11, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 12, 0, tzinfo=<UTC>), 'value': 8.93}, {'start': datetime.datetime(2024, 4, 10, 12, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 13, 0, tzinfo=<UTC>), 'value': 0.0}, {'start': datetime.datetime(2024, 4, 10, 13, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 14, 0, tzinfo=<UTC>), 'value': -0.1}, {'start': datetime.datetime(2024, 4, 10, 14, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 15, 0, tzinfo=<UTC>), 'value': 22.0}, {'start': datetime.datetime(2024, 4, 10, 15, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 16, 0, tzinfo=<UTC>), 'value': 64.9}, {'start': datetime.datetime(2024, 4, 10, 16, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 17, 0, tzinfo=<UTC>), 'value': 82.65}, {'start': datetime.datetime(2024, 4, 10, 17, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 18, 0, tzinfo=<UTC>), 'value': 145.46}, {'start': datetime.datetime(2024, 4, 10, 18, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 19, 0, tzinfo=<UTC>), 'value': 141.1}, {'start': datetime.datetime(2024, 4, 10, 19, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 20, 0, tzinfo=<UTC>), 'value': 88.28}, {'start': datetime.datetime(2024, 4, 10, 20, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 21, 0, tzinfo=<UTC>), 'value': 72.35}, {'start': datetime.datetime(2024, 4, 10, 21, 0, tzinfo=<UTC>), 'end': datetime.datetime(2024, 4, 10, 22, 0, tzinfo=<UTC>), 'value': 65.1}]
            time code    value
0   1712700000.0   da  0.01290
1   1712703600.0   da  0.00510
2   1712707200.0   da  0.00500
3   1712710800.0   da  0.00899
4   1712714400.0   da  0.02470
5   1712718000.0   da  0.04880
6   1712721600.0   da  0.07475
7   1712725200.0   da  0.09963
8   1712728800.0   da  0.10670
9   1712732400.0   da  0.08046
10  1712736000.0   da  0.06764
11  1712739600.0   da  0.05299
12  1712743200.0   da  0.01921
13  1712746800.0   da  0.00893
14  1712750400.0   da  0.00000
15  1712754000.0   da -0.00010
16  1712757600.0   da  0.02200
17  1712761200.0   da  0.06490
18  1712764800.0   da  0.08265
19  1712768400.0   da  0.14546
20  1712772000.0   da  0.14110
21  1712775600.0   da  0.08828
22  1712779200.0   da  0.07235
23  1712782800.0   da  0.06510

robbak2, 2024-04-10:
run-berekenen met debug

Day Ahead Optimalisering versie: 2024.3.4
Day Ahead Optimalisering gestart op: 10-04-2024 18:40:37
Debug = True
Er ontbreken voor een aantal uur gegevens (meteo en/of dynamische prijzen)
 controleer of alle gegevens zijn opgehaald

Prognose data:
                 tijd        time  temp  glob_rad   pv_rad  da_price
0 2024-04-10 18:00:00  1712764800  14.0      34.0  33.8122   0.08265
1 2024-04-10 19:00:00  1712768400  13.0       1.0   1.0000   0.14546
2 2024-04-10 20:00:00  1712772000  12.0       0.0   0.0000   0.14110
3 2024-04-10 21:00:00  1712775600  12.0       0.0   0.0000   0.08828
4 2024-04-10 22:00:00  1712779200  11.0       0.0   0.0000   0.07235
5 2024-04-10 23:00:00  1712782800  11.0       0.0   0.0000   0.06510

Prijs levering:

corneel27, 2024-04-10:
Sorry, maar met het aanpassen van het scherm waarmee je de prijzen kunt ophalen is een kleine maar zeer storende fout geintroduceerd bij het nordpool-onderdeel.
Er komt binnenkort een definitieve oplossing met een nieuwe versie, maar voorlopig zijn er twee oplossingen:

1. stel entsoe in als "source", maar dan moet je wel eerst een api key aanvragen
2. haal na 13:00 uur de prijzen op viahet "run"-scherm en vul bij "prijzen_vanaf" de datum van morgen in.

Kun je hiermee even verder?

robbak2, 2024-04-10:
Bedankt voor je reactie. Ik wacht even rustig op een update; met punt 2 kan ik wel verder.
Ter info: Schakelen tussen entsoe en nordpool doet het bij mij niet, met entsoe duurt het ca 30s en dan komt ie met “bad gateway” (ik heb een key want afgelopen weken heb ik met entsoe gedraaid totdat HA een update had waardoor het niet meer werkte)

corneel27, 2024-04-11:
Er is een nieuwe versie (2024.3.5), waarin het probleem met het ophalen van de nordpool data is opgelost.

closed 2024-04-11 by corneel27