corneel27/day-ahead #89: IndexError in day_ah_calc_optimum

hfvdv, 2024-04-18:
reconfigured baseload calc and setup all the entities. No issues in running calc before adding entities. Not sure yet why or how the below is related.

Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 1911, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 1872, in main
    day_ah.calc_optimum()
  File "/root/dao/webserver/../prog/day_ahead.py", line 1065, in calc_optimum
    model += c_l[u] == c_t_total[u] + b_l[u] * hour_fraction[u] + \
                                      ~~~^^^
IndexError: list index out of range

corneel27, 2024-04-18:
I think the calculations of the baseloads went wrong.
Do you have installed samba on HA?
Then you can connect with a file explorer the share and folder
 ```//addon_configs/xxxxxday_ahead_opt/daodata/baseloads```
You should see 7 files: baseload_0.json ... baseload_6.json each about 218 bytes 
When the baseload calculation has run (time set in scheduler??)) there must be a log of the run in:
 ```//addon_configs/xxxxxday_ahead_opt/daodata/log```
 With content like:
 ```
Day Ahead Optimalisatie gestart: 18-04-2024 14:30:00 :  calc_baseloads
Locatie: latitude 52.006443579881406 longitude:5.984225435624042
baseload voor weekdag 0 : [0.205, 0.392, 0.233, 0.494, 0.126, 0.129, 0.127, 0.178, 0.184, 0.366, 0.239, 0.256, 0.224, 0.208, 0.203, 0.235, 0.24, 0.675, 0.297, 0.256, 0.246, 0.191, 0.185, 0.159]
baseload voor weekdag 1 : [0.178, 0.437, 0.227, 0.555, 0.152, 0.136, 0.138, 0.172, 0.203, 0.378, 0.397, 0.377, 0.328, 0.318, 0.319, 0.247, 0.248, 0.803, 0.261, 0.222, 0.219, 0.216, 0.194, 0.164]
baseload voor weekdag 2 : [0.169, 0.487, 0.178, 0.674, 0.127, 0.131, 0.132, 0.198, 0.199, 0.173, 0.307, 0.192, 0.227, 0.298, 0.213, 0.234, 0.356, 0.908, 0.317, 0.247, 0.224, 0.22, 0.191, 0.185]
baseload voor weekdag 3 : [0.163, 0.501, 0.233, 0.628, 0.149, 0.135, 0.143, 0.176, 0.308, 0.377, 0.246, 0.283, 0.309, 0.254, 0.203, 0.218, 0.203, 0.656, 0.241, 0.269, 0.228, 0.214, 0.169, 0.182]
baseload voor weekdag 4 : [0.157, 0.441, 0.219, 0.567, 0.148, 0.154, 0.137, 0.186, 0.269, 0.283, 0.461, 0.471, 0.276, 0.251, 0.285, 0.244, 0.305, 0.473, 0.245, 0.234, 0.22, 0.176, 0.17, 0.159]
baseload voor weekdag 5 : [0.153, 0.387, 0.205, 0.463, 0.13, 0.126, 0.132, 0.17, 0.294, 0.276, 0.523, 0.325, 0.35, 0.427, 0.382, 0.24, 0.192, 0.447, 0.334, 0.287, 0.244, 0.205, 0.185, 0.18]
baseload voor weekdag 6 : [0.196, 0.394, 0.343, 0.436, 0.195, 0.13, 0.137, 0.197, 0.285, 0.245, 0.344, 0.384, 0.358, 0.325, 0.336, 0.327, 0.311, 0.507, 0.242, 0.286, 0.237, 0.21, 0.182, 0.16]
```
First this all must be ok and until that moment you must:
- set `` "use_calc_baseload": "False"``` and 
- set your baseload in options.json

hfvdv, 2024-04-19:
clear and thanks. As i came from a somewhat older version i had to do two things
-	create the baseload folder ( however unsure this had not happened yet ). When running the "bereken baseloads" from the gui. It stopped the optimization calculation and crashed. That issue is now resolved.
-	I was indeed missing the scheduler options as that was not yet included in the file. In addittion i updated and added the baseload calculation options ( set use_baseloadcalc to false for now 
-	i ran the "berekenen baseloads". The 7 baseload files from the gui arer present and empty. No log file present.

will wait for the scheduler calculating the baseload to see if logfile is generated inc population of the 7 baseload files.

hfvdv, 2024-04-19:
adjusted the scheduler. Log did show up however baseload voor weekdagen X : [] <- all remain 'empty'

I did have the following observation but perhaps not relevant. Not sure if historical data is used but when i run the reports the following give me an internal server error:
- gisteren
- deze week
- deze maand
- dit jaar
- dit contract jaar
- 365 dagen

all others work and return a table

corneel27, 2024-04-19:
There is a specific report you can run of all is well set: Reports Balans:
[image]

You caan also look in the logging of the calculation what went wrong.

miclnl, 2024-06-06:
Same issue here I think, no error when running calculate baseloads, but no data either;
```
2024-06-06 14:24:14 info: baseload voor weekdag 0 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 1 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 2 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 3 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 4 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 5 :
2024-06-06 14:24:14 info: 
2024-06-06 14:24:14 info: baseload voor weekdag 6 :
2024-06-06 14:24:14 info: 

```
```
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_0.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_1.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_2.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_3.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_4.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_5.json
-rw-r--r--    1 root     root             3 Jun  6 14:24 baseload_6.json
baseload # cat baseload_0.json
[]
```

corneel27, 2024-06-06:
Thank you for sending this error-message.
There is going something wrong with the calculation of your baseloads for every weekday.
Your logging should look like:
`2024-06-05 23:40:08 info: baseload voor weekdag 0 :
2024-06-05 23:40:08 info: 0.232 0.313 0.266 0.387 0.116 0.117 0.112 0.192 0.249 0.406 0.275 0.254 0.32 0.308 0.305 0.226 0.251 0.539 0.188 0.228 0.197 0.186 0.138 0.161 
2024-06-05 23:40:16 info: baseload voor weekdag 1 :
2024-06-05 23:40:16 info: 0.163 0.342 0.162 0.451 0.138 0.127 0.111 0.193 0.254 0.253 0.326 0.314 0.332 0.344 0.329 0.27 0.32 0.621 0.234 0.198 0.204 0.191 0.155 0.147 
2024-06-05 23:40:25 info: baseload voor weekdag 2 :
2024-06-05 23:40:25 info: 0.157 0.418 0.182 0.589 0.126 0.124 0.125 0.233 0.272 0.207 0.37 0.427 0.401 0.343 0.3 0.3 0.312 0.642 0.202 0.217 0.217 0.208 0.176 0.15 
2024-06-05 23:40:33 info: baseload voor weekdag 3 :
2024-06-05 23:40:33 info: 0.147 0.459 0.184 0.623 0.155 0.133 0.124 0.198 0.259 0.428 0.299 0.267 0.31 0.322 0.301 0.249 0.225 0.432 0.191 0.235 0.232 0.193 0.149 0.16 
....
`
and a file must have a content like (baseload_0.json):
`
[
  0.232,
  0.313,
  0.266,
  0.387,
  0.116,
  0.117,
  0.112,
  0.192,
  0.249,
  0.406,
  0.275,
  0.254,
  0.32,
  0.308,
  0.305,
  0.226,
  0.251,
  0.539,
  0.188,
  0.228,
  0.197,
  0.186,
  0.138,
  0.161
]
`
I shall make a better to understand error-message when calculating the baseloads.

To better understand your situation can you give me some more information?
I  have the following  questions:

1. Have you entered the home assistant sensors for in the settings for your grid consumption, production and for several consumption categories. This all should be the same sensors which you ave entered for the HA-energy-dasboard. For instance:
'''
 "report":{
    "entities grid consumption" : [
      "sensor.grid_consumption_low",
      "sensor.grid_consumption_high"
    ],
    "entities grid production" : [
      "sensor.grid_production_low",
      "sensor.grid_production_high"
    ],
    "entities solar production ac": [
      "sensor.solaredge_woning_ac_energy_kwh",
      "sensor.solaredge_garage_ac_energy_kwh_2"
    ],
    "entities solar production dc": [],
    "entities ev consumption" : ["sensor.fritz_dect_200_laadpunt_total_energy"],
    "entities wp consumption" : ["sensor.youless_meterstand"],
    "entities boiler consumption": [],
    "entities battery consumption": ["sensor.ess_grid_consumption"],
    "entities battery production": ["sensor.ess_grid_production"]
  },
'''
2. For how many weeks are there data for these sensors in your database?
3. For how many weeks are there price data in your database? You can add additional pricedata with the following actions:
  - set (for the time of this action) entsoe as price-source in your settings
  - in the Run-menu of the dashboard fill near the rule "Day ahead prjzen ophalen" in "prijzen vanaf" and "prijzen tot"; the periode for which you want additonal pricedata
  - click the corresponding checkbox
For all these: you should have at least data for 8 weeks to get reliable results for your baseload calculation.
The pricedata should be present  for "vorige maand" if you want to calculate your results for last month etc.

miclnl, 2024-06-07:
Thanks for trouble shooting!
1. I didn't. I followed the configuration section in docs. I added the sensors and the baseload is calculating correctly now :)
2. ~2 years more or less
3. It crashed when I tried any larger range than 2024-05-20 - today (at least it crashed at 05-01 and larger).
```
[2024-06-07 09:07:54 +0200] [12] [CRITICAL] WORKER TIMEOUT (pid:16)
[2024-06-07 09:07:54 +0200] [16] [INFO] Worker exiting (pid: 16)
[2024-06-07 09:07:54 +0200] [12] [ERROR] Worker (pid:16) exited with code 1
```

It's working when using 2024-05-20 - today and stepwise adding april and may afterwards.

I'm unfortunatelystill getting the same error when trying to open the reports-tab.

```
MariaDB [day_ahead]> select count(*),MIN(time),MAX(time) FROM `values`;
+----------+------------+------------+
| count(*) | MIN(time)  | MAX(time)  |
+----------+------------+------------+
|     2508 | 1704063600 | 1717862400 |
+----------+------------+------------+
```

corneel27, 2024-06-07:
There is a logging page of the addon, which you can reach in HA; "Settings" / "Add-ons" / "Day-ahead-optimization" and then choice "Logbook" (in the right upper corner). 
Wil you first try to reach the "report" option in the Dashboard and then look wat the "logbook" is telling.
Can you share that information?

corneel27, 2024-06-07:
Your min(time) is 1704063600.   Converted: 1 january 2024

miclnl, 2024-06-07:
This is from the logs
```
[2024-06-07 21:41:47,317] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 2190, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1486, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1484, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1469, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/webserver/app/routes.py", line 119, in menu
    return reports()
           ^^^^^^^^^
  File "/root/dao/webserver/../../dao/webserver/app/routes.py", line 284, in reports
    filtered_df = report.calc_grid_columns(report_df, active_interval, active_view)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_report.py", line 848, in calc_grid_columns
    col_4 = (row.consumption * (row.price + taxes_l + ol_l)) * (1 + btw / 100)
                                ^^^^^^^^^
AttributeError: 'Pandas' object has no attribute 'price'
```

I've downloaded everything between 1-4-2024 and 7-6-2024 now. 
```
MariaDB [day_ahead]> select count(*),MIN(time),MAX(time) FROM `values`;
+----------+------------+------------+
| count(*) | MIN(time)  | MAX(time)  |
+----------+------------+------------+
|      983 | 1714514400 | 1717927200 |
+----------+------------+------------+

Max = Sunday, June 9, 2024, is that correct?

```

 Still getting the same error, also on calc_optimum now

```
2024-06-07 22:33:26 debug: python pad:['/root/dao/prog', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '/root/dao/venv/day_ahead/lib/python3.11/site-packages', '/root']
2024-06-07 22:33:26 info: Day Ahead Optimalisering versie: 2024.5.4
2024-06-07 22:33:26 info: Day Ahead Optimalisering gestart op: 07-06-2024 22:33:26
2024-06-07 22:33:26 info: Day Ahead Optimalisatie gestart: 07-06-2024 22:33:26 taak: calc_optimum_met_debug
2024-06-07 22:33:26 debug: Starting new HTTP connection (1): supervisor:80
2024-06-07 22:33:26 debug: http://supervisor:80 "GET /core/api/ HTTP/1.1" 200 26
2024-06-07 22:33:26 debug: Starting new HTTP connection (1): supervisor:80
2024-06-07 22:33:26 debug: http://supervisor:80 "GET /core/api/config HTTP/1.1" 200 4702
2024-06-07 22:33:26 debug: Starting new HTTP connection (1): supervisor:80
2024-06-07 22:33:26 debug: http://supervisor:80 "POST /core/api/services/input_datetime/set_datetime HTTP/1.1" 200 2
2024-06-07 22:33:26 info: Debug = True
--- Logging error ---
Traceback (most recent call last):
  File "/usr/lib/python3.11/logging/__init__.py", line 1110, in emit
    msg = self.format(record)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 953, in format
    return fmt.format(record)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 687, in format
    record.message = record.getMessage()
                     ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/logging/__init__.py", line 377, in getMessage
    msg = msg % self.args
          ~~~~^~~~~~~~~~~
TypeError: not all arguments converted during string formatting
Call stack:
  File "/root/dao/webserver/../prog/day_ahead.py", line 1890, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 1868, in main
    da_calc.run_task_function("calc_optimum_met_debug")
  File "/root/dao/prog/da_base.py", line 416, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 364, in calc_optimum_met_debug
    dacalc.calc_optimum()
  File "/root/dao/prog/day_ahead.py", line 50, in calc_optimum
    logging.error("Er ontbreken voor een aantal uur gegevens (meteo en/of dynamische prijzen)\n",
Message: 'Er ontbreken voor een aantal uur gegevens (meteo en/of dynamische prijzen)\n'
Arguments: ('er kan niet worden gerekend',)
2024-06-07 22:33:28 fout: not all arguments converted during string formatting
2024-06-07 22:33:28 debug: Starting new HTTP connection (1): supervisor:80
2024-06-07 22:33:28 debug: http://supervisor:80 "POST /core/api/services/input_text/set_value HTTP/1.1" 200 2
2024-06-07 22:33:28 fout:  <class 'TypeError'>. not all arguments converted during string formatting, regelnummer: 416
2024-06-07 22:33:28 debug: Starting new HTTP connection (1): supervisor:80
2024-06-07 22:33:28 debug: http://supervisor:80 "POST /core/api/services/input_text/set_value HTTP/1.1" 200 2
```

corneel27, 2024-06-07:
Apparently there are prognose data missing.
Did you get and recieved the meteo-data?
Can you share the most recent logging of getting the meteo data?
Can you share the most recent logging of getting the price info?
Can you perform this query on the sql database:
```
SELECT `time`, `value`,  from_unixtime(`time`) 'begin' 
FROM `values` , `variabel` 
WHERE `variabel`.`code` = 'temp' AND `values`.`variabel` = `variabel`.`id` 
AND time >= UNIX_TIMESTAMP('2024-06-08 00:00:00')
ORDER BY `time` ;
```
 and the some query where you use **da** in place of **temp**?
 Can you share the results?

miclnl, 2024-06-08:
> Did you get and recieved the meteo-data?

I believe so.
```
2024-06-08 18:31:51 debug: python pad:['/root/dao/prog', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '/root/dao/venv/day_ahead/lib/python3.11/site-packages', '/root']
2024-06-08 18:31:51 info: Day Ahead Optimalisering versie: 2024.5.4
2024-06-08 18:31:51 info: Day Ahead Optimalisering gestart op: 08-06-2024 18:31:51
2024-06-08 18:31:51 debug: Locatie: ...
2024-06-08 18:31:51 info: Day Ahead Optimalisatie gestart: 08-06-2024 18:31:51 taak: get_meteo_data
2024-06-08 18:31:51 debug: Starting new HTTPS connection (1): data.meteoserver.nl:443
2024-06-08 18:31:51 debug: https://data.meteoserver.nl:443 "GET /api/uurverwachting.php?lat=.... HTTP/1.1" 200 None
2024-06-08 18:31:51 debug: {"plaatsnaam":[{"plaats":".."}],"data":[{"tijd":"1717866000","tijd_nl":"08-06-2024 19:00","offset":"13","loc":"none","temp":"16","winds":"5","windb":"3","windknp":"10","windkmh":"18","windr":"295","windrltr":"W","vis":"50000","neersl":"0.1","luchtd":"1010.7","luchtdmmhg":"758.1","luchtdinhg":"29.85","hw":"0","mw":"0","lw":"100","tw":"100","rv":"67","gr":"33","gr_w":"91","cape":"-","snd":"0","snv":"0","cond":"1","ico":"1","samenv":"Regen","icoon":"regen"},{"tijd":"1717869600","tijd_nl":"08-06-2024 20:00","offset":"14","loc":"none","temp":"15","winds":"4","windb":"3","windknp":"8","windkmh":"14.4","windr":"304","windrltr":"NW","vis":"50000","neersl":"0","luchtd":"1011.2","luchtdmmhg":"758.5","luchtdinhg":"29.86","hw":"0","mw":"0","lw":"100","tw":"100","rv":"74","gr":"5","gr_w":"14","cape":"-","snd":"0","snv":"0","cond":"2","ico":"2","samenv":"Bewolkt","icoon":"bewolkt"},{"tijd":"1717873200","tijd_nl":"08-06-2024 21:00","offset":"15","loc":"none","temp":"14","winds":"3","windb":"2","windknp":"6","windkmh":"10.8","windr":"311","windrltr":"NW","vis":"50000","neersl":"0","luchtd":"1011.8","luchtdmmhg":"758.9","luchtdinhg":"29.88","hw":"0","mw":"0","lw":"0","tw":"0","rv":"75","gr":"2","gr_w":"6","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Zonnig","icoon":"zonnig"},{"tijd":"1717876800","tijd_nl":"08-06-2024 22:00","offset":"16","loc":"none","temp":"13","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"281","windrltr":"W","vis":"50000","neersl":"0","luchtd":"1012.2","luchtdmmhg":"759.2","luchtdinhg":"29.89","hw":"0","mw":"0","lw":"0","tw":"0","rv":"80","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717880400","tijd_nl":"08-06-2024 23:00","offset":"17","loc":"none","temp":"11","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"260","windrltr":"W","vis":"50000","neersl":"0","luchtd":"1012.5","luchtdmmhg":"759.4","luchtdinhg":"29.9","hw":"0","mw":"0","lw":"0","tw":"0","rv":"85","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717884000","tijd_nl":"09-06-2024 00:00","offset":"18","loc":"none","temp":"10","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"256","windrltr":"ZW","vis":"50000","neersl":"0","luchtd":"1012.7","luchtdmmhg":"759.6","luchtdinhg":"29.91","hw":"0","mw":"0","lw":"0","tw":"0","rv":"89","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717887600","tijd_nl":"09-06-2024 01:00","offset":"19","loc":"none","temp":"9","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"249","windrltr":"ZW","vis":"50000","neersl":"0","luchtd":"1012.9","luchtdmmhg":"759.7","luchtdinhg":"29.91","hw":"0","mw":"0","lw":"0","tw":"0","rv":"92","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717891200","tijd_nl":"09-06-2024 02:00","offset":"20","loc":"none","temp":"8","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"249","windrltr":"ZW","vis":"50000","neersl":"0","luchtd":"1012.8","luchtdmmhg":"759.7","luchtdinhg":"29.91","hw":"0","mw":"0","lw":"0","tw":"0","rv":"93","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717894800","tijd_nl":"09-06-2024 03:00","offset":"21","loc":"none","temp":"8","winds":"2","windb":"2","windknp":"4","windkmh":"7.2","windr":"237","windrltr":"ZW","vis":"50000","neersl":"0","luchtd":"1012.7","luchtdmmhg":"759.6","luchtdinhg":"29.91","hw":"0","mw":"0","lw":"0","tw":"0","rv":"94","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717898400","tijd_nl":"09-06-2024 04:00","offset":"22","loc":"none","temp":"8","winds":"3","windb":"2","windknp":"6","windkmh":"10.8","windr":"232","windrltr":"ZW","vis":"5469","neersl":"0","luchtd":"1012.5","luchtdmmhg":"759.4","luchtdinhg":"29.9","hw":"0","mw":"0","lw":"0","tw":"0","rv":"94","gr":"0","gr_w":"0","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717902000","tijd_nl":"09-06-2024 05:00","offset":"23","loc":"none","temp":"8","winds":"3","windb":"2","windknp":"6","windkmh":"10.8","windr":"233","windrltr":"ZW","vis":"4784","neersl":"0","luchtd":"1012.3","luchtdmmhg":"759.3","luchtdinhg":"29.89","hw":"0","mw":"0","lw":"0","tw":"0","rv":"95","gr":"6","gr_w":"17","cape":"-","snd":"0","snv":"0","cond":"5","ico":"7","samenv":"Helder","icoon":"helderenacht"},{"tijd":"1717905600","tijd_nl":"09-06-2024 06:00","offset":"24" [... cut ...]
2024-06-08 18:38:40 debug: Starting new HTTP connection (1): supervisor:80
2024-06-08 18:38:40 debug: http://supervisor:80 "POST /core/api/services/input_text/set_value HTTP/1.1" 200 2
2024-06-08 18:38:40 fout:  <class 'IndexError'>. list index out of range, regelnummer: 416

2024-06-08 18:38:40 debug: Starting new HTTP connection (1): supervisor:80
2024-06-08 18:38:40 debug: http://supervisor:80 "POST /core/api/services/input_text/set_value HTTP/1.1" 200 2

```

Any way to check which list is out of range? Do I have another variable set wrong?

corneel27, 2024-06-08:
Thank you for this information.
From the logging above:

> 2024-06-08 18:38:37 info: Zelf berekende baseload
2024-06-08 18:38:37 info: lengte prognose arrays:
2024-06-08 18:38:37 info: uur: 30
2024-06-08 18:38:37 info: tijd: 30
2024-06-08 18:38:37 info: p_l: 30
2024-06-08 18:38:37 info: p_t: 30
2024-06-08 18:38:37 info: base: 24
2024-06-08 18:38:37 info: pv_ac: 30

The baseload array is just 24 items and neads at least 30 items (just as the others).
My conclusion there is something wrong with your baseload calculation.
Can you (with Samba) inspect the folder `94e386ba_day_ahead_opt/dao_data/baseload`?
Are there 7 files with each about 215 bytes?
Can you share the logging of the daily baseload calculation?

miclnl, 2024-06-09:
```
-rw-r--r--    1 root     root           231 Jun  9 09:30 baseload_0.json
-rw-r--r--    1 root     root           248 Jun  9 09:30 baseload_1.json
-rw-r--r--    1 root     root           259 Jun  9 09:30 baseload_2.json
-rw-r--r--    1 root     root           229 Jun  9 09:30 baseload_3.json
-rw-r--r--    1 root     root           230 Jun  9 09:30 baseload_4.json
-rw-r--r--    1 root     root           241 Jun  9 09:30 baseload_5.json
-rw-r--r--    1 root     root           201 Jun  9 09:30 baseload_6.json
```
length ([ and ] lines included)
```
baseload # wc -l *.json
       24 baseload_0.json
       26 baseload_1.json
       26 baseload_2.json
       24 baseload_3.json
       23 baseload_4.json
       25 baseload_5.json
       22 baseload_6.json
      170 total
```

logs
```
2024-06-09 14:12:39 info: Day Ahead Optimalisatie gestart: 09-06-2024 14:12:39 taak: calc_baseloads
2024-06-09 14:12:40 info: baseload voor weekdag 0 :
2024-06-09 14:12:40 info: 0.129 0.122 0.225 0.161 27.018 139.029 215.023 466.734 488.495 802.491 655.996 1187.184 1280.791 1124.401 720.083 326.784 391.681 237.384 66.82 12.34 0.358 0.149 
2024-06-09 14:12:41 info: baseload voor weekdag 1 :
2024-06-09 14:12:41 info: 0.153 0.153 0.135 0.12 0.311 0.255 24.246 114.516 310.31 808.796 810.549 828.653 1872.406 1267.592 576.365 775.859 1053.625 922.378 614.163 232.991 34.277 1.532 0.182 0.151 
2024-06-09 14:12:42 info: baseload voor weekdag 2 :
2024-06-09 14:12:42 info: 0.153 0.143 0.133 0.121 0.294 10.608 111.387 363.374 773.228 659.979 1176.178 1175.132 1792.233 1333.879 1747.615 1300.504 1549.555 1123.891 1070.594 540.226 221.72 27.678 0.143 0.118 
2024-06-09 14:12:43 info: baseload voor weekdag 3 :
2024-06-09 14:12:43 info: 0.141 0.142 0.115 0.131 0.341 16.03 729.506 827.393 1690.511 2049.392 1707.63 1167.806 779.6 501.036 1162.576 488.281 485.183 434.752 235.717 30.27 0.131 0.153 
2024-06-09 14:12:43 info: baseload voor weekdag 4 :
2024-06-09 14:12:43 info: 0.171 0.172 0.135 10.372 98.364 361.388 810.133 1281.507 1774.834 2046.558 1587.448 858.316 1083.774 1686.813 1222.234 1357.468 766.317 450.614 184.25 16.045 0.222 
2024-06-09 14:12:44 info: baseload voor weekdag 5 :
2024-06-09 14:12:44 info: 0.146 0.118 0.134 0.579 12.934 93.811 366.729 770.955 1337.536 1070.078 996.554 575.84 204.098 650.819 576.585 477.166 730.262 399.238 211.191 58.947 10.093 0.241 0.126 
2024-06-09 14:12:45 info: baseload voor weekdag 6 :
2024-06-09 14:12:45 info: 0.347 0.21 0.202 0.19 0.162 0.174 18.098 169.554 578.631 371.773 176.981 703.976 665.544 1548.912 1829.089 307.016 86.082 5.944 0.35 0.144 
2024-06-09 14:12:45 debug: Starting new HTTP connection (1): supervisor:80
2024-06-09 14:12:45 debug: http://supervisor:80 "POST /core/api/services/input_datetime/set_datetime HTTP/1.1" 200 2
```

corneel27, 2024-06-09:
when I give the command:
` ../baseload# wc -l *.json
`
I get:
  26 baseload_0.json
  26 baseload_1.json
  26 baseload_2.json
  26 baseload_3.json
  26 baseload_4.json
  26 baseload_5.json
  26 baseload_6.json
 182 total
There should be 24 hour values in each file.
Mine baseloads for monday:
2024-06-08 23:40:08 info: baseload voor weekdag 0 :
2024-06-08 23:40:08 info: 0.232 0.313 0.266 0.387 0.116 0.117 0.112 0.192 0.249 0.406 0.275 0.254 0.32 0.308 0.305 0.226 0.251 0.539 0.188 0.228 0.197 0.186 0.138 0.161 
These are kWh for each hour.
When I see your values they are much to high at some hours more than 1000 kWh. That means: you should have a MW connection to the grid.

My advice: 
- set the baseload calculation temporary to false and fill in for each hour a reasonable value
- then the program can calculate a use-able prediction
- try to set the your consumption- and productionsensors to correct magnitude (how does your energydashboard in HA look?
- your report in the dashboard with \Reports\Balans should have reasonable values. Some like this:
[image]

closed 2024-11-09 by corneel27