corneel27/day-ahead #96: Nieuwe installatie start niet op

georgeboot, 2024-05-26:
Current version: 2024.3.9

Logboek:

```
    self.config = Config(file_name)
                  ^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 9, in __init__
    self.options = json.load(file_json)
                   ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
               ^^^^^^^^^^^^^^^^^^^^^^
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 351 column 9 (char 8195)
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Traceback (most recent call last):
  File "/root/dao/prog/check_db.py", line 146, in <module>
    main()
  File "/root/dao/prog/check_db.py", line 140, in main
    checkdb = CheckDB()
              ^^^^^^^^^
  File "/root/dao/prog/check_db.py", line 9, in __init__
    self.config = Config("../data/options.json")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 9, in __init__
    self.options = json.load(file_json)
                   ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
               ^^^^^^^^^^^^^^^^^^^^^^
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 351 column 9 (char 8195)
Setting up watches.
Watches established.
Failed to read config file: gunicorn_config.py
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/app/base.py", line 111, in get_config_from_filename
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/dao/webserver/gunicorn_config.py", line 5, in <module>
    port = get_config(app_datapath + "options.json", ["dashboard", "port"], 5000)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 36, in get_config
    config = Config(file_name=file_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 9, in __init__
    self.options = json.load(file_json)
                   ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
               ^^^^^^^^^^^^^^^^^^^^^^
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 351 column 9 (char 8195)
Traceback (most recent call last):
  File "/root/dao/prog/day_ahead.py", line 2145, in <module>
    main()
  File "/root/dao/prog/day_ahead.py", line 2094, in main
    day_ah = DayAheadOpt("../data/options.json")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/day_ahead.py", line 36, in __init__
    self.config = Config(file_name)
                  ^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 9, in __init__
    self.options = json.load(file_json)
                   ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 293, in load
    return loads(fp.read(),
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
               ^^^^^^^^^^^^^^^^^^^^^^
json.decoder.JSONDecodeError: Expecting ',' delimiter: line 351 column 9 (char 8195)
```

Ik snap bij HomeAssistantOS nooit hoe ik bij de ruwe storage kan komen, dus het is me niet gelukt de bug zelf te fixen.

georgeboot, 2024-05-26:
De fix voor #94 in de nieuwste release lijkt niet helemaal door te komen.
In de commit history zie ik de wijziging ook niet terug, op master missen de comma's nog,

corneel27, 2024-05-26:
Ik heb de laatste develop-versie (in de branche addon) doorgezet naar main.
Daar zijn de komma-fouten hersteld (net nog gechecked).
Dus nu moet het wel werken.
Als je nog vragen hebt: ik hoor het graag.

closed 2024-05-28 by georgeboot