corneel27/day-ahead #98: PV uitschakelen niet mogelijk

georgeboot, 2024-05-28:
Ik heb een hybride omvormer van Growatt waarop ik de pv niet uit kan schakelen.

Ik kan wel vanaf het net laden (en dit ook desgewenst uitschakelen), maar de productie van de panelen gaat ook naar de accu.
Ik heb het aangekaart bij Growatt en ze gaan er naar kijken, maar voor nu lijkt het dus niet mogelijk.

Hoe vertel ik day-ahead dat dit niet mogelijk is zodat hij daar in de planning rekening mee houdt?

corneel27, 2024-05-28:
Ik zag ook jou oproep op tweakers om de pv-input van de Growatt uit te kunnen zetten.

Ik ben even in de code gedoken.
De - voor mij - makkelijkste manier (en ik denk ook voor jou) is het weglaten of uitcommentarieren van de regel 
'`"entity pv switch": "input_boolean.pv_woning_aan_uit"`
Weglaten spreekt voor zich. Uitcommentarieren  werkt bijvoorbeeld als volgt:
'`"//entity pv switch": "input_boolean.pv_woning_aan_uit"`
Kan dat voor jou werken?
Laat het me weten dan ga ik dat zo maken.
Mogelijk heb je het donderdag al nodig, dan is het een zondag in Duitsland

georgeboot, 2024-05-28:
Ja prima hoor, wat het makkelijkst is.

Wil overigens best helpen om de boel wat om te bouwen (zelf (ook?) softwareontwikkelaar) maar liever eerst even draaien krijgen en wat stoeien voordat ik me er aan waag. Ook totaal niet bekend met HA add-ons enz. Wel met python.

corneel27, 2024-05-28:
Ok, ga ik het zo maken.

Je kunt de software zo kopieren van Github en in een folder plaatsen in je eigen ontwikkelomgeving (ik gebruik PyCharm).
Je kunt met git ook een lokale kloon installeren en wijzigingen voorstellen met een push request.
Je  kunt dan het beste de addon branche klonen, want die gebruik ik als ontwikkelversie.
Als je wilt kan ik je  uitnodigen als "contributor".

georgeboot, 2024-05-31:
Bedankt!

closed 2024-05-31 by georgeboot