corneel27/day-ahead #99: ophalen prijzen sinds update 2024.5.0

sjampeter, 2024-05-28:
het lijkt dat sinds de update, welke gistermiddag geinstalleerd is (denk ik toch), de prijzen in een verkeerde "array" geplaatst worden.
tot 13:00 vanmiddag liep het programma prima door, echter sinds het ophalen nieuwe prijzen gaat het mis.
enstoe, nordpool of easyenergy maakt geen verschil.
ligt het aan mij?

[image]

pvhpsb, 2024-05-28:
Ik zie hetzelfde probleem, gestopt na 13:00 uur

sjampeter, 2024-05-28:
rond 16 uur werkte het weer. geen idee wat er gebeurd is

sjampeter, 2024-05-30:
vandaag met update wederom hetzelfde beeld.

[image]

draai nu op versie 2024.5.2
ik heb niks veranderd aan de instellingen

robbak2, 2024-05-30:
Bij mij is het ook gestopt; ik heb geen wijziging doorgevoerd
[image]

corneel27, 2024-05-30:
Probleem is dat de meteodata niet meer tot en morgen avond binnenkomen, maar geen data meeleveren voor morgen na 19:00 uur .
Nu staat versie 2024.5.3 klaar, daarin rekent hij nu tot morgen 19:00 uur.
Ik ga contact opnemen met meteoserver wat er aan de hand is, want dit gebeurt tevaak.

corneel27, 2024-05-30:
Inmiddels heb ik contact gehad met Meteoserver.
Dit is hun reactie:

`Dank je voor je melding, er was de afgelopen dagen gedoe bij het KNMI dataplatform waardoor Harmonie korter liep. Dat wordt op dit moment ingehaald. Maar zou het niet wat voor je zijn om GFS te gebruiken? Dan kun je tot 4 dagen in de toekomst uurverwachtingen opvragen.`

Bij het ophalen van de data om 15:55 was het nog niet ingehaald.
Ik heb nog even naar het verschil tussen die twee modellen (Harmonie en GFS) gekeken:

- GFS rekent per vierkant van 70x70 km over de hele wereld, maar wel 4 dagen vooruit
- Harmonie rekent per 5 x 5 km alleen in de Benelux en max 41 uur vooruit

Als jullie de keuze willen hebben kan ik er een instelling van maken.

Maar gelukkig rekent DAO nu wel met de nieuwe versie, maar helaas maar tot en met 19:00 uur.

robbak2, 2024-05-30:
Het lijkt me sowieso verstandig om te proberen te voorkomen dat de berekening afkakt bij het ontbreken van wat gegevens. Misschien kan dat door de in de berekening gebruikte lijst/array te initialiseren met waarden of schattingen

corneel27, 2024-05-30:
Wat ik kan doen is: blijven gebruik maken van Harmonie model. Bij het ontbreken van data deze aanvullen met data van het GFS model. 
Lijkt dat wat?

robbak2, 2024-05-30:
Ja, dat klinkt wel goed

corneel27, 2024-05-30:
Als je nu via het Dashboard en het Run-menu de meteoprognoses ophaalt is alles weer bij tot en met morgenavond 24:00 uur.
Intussen ga ik aan de gang met versie 2024.5.4, waarin ontbrekende gegevens van het Hramonie-model worden aangevuld met gegevens van het GF-model.

corneel27, 2024-06-01:
Opgelost met versie 2024.5.4

closed 2024-06-01 by corneel27