corneel27/day-ahead #109: list index out of range

thoutenbos, 2024-07-15:
I keep getting this error, not sure what I am missing. Tried to start with a very basic config only but still no luck. Also it was completely unclear to what file this line nr 416 is referring.

Any clues on that?

```
2024-07-15 10:18:51 info: Doorzetten van alle settings naar HA
2024-07-15 10:18:52 fout: list index out of range
2024-07-15 10:18:52 fout:  <class 'IndexError'>. list index out of range, regelnummer: 416
```

Here is my config

```
{
  "homeassistant": {},
  "database da": {
    "server": "172.24.0.201",
    "database": "day_ahead",
    "username": "hassuser",
    "password": "!secret db_da_password"
  },
  "database ha": {
    "server": "172.24.0.201",
    "database": "haos",
    "username": "hassuser",
    "password": "!secret db_ha_password"
  },
  "meteoserver-key": "!secret meteoserver-key",
  "prices": {
    "source day ahead": "nordpool",
    "entsoe-api-key": "!secret entsoe-api-key",
    "regular high": 0.0431,
    "regular low": 0.0431,
    "switch to low": 23,
    "energy taxes delivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.1088
    },
    "energy taxes redelivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.1088
    },
    "cost supplier delivery": {
      "2022-01-01": 0.002,
      "2023-03-01": 0.018,
      "2024-04-01": 0.0175
    },
    "cost supplier redelivery": {
      "2022-01-01": 0.002,
      "2023-03-01": 0.018,
      "2024-04-01": 0.0175
    },
    "vat": {
      "2022-01-01": 21,
      "2022-07-01": 9,
      "2023-01-01": 21
    },
    "last invoice": "2023-10-23",
    "tax refund": "True"
  },
  "logging level": "info",
  "use_calc_baseload": "False",
  "baseload calc periode": 56,
  "baseload": [
    0.14,
    0.38,
    0.26,
    0.42,
    0.15,
    0.12,
    0.13,
    0.15,
    0.23,
    0.26,
    0.31,
    0.32,
    0.31,
    0.23,
    0.26,
    0.21,
    0.21,
    0.54,
    0.26,
    0.26,
    0.22,
    0.19,
    0.18,
    0.16
  ],
  "graphical backend": "",
  "graphics": {
    "style": "dark_background",
    "show": "true",
    "prices delivery": "True",
    "prices redelivery": "True",
    "average delivery": "True"
  },
  "strategy": "minimize cost",
  "notifications": {
    "notification entity": "",
    "opstarten": "False",
    "berekening": "False",
    "last activity entity": "input_datetime.dao_last_activity"
  },
  "grid": {
    "max_power": 9
  },
  "history": {
    "save days": 7
  },
  "dashboard": {
    "port": 5000
  },
  "boiler": {
    "boiler present": "False"
  },
  "heating": {
    "heater present": "False"
  },
  "battery": [],
  "solar": [
    {
      "name": "Dak",
      "tilt": 12,
      "orientation": 154,
      "capacity": 3.2,
      "yield": 0.01,
      "entity pv switch": "input_boolean.papegaai_pv_on_off"
    }
  ],
  "electric vehicle": [],
  "machines": [],
  "tibber": {
    "api_token": "!secret tibber_api_token"
  },
  "report": {
    "entities grid consumption": [],
    "entities grid production": [],
    "entities solar production ac": [],
    "entities solar production dc": [],
    "entities ev consumption": [],
    "entities battery consumption": [],
    "entities battery production": []
  },
  "scheduler": {
    "1050": "get_tibber_data",
    "1150": "get_tibber_data",
    "1250": "get_tibber_data",
    "1255": "get_day_ahead_prices",
    "1350": "get_tibber_data",
    "1355": "get_day_ahead_prices",
    "1450": "get_tibber_data",
    "1455": "get_day_ahead_prices",
    "1554": "get_day_ahead_prices",
    "1555": "get_meteo_data",
    "1655": "get_day_ahead_prices",
    "2155": "get_meteo_data",
    "2359": "clean_data",
    "0355": "get_meteo_data",
    "0955": "get_meteo_data",
    "0950": "get_tibber_data",
    "xx00": "calc_optimum",
    "0930": "calc_baseloads"
  }
}
```

corneel27, 2024-07-15:
Sorry, my mistake
Thank you for the report of the error.
I didn't test the programma without a battery in the settings.
I found the error en solved it.
This evening I will do some further tests and put a new version on github.
Can you wait so long?
If you can not wait you can add a dummy battery to your settings.

thoutenbos, 2024-07-15:
Aaah thanks! At least I am glad I opened this issue then, as I was trying soo many things haha

Yes tonight is fine, no rush I am just testing now so no workloads connected yet. Eventually I will also configure a battery ofcourse, but wanted to start simple

corneel27, 2024-07-15:
I have published version 2024.5.6, where this issue is adressed.

closed 2024-07-15 by corneel27

thoutenbos, 2024-07-16:
All good now, thanks!