corneel27/day-ahead #118: Install van versie 2024.8.4 geeft alleen fouten

HerikoParaone, 2024-09-02:
table.__init__(name, metadata, *args, _no_init=False, **kw)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py", line 857, in __init__
    self._autoload(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py", line 887, in _autoload
    insp = inspection.inspect(autoload_with)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/inspection.py", line 140, in inspect
    ret = reg(subject)
          ^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py", line 303, in _engine_insp
    return Inspector._construct(Inspector._init_engine, bind)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py", line 236, in _construct
    init(self, bind)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py", line 247, in _init_engine
2024-09-02 15:00:04 fout: File: /root/dao/prog/da_scheduler.py, line 58, in <module>
    engine.connect().close()
    ^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 3278, in connect
    return self._connection_cls(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 148, in __init__
    Connection._handle_dbapi_exception_noconnection(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2442, in _handle_dbapi_exception_noconnection
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 146, in __init__
    self._dbapi_connection = engine.raw_connection()
                             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 3302, in raw_connection
    return self.pool.connect()
           ^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 449, in connect
    return _ConnectionFairy._checkout(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 1263, in _checkout
    fairy = _ConnectionRecord.checkout(pool)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 712, in checkout
    rec = pool._do_get()
          ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py", line 179, in _do_get
    with util.safe_reraise():
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py", line 177, in _do_get
    return self._create_connection()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 390, in _create_connection
    return _ConnectionRecord(self)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 674, in __init__
    self.__connect()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 900, in __connect
    with util.safe_reraise():
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 896, in __connect
    self.dbapi_connection = connection = pool._invoke_creator(self)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/create.py", line 643, in connect
    return dialect.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 621, in connect
    return self.loaded_dbapi.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pymysql/connections.py", line 361, in __init__
    self.connect()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pymysql/connections.py", line 716, in connect
    raise exc
sqlalchemy.exc.OperationalError: (pymysql.err.OperationalError) (2003, "Can't connect to MySQL server on 'core-mariadb' ([Errno -5] No address associated with hostname)")
(Background on this error at: https://sqlalche.me/e/20/e3q8)
2024-09-02 15:00:04 fout: File: /root/dao/prog/da_scheduler.py, line 54, in main
2024-09-02 15:00:04 fout: File: /root/dao/prog/da_scheduler.py, line 42, in scheduler
2024-09-02 15:00:04 fout: File: /root/dao/prog/da_base.py, line 426, in run_task_function
2024-09-02 15:00:04 fout: File: /root/dao/prog/da_base.py, line 375, in calc_optimum
2024-09-02 15:00:04 fout: File: /root/dao/prog/day_ahead.py, line 49, in calc_optimum
2024-09-02 15:00:04 fout: File: /root/dao/prog/db_manager.py, line 197, in get_prognose_data
2024-09-02 15:00:04 fout: File: <string>, line 2, in __new__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/deprecations.py, line 281, in warned
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py, line 427, in __new__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py, line 481, in _new
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py, line 857, in __init__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/schema.py, line 887, in _autoload
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/inspection.py, line 140, in inspect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py, line 303, in _engine_insp
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py, line 236, in _construct
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/reflection.py, line 247, in _init_engine
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py, line 3278, in connect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py, line 148, in __init__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py, line 3302, in raw_connection
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 449, in connect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 1263, in _checkout
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 712, in checkout
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py, line 179, in _do_get
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 390, in _create_connection
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 674, in __init__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py, line 900, in __connect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/create.py, line 643, in connect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py, line 621, in connect
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/pymysql/connections.py, line 361, in __init__
2024-09-02 15:00:04 fout: File: /root/dao/venv/day_ahead/lib/python3.11/site-packages/pymysql/connections.py, line 716, in connect

corneel27, 2024-09-02:
Ik snap dat je in deze brei niet makkellijk de kern van de fout opspoort, maar dit is de kern:
```sqlalchemy.exc.OperationalError: (pymysql.err.OperationalError) (2003, "Can't connect to MySQL server on 'core-mariadb' ([Errno -5] No address associated with hostname)")```
Waarschijnlijk komt de database definitie zoals je die in settings.json hebt staan niet overeen met jouw situatie.
Ik heb daarom de volgende vragen voor je:
- Kun je het eerste deel van je settings.json delen. Het gaat met name om wat je hebt ingevuld bij:
   -  ` "database da": {....}` en  
   -  `  "database ha": {....}" `
- Het programma gaat er nu vanuit (conform je huidige instellingen) dat je de mariadb addon hebt geinstalleerd. Klopt dat?
- Maakt jouw Home Assistant gebruik van mariadb of gebruik je sqlite in HA (dat is in HA de default db-engine)
Ik hoor het graag.

HerikoParaone, 2024-09-03:
Alle respect voor wat je hebt gemaakt.
Als simpele ziel probeer ik te begrijpen wat er gedaan moet worden. Als ik van boven naar beneden lees, kom IK daar niet uit.
Toevallig zie ik iets van entsoe-api-key staan. Ik gok erop,dat ik via de website transparency.entsoe.eu een api-key moet zien te krijgen. Klopt dit?

corneel27, 2024-09-03:
Heb je DOCS.md gelezen?
https://github.com/corneel27/day-ahead/blob/main/dao/DOCS.md
Misschien moet ik een simpele startup-tutorial maken.
Ben je daar mee geholpen?

corneel27, 2024-09-03:
Entsoe kun je gebruiken om de day-ahead prijzen op te halen en dan heb je een key nodiig.
Maar als je de prijzen ophaalt bij nordpool heb je geen key nodig.

HerikoParaone, 2024-09-03:
> Heb je DOCS.md gelezen? https://github.com/corneel27/day-ahead/blob/main/dao/DOCS.md Misschien moet ik een simpele startup-tutorial maken. Ben je daar mee geholpen?

Ja, die heb ik doorgenomen. Ik kom daar (nog niet) snel verder mee.

Met een simpele startup-tutorial zou ik geweldig geholpen worden,

corneel27, 2024-09-04:
Ik ben bezig met een start-up tutorial als een van de eerste hoofdstukken in DOCS.md.
Kost wel een paar dagen tijd.
Graag nog even geduld.
Mocht je intussen concrete vragen hebben dan hoor ik het graag.

HerikoParaone, 2024-09-04:
Geen probleem. Ik ben benieuwd

corneel27, 2024-09-11:
Ik heb zojuist versie 2024.8.6 gepubliceerd met daarin (zie changelog) o.a. een flinke uitbreiding van DOCS.md zodat je aan de hand wordt genomen met het configureren van je eigen instellingen.
Zie met name het hoofdstuk **DAO eerste keer starten**
Het beste kun je blanco beginnen: de oude addon verwijderen en opnieuw starten.
Ik hoor graag als je nog ergens tegenaan loopt.
Succes!

corneel27, 2024-10-04:
Ik heb een nieuwe versie gepubliceerd (2024.10.0) met in DOCS.md een startup tutorial.

closed 2024-11-09 by corneel27