corneel27/day-ahead #121: installatie 2024.8.6 lukt niet

KRline, 2024-09-12:
Fout code:
Failed to install add-on
The command '/bin/bash -o pipefail -c uv pip install -r /tmp/requirements.txt' returned a non-zero code: 2

HA en core enz... op laatste versies gebracht, herstart... steeds zelfde probleem.

enig idee hoe op te lossen?

Kris

corneel27, 2024-09-12:
Vervelend dat het installeren fout gaat.
Ik heb zojuist een blanco installatie van DAO getest op een linux machine, waar ik HA heb draaien in een virtual box:
werkt probleemloos.
Ook op een raspberry PI4 werkt het probleemloos.
Dus mijn vraag aan jou is:
- wat is de hardware waar HA op draait?
  - dao werkt alleen op systemen met arm64, amd63 en i386 processor (dus geen rapsberry pi3)
- wat is de opslag van HA (sd-cards kunnen problemen geven)

Misschien kan ik je verder helpen
Ik hoor graag van je

KRline, 2024-09-12:
Werk op een Intel NUC met proxmox virtualizatie, HA operating system
CPU(s) 4 x Intel(R) Core(TM) i3-4130 CPU @ 3.40GHz (1 Socket)  (2 CPU voor de VM)
RAM: 8GB (5 voor de VM)
HD: 98GB (42 voor de VM)
Kernel Version Linux 6.5.13-3-pve (2024-03-20T10:45Z)
Boot Mode EFI

corneel27, 2024-09-12:
Dat zou allemaal geen problemen mogen geven.
Je bent wel de eerste met proxmox (voor zover ik weet, maar ik weet ook niet alles ;-) ).
Ik kan hier ook niet zo snel een promox installeren (zou kunnen in een virtual box, dus een vm-machine in een vm-machine..)
Dat moet ik eerst maar eens testen.
Ik heb wel nog een hint op internet gevonden omdat mogelijk bij jou een lib-file ontbreekt.
Ik zal de dockerfile daarop aanpassen.
Maar dat wil ik dan weer eerst goed testen voordat ik dat publiceer. Dus dat kost me minimaal een dag.
Kun je daarop wachten?
Ben jij bereid om een test-installatie daarvan te doen ((een extra addon die je daarna weer verwijderdt)?

KRline, 2024-09-12:
Ik heb tijd zat.
Woon op een varend schip, nu in Frankrijk, en pas midden october terug in Belgie waar ik walstroom heb (en eigen digitale teller)
Ook de ENTSO-e API is pas gisteren aangevraagd.
Kan dus enkel testen of de DAO installatie lukt en of ik dan verbinding kan krijgen met ENTSO-e platform voor de prijzen per uur.
Geen probleem om iets te testen, maak wel een image van de proxmox VM daarvoor.

corneel27, 2024-09-12:
Ik heb in de addon-branche (dat is de develop branche) een update geplaatst, die mogelijk jouw probleem oplost (ik heb het nog niet op promox kunnen testen).
Zou je deze willen testen?
Dat gaat als volgt:
- je gaat in HA naar Instellingen/Add-ons
- je kiest voor de add-on winkel
- je kiest in het puntjes-menu rechtsboven voor repositories
- alles net als bij het toevoegen van de standaard addon
- maar nu vul je in `https://github.com/corneel27/day-ahead#addon` (je zet er dus `#addon` achter)
- klik op `Toevoegen`
- verder weer hetzelfde als bij de standaard addon

Laat je me weten of dit werkt?
Ik hoor het graag .

KRline, 2024-09-13:
Dag Corneel,
repository toegevoegd, oude repository gewist, HA herstart, DAO installatie... zelfde probleem
The command '/bin/bash -o pipefail -c uv pip install -r /tmp/requirements.txt' returned a non-zero code: 2

Ik kijk even of het mogelijk is om met de CLI  in HAOS (of andere command line) manueel iets te installeren. Heb dit nog nooit geprobeerd en weet ook niet of het mogelijk is om iets zoals 'pip install xyzlibrary" manueel uit te voeren. Of en volledig manuele installatie van de add-on op een of andere manier eventueel...

KRline, 2024-09-13:
Kan het zijn dat het is omdat GCC compiler niet beschikbaar is in HAOS (HomeAssistant Operating System)?
Ik heb geprobeerd alle dependencies in je file "requirements.txt" manueel te installeren via pip install. Lukte voor bijna alles, behalve vor "mip". Daar krijg ik een foutmelding bij het installeren van cffi:    " warning: build_py: byte-compiling is disabled, skipping."  en iets verder  "error: command 'gcc' failed: No such file or directory"

KRline, 2024-09-13:
GCC toegevoegd via "apk add gcc"
mip 1.16rc0 installatie via pip install werkte nu zonder foutmelding
terug geprobeerd DAO te installeren -, zelfde foutmelding   .... non-zero code: 2

corneel27, 2024-09-13:
Dank voor al je uitzoekwerk!
Ik ben bezig om een docker-file te maken waarmee je exact kon doen wat jij nu gedaan hebt maar dan in de addon!
gcc moet niet in HAOS maar in de addon beschikbaar zijn (dat is een afgeschermde sandbox)
Maar in de laatste dockerfile heb ik gcc al toegevoegd zie regel 15:
`RUN apt-get install -y postgresql-client sqlite3 libpq-dev gcc build-essential`
Omdat het een debian-image is werkt het niet met apk (dat is alpine linux), maar met apt-get.
Maar tot nu toe werkt mip met een amd64 processor probleemloos onder debian: de benodigde binary wordt automatisch geinstalleerd.
Dus ik ga nog even verder met het maken van een dockerfile zodat jij op jouw machine kunt testen waar het fout gaat.

corneel27, 2024-09-13:
Ik heb een nieuwe versie in de addon-branche gezet.
Maar om de testen moet je deze niet via de standaard-weg installeren maar als lokale addon.
Dat werkt als volgt:
- ga naar deze pagina op github: `https://github.com/corneel27/day-ahead/tree/addon`
- klik op de groene knop met "code"en download de zip-file
- daarin vind je een map dao
- mbv van samba kun je op je ha-machine bij de map "addons"
- kopieer de hele map "dao" (met alle bestanden en submappen) naar addons, dus je hebt dan `.../addons/dao/..`
- in de map dao staat een bestand "Dockerfile"
- kopieer /overschrijf vanaf `COPY requirements.txt /tmp/` alle regels daarondermet onderstaande regels

```
COPY requirements.txt /tmp/
COPY run /root/dao/prog/
WORKDIR /root/dao/prog
RUN chmod a+x run_test.sh
ENTRYPOINT  ["/bin/bash", "./run_test.sh"]
```
Nu ga je in HA naar Settings/Add-ons, add-on-winkel
Klik op "Controleer op add-ons" en dan moet ie bij "Local add-ons" verschijnen.
Klik erop en installeer en na enkele minuten "starten".
Als alles goed is gegaan, ga ja naar je HA-terminal en type:
`docker exec -it addon_local_day_ahead_opt bash`
Je krijgt nu een commandline **in** de addon.
type daarin `pip3 install -r /tmp/requirements.txt`
En dan moet duidelijk worden wat er bij jou fout gaat.
Ik ben benieuwd.
Groeten vann Cees

KRline, 2024-09-14:
Dag Cees,
Alles gaat goed tot en met het starten van de add-on en het starten van de commandline in de addon.
Het commando pip3 install... geeft terug fouten:
➜  ~ pip3 install -r /tmp/requirements.txt
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '/tmp/requirements.txt'

Met file explorer  in de /tmp directory gaan kijken, geen file 'requirements.txt' te zien.
File manueel in de /tmp folder gezet, zonder resultaat.

Er was wel een melding bij het starten van docker in de SSH terminal dat de terminal in protection mode stond. Protection mode in SSH terminal uitgeschakeld en herstart. Terug Docker... en pip3 install... alles verloopt nu zonder fouten (de requirements.txt file stond wel nog manueel in de /tmp directory).
Bliijkbaar dus ergens iets met rechten op files en directories... Raar dat andere addons er geen problemen mee hebben

Alles verwijderd, uninstall addon, ook de files in de locale faddon en /tmp older
herstarten ha
Nazien of ssh terminal NIET in protection mode staat
Terug originele DAO repository install getest, lukt niet.
Download main van github, dao folder lokaal geplaatst, lijn 15 van docker aangepast met je info voor gcc.
Installatie lokaal... terug zelfde error  code.

Vermoedelijk dus ergens iets met rechten of zo op de /tmp directory

KRline, 2024-09-14:
Wat verder getest met lokale installatie.
Je originele DAO van github lokaal geplaatst

Als ik in dockerfile 1 lijn aanpas, dan zijn er geen problemen bij installatie:

RUN pip3 install -r /tmp/requirements.txt
in plaats van
RUN uv pip install -r /tmp/requirements.txt

Maar bij het openen van de web_UI krijg ik dan wel een bad gateway 502 error en kan dus niets verder doen

Heb nog eens geprobeerd met 
RUN uv pip3 install -r /tmp/requirements.txt
maar instalaltie mislukt (originele error melding)

Dan geprobeerd
RUN pip install -r /tmp/requirements.txt
Installatie loopt schijnbaar ok en kan addon starten maar Web UI terug 502 bad gateway

Lijkt me iets te zijn met die uv (vervanging van pip?).

KRline, 2024-09-14:
Hallo,

laatste test:
- Uninstall DAO
- Verwijder folder /addon/dao
- download laatste versie 2024.8.6 van github, unzip,
- upload folder dao naar /addons/
- aanpassen lijn in Dockerfile (verwijderen uv)     RUN pip install -r /tmp/requirements.txt
- herstarten HA
- installatie DAO vanuit locale repository  /addons/dao
- installatie zonder foutmeldingen
- start DAO lijkt OK  (groene punt rechts boven, onderaan opties stop, restart, open web ui, uninstall, rebuild

LOG gegevens:

=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data doesn't exist, made
ln: failed to create symbolic link 'app/static/data': File exists
Setting up watches.
Watches established.
Failed to read config file: gunicorn_config.py
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/app/base.py", line 111, in get_config_from_filename
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/dao/webserver/gunicorn_config.py", line 5, in <module>
    port = get_config(app_datapath + "options.json", ["dashboard", "port"], 5000)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 38, in get_config
    config = Config(file_name=file_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 8, in __init__
    file_json = open(file_name)
                ^^^^^^^^^^^^^^^
NotADirectoryError: [Errno 20] Not a directory: 'app/static/data/options.json'

En ook een foutmelding bij openen Web UI:   502: bad gateway
met daarbij de pop-up melding:  The add-on seems to not be ready, it might still be starting. Do you want to try again?

Verder  geraak ik niet

corneel27, 2024-09-14:
Ik ben vanaf vandaag tot en met a.s. donderdag niet thuis (een paar dagen op de wadden..) dus ik kan  niet mee testen op mijn HA machine.

Als je de Dockerfile aanpast moet de regel **`COPY requirements.txt /tmp/`** er altijd instaan want daarmee worden de requirements.txt naar de /tmp/ directory gekopieerd.
**uv** is een pythton module die sneller werkt dan pip/pip3, maar die heb je niet perse nodig.
Maar als je hem gebruikt moet uv eerst worden geinstalleerd met **`RUN pip3 install uv`**.
Als je uv gebruikt is het commando (in Dockerfile voorafgegaan door `RUN` (in commandline zonder `RUN`):
**`RUN uv pip install -r /tmp/requirements.txt`** (dus niet pip3, maar pip)
Als je webui wilt openen zal ergens in Dockerfile de regel **`EXPOSE 5000`** moeten staan anders blijft alles dicht.

Ik denk dat je met je laatste poging dicht bij de oplossing bent.
Alleen bij het verwijderen van de addon voor die laatste poging heb je waarschijnlijk niet alles verwijderd.
Er is een map "dao\dao_data` in de folder samba-share addon_configs, waarin na een installatie een aantal  options-bestanden staan en die worden niet verwijderd. Dat is een bewuste keuze om zo de instellingen mee te nemen naar een volgende update.
Ik vermoed dat daar iets fout is gegaan.
Je kunt het beste de map `dao_data` verwijderen en dan je laatste installatie herhalen met de volgende aanpassingen in je Dockefile:
- uv niet installeren
- **`RUN pip3 install -r /tmp/requirements.txt`** (dus zonder uv)

Ik ben benieuwd.
Groeten van Cees

KRline, 2024-09-15:
Dag Cees
alvast prettige dagen op de Wadden.
Volgende geprobeerd:
- Wissen /addon/dao folder
- Wissen /addon_configs/local_day_ahead_opt folder
- download versie van Github en plaatsen dao folder in /addon
- Aanpassen Dockerfile lijnen 57 en 58
   57     # RUN pip3 install uv
   58    RUN pip3 install -r /tmp/requirements.txt
- Herstarten HA (voor alle zekerheid)
- Installatie van DAO uit locale repository.  Installatie lijkt OK.   
- Uitschakelen auto start DAO
- herstarten HA
- Logfile DAO:  veel fouten
- starten DAO, lijkt OK
- Logfile: DAO, nog meer fouten
- Open Web UI: 502: bad gateway en pop-up "The add-on seems to not be ready, it might still be starting. Do you want to try again?"

Via file browser in de /addon/dao en in de /addon_configs/local_day_ahead_opt folders even wat JSON files bekeken. Databases en paswoorden manueel ingevuld, zonder resultaat.
Ik merk dat er ook IP addressen instaan die helemaal niet van mijn in stallatie zijn, en ook poort adressen zoals 8123.   Nu heb ik de poort van HA wel veranderd naar 443 om zonder problemen via https:// toegang te krijgen tot mijn Home Assistant in het lokaal netwerk. Kan dit misschien het probleem zijn dat de webserver van DAO niet werkt?

corneel27, 2024-09-20:
Ik ben weer thuis.
Ik ben wel benieuwd wat er in de logfiles met fouten staat. 
Dat geeft mij waarschijnlijk inzicht wat/waar het fout gaat.
Kun je die met me delen (misschien beter als bijlage).
Ik heb me verder ook verdiept in het https-gebeuren.
Jij hebt waarschijnlijk certificaat-bestanden staan in de ssl-map.
Ik zal een extra instelling maken en de software aanpassen zodat het met certificaten in de ssl-map https-versleutelde webpaginas kan genereren.
Maar dat kost me wel enkele dagen.

KRline, 2024-09-21:
Hallo,
gezien HTTPS staan er inderdaad certificaten in de ssl map.
Installatie via Github lukt dus niet. Map lokaal gezet en dan installatie zonder UV lukt wel maar verder geen Web UI (error 502: bad gateway)
In bijlage ook errorlog.
[errorlog DAO.txt](https://github.com/user-attachments/files/17082992/errorlog.DAO.txt)
Neem je tijd, ik ben momenteel tot 30 sept zeer druk bezig met klanten op cruise en geen/weinig tijd om HA te testen.

KRline, 2024-10-17:
Installatie  van versie 2024.10.4 is gisteren dan toch gelukt.
Weet niet of dit aan de update van DAO ligt, of aan het feit dat mijn (mobiele) internetverbinding nu veel sneller is: vroeger begrensd op 500KB/sec en nu gaat het tot 8MB/sec. (QoS filters aangepast voor dit toestel).
Nu nog configureren en mogelijkheden uitzoeken...

corneel27, 2024-10-17:
Er zijn de laatste maand veel issues opgelost. Ook staat er in DOCS.md (misschien ook iets voor jou?) een tutorial hoe je stapsgewijs je configuratie opbouwt en uitbouwt.
Als je nog vragen hebt hoor ik het graag.
Succes!

closed 2024-10-17 by corneel27