corneel27/day-ahead #123: start en end window

sjampeter, 2024-09-20:
zou het kunnen dat er met een nieuwe update, er iets veranderd is met betrekking tot start en end window voor "machines"?
ik stuur mijn warmtepomp via node-red aan om op het juiste moment te gaan verwarmen.

mijn start en end window staan op 08:00 tot 20:00. 
dat wil zeggen dat de warmtepomp in dit window een run kan doen.

het vreemde is, en dit was volgens mij vroeger niet zo, dat rond 13 uur nieuwe prijzen binnen komen. zodra dit gebeurt wordt er al voor de volgende dag (dus volgend window) een optimum gecalculeerd.
mijn besturing raakt hier behoorlijk van in de war.

volgens mij was het bij eerdere versies zo dat  er pas een nieuw tijdstip berekend werd na (in mijn geval) 20:00.

kan iemand dit bevestigen?

corneel27, 2024-09-21:
Er is niets meer aan de software op dit onderdeel veranderd sinds de introductie.
Maar er kan best nog een foutje in de software zitten.
Ik wil graag het probleem reproduceren en zien wat er fout gaat.
Daarvoor heb ik nodig:
- Jouw definitie van de warmtepomp in de settings. Je mag de hele options.json als bijlage meesturen als je de secrets er niet in hebt staan.
- een screenshot van het scherm in HA waarmee je de (het?)  window definieert.
Wil je dat opsturen?

sjampeter, 2024-09-21:
Dag Corneel.
uiteraard wil ik dit sturen, geen enkel probleem
tevens enkele screenshots van voorbeeld situatie van gisteren.

[image]
situatie om 08:00: wp aan berekent op 12:45

[image]
situatie om 09:00, wp aan is verdwenen. waarde staat echter vast in "calculated_start_warmtepomp"

[image]
situatie om 13:20, het eerste moment dat er nieuwe prijzen binnen zijn. te zien is dat er voor de volgende dag al een tijdstip berekent wordt voor de wp. deze wordt ook al doorgezet naar "calculated_start_warmtepomp" . dit doorzetten mag volgens mij enkel gebeuren na "end_window_warmtepomp" . doordat het al eerder doorgezet wordt, raakt de verder aansturing (via node-red) behoorlijk in de war. 

[image]
zie hier de import node  "calculated_start_warmtepomp" van node-red. hier wordt heel de toko op aangestuurd. deze is van vandaag, net voor 13:00 uur.
[image]
deze van 13:22. 
ik dacht dat deze tijd pas na end window warmtepomp doorgegeven werd, echter is dit dus niet zo.

ik kan het trouwens, nu ik het probleem wat beter in beeld heb, oplossen via node-red.
vraag blijft of dit gewenst gedrag is?

[image]

[image]
dit zijn de entiteiten.

het vreemde is dat het spul probleemloos gedraaid heeft.

corneel27, 2024-09-22:
Dank voor de heldere beschrijving van het issue.
Ik heb ernaar gekeken en het is een fout die er al vanaf het begin in zit.
Ik ga een en ander aanpassen en als het klaar is komt er een nieuwe versie van de de software.

sjampeter, 2024-09-22:
kijk aan. 
blij dat ik heb kunnen bijdragen aan deze software.
waarom het hiervoor probleemloos draaide, heb ik nog geen idee van.
in ieder geval zal straks het probleem vanuit de software niet meer aanwezig zijn.
ik duik nog even in mijn eigen aansturing van de wp.
bedankt.

corneel27, 2024-09-22:
Ik heb op de develop-branche (de naam van die branche is `addon`) een aanpassing gezet, die de fout zou moet oplossen.
Ik heb het hier lokaal getest en daar werkt ie probleemloos.
Zou jij deze develop-versie willen testen.
Daarvoor ga je - naast de productie versie - een tweede versie installeren.
Dat werkt als volgt:
- je gaat in HA naar Instellingen/Add-ons
- je kiest voor de add-on winkel
- je kiest in het puntjes-menu rechtsboven voor repositories
- alles net als bij het toevoegen van de standaard addon
- maar nu vul je in `https://github.com/corneel27/day-ahead#addon` (je zet er dus `#addon` achter)
- klik op `Toevoegen`
- verder weer hetzelfde als bij de standaard addon
- als ie draait moet je de instellingen (options.json en secrets.json) kopieren naar de nieuwe addon.
- daarna de addon herstarten

Laat je me weten of dit werkt?
Ik hoor het graag .

Let op: als de tweede (develop) addon goed werkt kun je het beste de eerste addon (productie) tijdelijk even stopzetten, anders gaan die twee elkaar in de weg zitten. 
Als het allemaal goed werkt zet ik alles door naar productie en kun je de test/develop versie weer verwijderen.

corneel27, 2024-09-24:
Er zit nog een klein foutje in: de in een vorige berekening ingeplande inzet van een apparaat (bij jou je wp/sww) die niet meer kan worden gewijzigd (omdat het begintijdstip van de berekening voorbij het begintijdstip van de inzet ligt, wordt niet meegenomen in de rapportages grafieken en tabellen.
Dat ben ik nu aan het herstellen. Wellicht kan ik dit samen met jouw opmerkingen/suggesties meenemen in een volgende versie.
Groeten van Cees

sjampeter, 2024-09-24:
hmm. dat deed ie in de vorige versies toch ook al niet? opzich geen drama, maar voor de volledigheid wel zo mooi inderdaad

sjampeter, 2024-09-25:
mijn end_window_warmtepomp staat op 19:59. toch wordt om exaxt 18:00 de nieuwe startwaarde doorgegeven. lijkt dus nog niet helemaal te kloppen.
[image]
om 17:00 nog geen calculatie voor start warmtepomp.
[image]
om 18:00 dus wel.
[image]
hier de instelling van het end_window.
waar die 18:00 vandaan komt, geen idee.

[image]
oude calculated_start blijft staan tot 13:00 (dan wordt de nieuwe berekening voor de volgende 24 uur gestart)
[image]
om 13:00 is ie weg.

corneel27, 2024-09-25:
Dank voor je testen en feedback!
Ik heb versie 2024.8.7.dev_b in de addon-branche gepubliceerd.
Hierin zijn een aantal correcties doorgevoerd.
Ook bovenstaande fouten zouden weg moeten zijn.
Of ik die laatste fout heb iopgelost heb ik nog mijn twijfels, wat is de begintijd van je window dan kan ik het beter testen.
Je kunt (als je de addon-versie nog heb staan) deze gewoon updaten.

corneel27, 2024-09-26:
Dus de functionaliteit kwa schakel-berekening komt nu overeen met jouw wensen of mag hij pas voor de volgende dag berekenen als de eindtijd van het window voor de huidige dag is verstreken?

sjampeter, 2024-09-28:
feitelijk zou je zeggen dat ie pas een nieuwe berekening mag pushen nadat het end_window_warmtepomp voorbij is.
volgens mij zou deze voor de exacte berekening wel al meegenomen moeten worden in de nieuwe berekening op het moment dat de nieuwe prijzen bekend zijn. in mijn geval zou dat rond 13:00 uur tot 13:20 zijn, echter mag ie pas "gepushed" worden na end_window_warmtepomp.
am i right?

nu doet ie dit als het calculated_end_warmtepomp gepaseerd is, wat opzich bij mijn verdere aansturing geen problemen geeft.

voor mij is het prima werkbaar nu, echter denk ik dat we er in theorie nog net niet zijn.

corneel27, 2024-09-29:
Ik ben het helemaal met je eens.
Ik heb versie 2024.8.7.dev_c gepushed naar de addon branche.
Daarin zijn deze aanpassingen doorgevoerd (zie changelog).
Wil jij hem nog even testen?
Bij akkoord zal ik een nieuwe versie (wordt waarschijnlijk 2024.10.0) doorzetten naar de main-branche.

sjampeter, 2024-09-30:
ik zet hem erop. ik laat mijn bevindingen weten.

sjampeter, 2024-10-01:
nog een kleinigheidje tussendoor, maar ik krijg nog wat bijzondere grafieken te zien. 
volgens mij is dit gekomen sinds er wat gesleuteld is aan de weerdata?
onderstaand de instellingen : 
[image]
onderstaand het resultaat.
[image]

doet het script eigenlijk steeds op deze tijdstippen.  foutje in het script? of ligt het bij mij (meteo data komt wel binnen namelijk)

corneel27, 2024-10-01:
Ik heb deze fout ook gehad.
Het probleem zit bij Meteoserver.
Ik heb een paar emails naar hun gestuurd, maar geen reactie.
Ik heb het eindelijk opgelost door het tijdstip van het ophalen van de meteodata te verschuiven:
```
    "0430": "get_meteo_data",
    "1030": "get_meteo_data",
    "1630": "get_meteo_data",
    "2230": "get_meteo_data",
```
Ik zal dit bij de volgende grote update meenemen als extra mededeling in de changelog.

sjampeter, 2024-10-03:
ik heb dezelfde "get_meteo_data" tijdstippen ingesteld, echter veranderd er niks. voor mij geen probleem hoor.

versie c draait stabiel bij mij.
wel blijft hij consequent om 18:00 de nieuwe calculated_start_warmtepomp door geven.
[image]

dit is nog steeds gebaseerd op de calculated_stop_warmtepomp lijkt het.
[image]

zou feitelijk pas na het end_window_warmtepomp mogen gebeuren.
ik dacht dat je dit getracht had te bereiken in versie c?

voor mij niet al te groot probleem, en als het voor jou ook goed is, kun je hem wat mij betreft doordrukken naar de main.

corneel27, 2024-10-04:
Dank voor de terugkoppeling.
> zou feitelijk pas na het end_window_warmtepomp mogen gebeuren.
> ik dacht dat je dit getracht had te bereiken in versie c?
Er zit nu iemand te wachten op een update die niet een "showstopper" repareert.
Dus vandaag komt er een nieuwe main-versie waarin jouw melding nog niet is gerepareerd.
Ik ga jou melding in een volgende test-versie oplossen, dus stay tuned op de "addon" branche.

sjampeter, 2025-04-17:
vandaag toch weer een koude boiler.
wat er volgens mij gebeurd is het volgende.

onderstaande was de gecalculeerde situatie start warmtepomp op basis van calculatie vorige dag (bij ophalen prijzen)
de wp zou dus om 13:45 starten.
deze waarde wordt ook netjes doorgestuurd naar de helper.
[image]

echter wordt er de volgende dag weer een nieuwe calculatie gemaakt.
zie onderstaande screenshot.
deze calculeerd om 13:06 dat de starttijd voor de volgende dag (toevallig) ook 13:45 moet zijn..
deze data wordt weggeschreven naar de helper et voila. de eerdere geplande run wordt niet gedaan.
[image]

de nieuwe calculatie voor de volgende dag mag mijns inziens pas ingeschoten worden na de gecalculeerde 1e starttijd.

makes sense?

corneel27, 2025-04-17:
Mooi dat je dit ontdekt hebt, dit is echt een wat je noemt een "worse case" scenario.
Voor vaatwassers en wasmachines zou dit zo nooit gebeuren, maar een bij een boiler zeker.

Volg je inmiddels ook het topic over DAO op tweakers?
https://gathering.tweakers.net/forum/list_messages/2293338

Daar kwam een dag of wat terug een vraag voorbij over het toepassen van een "machine" voor een boiler.
Misschien dat de automation die ik daarvoor heb gesuggereerd ook voor de korte termijn voor jou kan werken:
https://gathering.tweakers.net/forum/list_message/82199448#82199448

Ik hoor het graag.

sjampeter, 2025-04-18:
de draad op tweakers ken ik zeker wel. 
heb nog wel een paar ideeën maar wacht even tot het wat rustiger is.
betreft overigens rekenen met meerdere contracten.
ik heb een sap, dus werkelijkheid is vaak net wat anders bij mij.

de voorgestelde oplossing is niet hetgeen wat werkt voor mij
het betreft een panasonic warmptepomp welke volledig door node-red (en heishamon) aangestuurd wordt.
het enige wat ik nodig heb is een "go" om het buffervat weer te verwarmen op het goedkoopste moment. (en dus heel het script in node-red te starten)

feitelijk is het nog steeds hetzelfde probleem zoals vermeld in de eerste post van deze draad.
bijzonder dat dit in de tussentijd niet meer voorgekomen is.
ik vermoed dat de beste starttijd steeds rond 13:00 gelegen heeft.
erover terugdenkend, de momenten dat het toch mis gegaan is ( en ik heb hier eigenlijk niet verder naar gekeken wat de oorzaak was) zou weleens hetzelfde issue kunnen geweest zijn.
het gebeurt niet vaak, maar als het gebeurt...... koude doushe.

by the way, startwindow staat op 22:00 en endwindow op 19:00

sjampeter, 2025-04-18:
volgens mij moet het zo zijn : 

calculated-start mag pas doorgegeven worden na end-window. calculatie moet uiteraard wel al rond 13:00 gedaan worden.

toch?

sjampeter, 2025-04-27:
is dit nog iets wat meerdere gebruikers ervaren?
krijg je dit goed in een nieuwe versie?
indien dit zo blijft, zal ik aan de achterzijde iets moeten bouwen wat dit tegen gaat. (in node red)

corneel27, 2025-04-27:
Ik heb in de addon-branche versie 2025.4.3.rc3 gezet, die bovenstaand issue zou moeten repareren.
Wil jij dit testen?

sjampeter, 2025-04-28:
ja hoor, ga ik doen. keep you posted.

sjampeter, 2025-05-02:
eerste bevindingen : 

vanaf het eerste moment dat calculated start bereikt is, wordt er elk uur een nieuw calculated start geschreven.
zie screenshot.

[image]

ook goed te zien in onderstaande grafieken. eerst 1 orgineel vorige dag berekent. berekende start wp 12:45

[image]

om 13:00 het volgende beeld. 
te zien is dat er opeens een kwartier bijgekomen is.

[image]

dan worden er nieuwe prijzen binnengehaald, en volgt om 13:20 het volgende beeld

[image]

vervolgens wordt er elk uur, tot (volgens mij steeds ) 17:00 de calculated start opgeschoven.
overigens wordt er gelukkig geen nieuwe berekening gemaakt voor de voor de volgende dag.

[image]

[image]

[image]

en zie hieronder die van 17:00, hier wordt voor de volgende dag de nieuwe calculated start weggeschreven en "verdwijnt" die van de huidige dag.

[image]

om 18:00 is alles ok.

[image]

momenteel lijkt het dus werkbaar, echter werkt het script volgens mij nog niet helemaal goed.

any thoughts?

sjampeter, 2025-05-19:
ondertussen via tweakers.net opgelost. mag dicht!

closed 2025-05-19 by sjampeter