corneel27/day-ahead #127: Stilte uren met gereduceerde max load

thoutenbos, 2024-10-07:
Is het mogelijk om functionaliteit toe te voegen waarbij je voor bepaalde uren dmv configuratie een maximaal vermogen aan kunt geven voor had accu laden/ontladen? Op vol vermogen maakt de omvormer best wat geluid in een verder geheel stil huis (nacht) en een tijdelijke reductie van vermogen tijdens de nachturen zou welkom zijn. Met een reductie tot 0 wordt het opladen/ontladen niet toegepast in dat uur.

Bijvoorbeeld:
```
"quiethours": {
  "2200": "2000",
  "2300": "1000",
  "0000": "0",
 ...
}
```

corneel27, 2024-10-07:
Ik heb inmiddels even naar de code gekeken en dat gaat wel lukken.
Ik weet niet of je meer thuisbatterijen hebt, maar ik ga het zo implementeren dat je het per batterij moet opgeven.
Verder:
- ik vind "reduced hours" een betere benaming dan "quiethours" (dat is een kwestie van smaak)
- Je geeft het per klokuur ("0".."23") op
- Uren die niet voorkomen hebben geen reductie
- Als de sleutel (key) "reduced hours" niet is gedefinieerd is er geen reductie (dus geen "breaking change"). 

Akkoord?

thoutenbos, 2024-10-07:
Ja dat zou top zijn!!! :-)

corneel27, 2024-10-10:
Ik heb versie 2024.10.2 op github gezet.
Daarin zit de "optional feature" met "reduced hours", zie DOCS.md

closed 2024-10-10 by corneel27

thoutenbos, 2024-10-10:
Bedankt!

thoutenbos, 2024-10-10:
Meteen even getest en lijkt perfect te werken (in planning)
Kan ik weer rustig slapen :-)