corneel27/day-ahead #142: 'Pandas' object has no attribute 'price'

thoutenbos, 2024-11-04:
Hi Cees,

Helaas werkt de reports pagina bij mij niet (meer)
Ik krijg dan een "Internal Server Error"
In de docker logs van de dao container is de volgende stacktrace zichtbaar

Enig idee?

Gr,
Thijs

```
[2024-11-04 20:32:30,741] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/webserver/app/routes.py", line 122, in menu
    return reports()
           ^^^^^^^^^
  File "/root/dao/webserver/../../dao/webserver/app/routes.py", line 288, in reports
    filtered_df = report.calc_grid_columns(report_df, active_interval, active_view)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_report.py", line 1087, in calc_grid_columns
    col_5 = (row.production * (row.price + taxes_t + ol_t)) * (1 + btw / 100)
                               ^^^^^^^^^
AttributeError: 'Pandas' object has no attribute 'price'
```

corneel27, 2024-11-05:
Het lijkt erop dat er geen prijzen zijn in jouw database.
Worden deze nog wel dagelijks opgehaald?
Kun je nog wel een optimalisatieberekening uitvoeren (of heb je dat uitgezet)?
Kijk daarvoor in de logging van de bewerkingen van de addon:
- Ga naar het dashboard/webgui van de dao
- Klik op Home/ en dan op het symbool voor de "tabel"
- [image]
- Je kunt nu bladeren met de pijltjes toetsen naar de logging van rond 13:00 uur en dan zie je of de prijzen zijn opgehaald 
- [image]

thoutenbos, 2024-11-05:
Prijzen ophalen gaat prima. Ook de optimalisatieberekening loopt elk uur zonder problemen en deze dagen staat de boel flink te stampen door de grote prijsfluctuaties, dus dat werkt uitstekend! :-)

Misschien mis ik alleen van een aantal uren de prijzen?
Het ophalen is wel eens een dag mis gegaan namelijk en ik heb wel eens moeten switchen tussen entso-e en nordpool

[image]

thoutenbos, 2024-11-05:
Ik zit ook even door de code te kijken maar de default is volgens mij alleen "vandaag" en zelfs daar gaat het al fout. Terwijl die prijzen echt aanwezig zijn... 

Heb je nog een tip om te troubleshooten? Ik kan lokaal wel tijdelijk wat debug code toevoegen eventueel

```
MariaDB [day_ahead]> select * from `values` where `variabel` = 3 and time >= 1730761200;
+-------+----------+------------+---------+
| id    | variabel | time       | value   |
+-------+----------+------------+---------+
| 11459 |        3 | 1730761200 | 0.08363 |
| 11460 |        3 | 1730764800 |   0.094 |
| 11461 |        3 | 1730768400 | 0.09068 |
| 11462 |        3 | 1730772000 |  0.0913 |
| 11463 |        3 | 1730775600 |   0.094 |
| 11464 |        3 | 1730779200 | 0.09609 |
| 11465 |        3 | 1730782800 |   0.106 |
| 11466 |        3 | 1730786400 | 0.13599 |
| 11467 |        3 | 1730790000 | 0.13621 |
| 11468 |        3 | 1730793600 | 0.11823 |
| 11469 |        3 | 1730797200 | 0.10587 |
| 11470 |        3 | 1730800800 | 0.09528 |
| 11471 |        3 | 1730804400 | 0.09492 |
| 11472 |        3 | 1730808000 | 0.09925 |
| 11473 |        3 | 1730811600 | 0.10798 |
| 11474 |        3 | 1730815200 | 0.14986 |
| 11475 |        3 | 1730818800 | 0.30324 |
| 11476 |        3 | 1730822400 | 0.47299 |
| 11477 |        3 | 1730826000 | 0.43102 |
| 11478 |        3 | 1730829600 | 0.32033 |
| 11479 |        3 | 1730833200 |  0.1697 |
| 11480 |        3 | 1730836800 |  0.1299 |
| 11481 |        3 | 1730840400 | 0.11777 |
| 11482 |        3 | 1730844000 | 0.11003 |
| 11675 |        3 | 1730847600 |  0.1005 |
| 11676 |        3 | 1730851200 | 0.09905 |
| 11677 |        3 | 1730854800 |  0.0954 |
| 11678 |        3 | 1730858400 | 0.09583 |
| 11679 |        3 | 1730862000 | 0.09883 |
| 11680 |        3 | 1730865600 | 0.09291 |
| 11681 |        3 | 1730869200 |   0.121 |
| 11682 |        3 | 1730872800 | 0.21578 |
| 11683 |        3 | 1730876400 |   0.194 |
| 11684 |        3 | 1730880000 | 0.14197 |
| 11685 |        3 | 1730883600 | 0.13092 |
| 11686 |        3 | 1730887200 | 0.12212 |
| 11687 |        3 | 1730890800 | 0.12147 |
| 11688 |        3 | 1730894400 | 0.12411 |
| 11689 |        3 | 1730898000 | 0.13688 |
| 11690 |        3 | 1730901600 |  0.1907 |
| 11691 |        3 | 1730905200 | 0.31569 |
| 11692 |        3 | 1730908800 |    0.55 |
| 11693 |        3 | 1730912400 | 0.54585 |
| 11694 |        3 | 1730916000 |  0.3335 |
| 11695 |        3 | 1730919600 | 0.17562 |
| 11696 |        3 | 1730923200 | 0.12897 |
| 11697 |        3 | 1730926800 |   0.113 |
| 11698 |        3 | 1730930400 |  0.1087 |
+-------+----------+------------+---------+
48 rows in set (0.002 sec)
```

thoutenbos, 2024-11-05:
Ik heb ook nog even een dump gemaakt van de `report_df` value die de methode `calc_grid_columns` meekrijgt en ik zie daar inderdaad geen prices column

```
          uur               vanaf                 tot  consumption  production      cost    profit datasoort  netto_consumption  netto_const
00:00   00:00 2024-11-05 00:00:00 2024-11-05 00:00:00        2.749         NaN  0.708267       NaN  recorded                NaN          NaN
01:00   01:00 2024-11-05 01:00:00 2024-11-05 01:00:00        2.092         NaN  0.565244       NaN  recorded                NaN          NaN
02:00   02:00 2024-11-05 02:00:00 2024-11-05 02:00:00        2.635         NaN  0.701373       NaN  recorded                NaN          NaN
03:00   03:00 2024-11-05 03:00:00 2024-11-05 03:00:00        2.637         NaN  0.703884       NaN  recorded                NaN          NaN
04:00   04:00 2024-11-05 04:00:00 2024-11-05 04:00:00        0.954         NaN  0.257764       NaN  recorded                NaN          NaN
05:00   05:00 2024-11-05 05:00:00 2024-11-05 05:00:00        0.111         NaN  0.030272       NaN  recorded                NaN          NaN
06:00   06:00 2024-11-05 06:00:00 2024-11-05 06:00:00        0.112         NaN  0.031888       NaN  recorded                NaN          NaN
07:00   07:00 2024-11-05 07:00:00 2024-11-05 07:00:00        0.131         NaN  0.042051       NaN  recorded                NaN          NaN
08:00   08:00 2024-11-05 08:00:00 2024-11-05 08:00:00        0.131         NaN  0.042086       NaN  recorded                NaN          NaN
09:00   09:00 2024-11-05 09:00:00 2024-11-05 09:00:00        0.073       0.018  0.021864  0.005391  recorded              0.055     0.016473
10:00   10:00 2024-11-05 10:00:00 2024-11-05 10:00:00        0.063       0.120  0.017927  0.034147  recorded             -0.057    -0.016220
11:00   11:00 2024-11-05 11:00:00 2024-11-05 11:00:00        0.017       0.164  0.004620  0.044566  recorded             -0.147    -0.039946
12:00   12:00 2024-11-05 12:00:00 2024-11-05 12:00:00        0.003       0.166  0.000814  0.045037  recorded             -0.163    -0.044223
13:00   13:00 2024-11-05 13:00:00 2024-11-05 13:00:00        0.205       0.098  0.056692  0.027101  recorded              0.107     0.029590
14:00   14:00 2024-11-05 14:00:00 2024-11-05 14:00:00        0.443       0.013  0.127189  0.003732  recorded              0.430     0.123457
15:00   15:00 2024-11-05 15:00:00 2024-11-05 15:00:00        0.057       0.001  0.019254  0.000338  recorded              0.056     0.018916
16:00   16:00 2024-11-05 16:00:00 2024-11-05 16:00:00        0.000       1.871  0.000000  0.979232  recorded             -1.871    -0.979232
17:00   17:00 2024-11-05 17:00:00 2024-11-05 17:00:00        0.000       3.761  0.000000  2.740907  recorded             -3.761    -2.740907
18:00   18:00 2024-11-05 18:00:00 2024-11-05 18:00:00        0.000       3.744  0.000000  2.538384  recorded             -3.744    -2.538384
0       19:00 2024-11-05 19:00:00 2024-11-05 20:00:00        0.002       3.100  0.001088  1.686562  recorded             -3.098    -1.685474
```

thoutenbos, 2024-11-05:
Het probleem lijkt te zijn dat die `profit` column bij mij `NaN` geeft als er geen productie is. Dit triggert een codepad dat volgens mij nooit kan werken doordat de `price` column in de datatable weggefilterd wordt in de aggregate calculatie binnen `recalc_df_ha`. Ik vermoed dat je zelf geen NaN values hebt bij production en hier daarom ook niet tegaan loopt.

Codepad dat `price` column verwacht bij NaN: https://github.com/corneel27/day-ahead/blob/main/dao/prog/da_report.py#L1086
Hier wordt `price` column toegevoegd aan de dataset: https://github.com/corneel27/day-ahead/blob/main/dao/prog/da_report.py#L1010
En direct daarna weer gefilterd via de aggregate: https://github.com/corneel27/day-ahead/blob/main/dao/prog/da_report.py#L552

corneel27, 2024-11-05:
Mooi, dat je je zo verdiept in de code.
Ben ik echt blij mee!
Het programma probeert eerst consumption en production data op te halen uit de day_ahead database.
Dat zijn dan data die dagelijks beschikbaar komen van Tibber (als je een Tibber abo hebt en die data dagelijks daar ophaalt).
Zijn die data er niet dan pakt DAO de data uit de database van HA. HA houdt uurlijks de database bij.
Die data worden dan mbv de prijzen uit de day-ahead database gebruikt om de kosten en de opbrengsten te berekenen.
Het lijkt erop dat er op 5 nov tussen 0.00 en 9.00 uur geen productie data aanwezig zijn.
Je kunt deze query loslaten op de ha-database:
```
            sql = "SELECT FROM_UNIXTIME(t2.`start_ts`) 'tijd', " \
                  "FROM_UNIXTIME(t2.`start_ts`) 'tot', " \
              "round(greatest(t2.`state` - t1.`state`, 0),3) '" + col_name + "' " \
              "FROM `statistics` t1,`statistics` t2, `statistics_meta` " \
              "WHERE statistics_meta.`id` = t1.`metadata_id` AND statistics_meta.`id` = t2.`metadata_id` " \
              "AND statistics_meta.`statistic_id` = '<sensor_entity_id>' " \
              "AND (t2.`start_ts` = t1.`start_ts` + 3600) " \
              "AND t1.`state` IS NOT null AND t2.`state` IS NOT null " \
              "AND t1.`start_ts` >= UNIX_TIMESTAMP('2024-11-05 00:00') - 3600 " \
              "AND t1.`start_ts` < UNIX_TIMESTAMP('2024-11-06 00:00) - 3600 " \
              "ORDER BY t1.`start_ts`;
```
Je vervangt "sensor_entity_id" door een entity die jij bij production in je settings hebt opgegeven.
Als je bij reports in je instellingen twee sensors voor production hebt opgegeven kun je deze query twee keer uitvoeren.
Ik ben benieuwd wat er uitkomt.
Krijg je wel production (teruglevering) data te zien in het energiedashboard van HA?

corneel27, 2024-11-05:
Ik had nog ergens een beter geformatteerde query liggen.
Die heb ik aangepast:
```
SELECT 
    FROM_UNIXTIME(t2.`start_ts`) 'tijd', 
    FROM_UNIXTIME(t2.`start_ts`) 'tot', 
    round(greatest(t2.`state` - t1.`state`, 0), 3) 'prod_high' 
FROM 
    `statistics` t1,
    `statistics` t2, 
    `statistics_meta` 
WHERE 
    statistics_meta.`id` = t1.`metadata_id` 
    AND statistics_meta.`id` = t2.`metadata_id` 
    AND statistics_meta.`statistic_id` = 'sensor.grid_production_high' 
    AND (t2.`start_ts` = t1.`start_ts` + 3600) 
    AND t1.`state` IS NOT null 
    AND t2.`state` IS NOT null 
    AND t1.`start_ts` >= UNIX_TIMESTAMP('2024-11-05 00:00:00') - 3600 
    AND t1.`start_ts` < UNIX_TIMESTAMP('2024-11-06 00:00:00') - 3600 
ORDER BY 
    t1.`start_ts`;
``

thoutenbos, 2024-11-06:
Thanks! Ik heb inderdaad Tibber dus die krijgt misschien voorkeur boven HA dan

Ik heb drie entities voor consumption / production en allen bevatten ze data voor alle uren

```
    "entities grid consumption": [
      "sensor.p1_meter_3c39e7241cfc_total_power_import"
    ],
    "entities grid production": [
      "sensor.p1_meter_3c39e7241cfc_total_power_export"
    ],
    "entities solar production ac": [
      "sensor.kwh_meter_5c2faf0ebcde_total_power_export"
    ],
```

```
MariaDB [haos]> SELECT
    ->     FROM_UNIXTIME(t2.`start_ts`) 'tijd',
    ->     FROM_UNIXTIME(t2.`start_ts`) 'tot',
    ->     round(greatest(t2.`state` - t1.`state`, 0), 3) 'prod_high'
    -> FROM
    ->     `statistics` t1,
    ->     `statistics` t2,
    ->     `statistics_meta`
    -> WHERE
    ->     statistics_meta.`id` = t1.`metadata_id`
    ->     AND statistics_meta.`id` = t2.`metadata_id`
    ->     AND statistics_meta.`statistic_id` = 'sensor.p1_meter_3c39e7241cfc_total_power_import'
    ->     AND (t2.`start_ts` = t1.`start_ts` + 3600)
    ->     AND t1.`state` IS NOT null
    ->     AND t2.`state` IS NOT null
    ->     AND t1.`start_ts` >= UNIX_TIMESTAMP('2024-11-05 00:00:00') - 3600
    ->     AND t1.`start_ts` < UNIX_TIMESTAMP('2024-11-06 00:00:00') - 3600
    -> ORDER BY
    ->     t1.`start_ts`;
+---------------------+---------------------+-----------+
| tijd                | tot                 | prod_high |
+---------------------+---------------------+-----------+
| 2024-11-05 00:00:00 | 2024-11-05 00:00:00 |     2.748 |
| 2024-11-05 01:00:00 | 2024-11-05 01:00:00 |     2.094 |
| 2024-11-05 02:00:00 | 2024-11-05 02:00:00 |     2.634 |
| 2024-11-05 03:00:00 | 2024-11-05 03:00:00 |     2.634 |
| 2024-11-05 04:00:00 | 2024-11-05 04:00:00 |     0.959 |
| 2024-11-05 05:00:00 | 2024-11-05 05:00:00 |     0.110 |
| 2024-11-05 06:00:00 | 2024-11-05 06:00:00 |     0.113 |
| 2024-11-05 07:00:00 | 2024-11-05 07:00:00 |     0.131 |
| 2024-11-05 08:00:00 | 2024-11-05 08:00:00 |     0.131 |
| 2024-11-05 09:00:00 | 2024-11-05 09:00:00 |     0.073 |
| 2024-11-05 10:00:00 | 2024-11-05 10:00:00 |     0.063 |
| 2024-11-05 11:00:00 | 2024-11-05 11:00:00 |     0.017 |
| 2024-11-05 12:00:00 | 2024-11-05 12:00:00 |     0.003 |
| 2024-11-05 13:00:00 | 2024-11-05 13:00:00 |     0.203 |
| 2024-11-05 14:00:00 | 2024-11-05 14:00:00 |     0.445 |
| 2024-11-05 15:00:00 | 2024-11-05 15:00:00 |     0.057 |
| 2024-11-05 16:00:00 | 2024-11-05 16:00:00 |     0.000 |
| 2024-11-05 17:00:00 | 2024-11-05 17:00:00 |     0.000 |
| 2024-11-05 18:00:00 | 2024-11-05 18:00:00 |     0.000 |
| 2024-11-05 19:00:00 | 2024-11-05 19:00:00 |     0.002 |
| 2024-11-05 20:00:00 | 2024-11-05 20:00:00 |     0.147 |
| 2024-11-05 21:00:00 | 2024-11-05 21:00:00 |     0.288 |
| 2024-11-05 22:00:00 | 2024-11-05 22:00:00 |     0.168 |
| 2024-11-05 23:00:00 | 2024-11-05 23:00:00 |     0.148 |
+---------------------+---------------------+-----------+
24 rows in set (0.005 sec)

MariaDB [haos]> SELECT
    ->     FROM_UNIXTIME(t2.`start_ts`) 'tijd',
    ->     FROM_UNIXTIME(t2.`start_ts`) 'tot',
    ->     round(greatest(t2.`state` - t1.`state`, 0), 3) 'prod_high'
[... 85 lines cut ...]
| 2024-11-05 18:00:00 | 2024-11-05 18:00:00 |     3.745 |
| 2024-11-05 19:00:00 | 2024-11-05 19:00:00 |     3.100 |
| 2024-11-05 20:00:00 | 2024-11-05 20:00:00 |     0.000 |
| 2024-11-05 21:00:00 | 2024-11-05 21:00:00 |     0.000 |
| 2024-11-05 22:00:00 | 2024-11-05 22:00:00 |     0.000 |
| 2024-11-05 23:00:00 | 2024-11-05 23:00:00 |     0.000 |
+---------------------+---------------------+-----------+
24 rows in set (0.002 sec)
```

Echter bij de Tibber production data van de huidige dag heb ik geen rijen voor uren waarin geen productie was. Hier lijkt het op stuk te gaan omdat deze dan als `NaN` in de datatabel terecht komen.

```
MariaDB [day_ahead]> select *, FROM_UNIXTIME(time)  from `values` where variabel = 2 and time >= 1730836800;
+-------+----------+------------+-------+---------------------+
| id    | variabel | time       | value | FROM_UNIXTIME(time) |
+-------+----------+------------+-------+---------------------+
| 11803 |        2 | 1730836800 |     0 | 2024-11-05 21:00:00 |
| 11805 |        2 | 1730840400 |     0 | 2024-11-05 22:00:00 |
| 11807 |        2 | 1730844000 |     0 | 2024-11-05 23:00:00 |
| 11809 |        2 | 1730872800 | 1.868 | 2024-11-06 07:00:00 |
| 11811 |        2 | 1730876400 | 0.003 | 2024-11-06 08:00:00 |
+-------+----------+------------+-------+---------------------+
5 rows in set (0.001 sec)
```

corneel27, 2024-11-06:
Voor vandaag zijn er(nog) geen Tibber data. 
Die Tibber data staan niet in de HA-database, maar in de day-ahead database.
Dus valt DAO terug op data uit de  HA database. Als daar voor vandaag ook geen data zijn, dan hebben we een probleem.
Misschien is de database corrupt. Kun je health-check doen?

corneel27, 2024-11-06:
Er is dus iets fout gegaan met het ophalen van data van Tibber.
Waarschijnlijk een fout aan de kant van Tibber.
Wat je kunt proberen is het volgende:

1. Verwijder alle consumption en production data in de day-ahead database vanaf gisterochtend 5 november 0 uur met een delete-query. Lukt dat of heb je daar hulp bij nodig?
2. Haal opnieuw Tibber-data op via het dashboard, run-menu.

thoutenbos, 2024-11-07:
Ik heb het geprobeerd en Tibber-data ook even in debug gedraaid
Echter voor de huidige dag krijg ik geen voor de uren waarin production 0 is geen records terug van Tibber
Deze lijken pas de volgende dag aangevuld te worden

Het laatste production 0 record is het laatste uur van gister (er is vandaag ook nog geen uur met overschot geweest door het grauwe weer)

```
2024-11-07 09:35:16 info: {'from': '2024-11-06T17:00:00.000+01:00', 'profit': 3.108626246, 'production': 3.782} 1730908800 3.782
2024-11-07 09:35:16 info: {'from': '2024-11-06T17:00:00.000+01:00', 'profit': 3.108626246, 'production': 3.782} 1730908800 3.108626246
2024-11-07 09:35:16 info: {'from': '2024-11-06T18:00:00.000+01:00', 'profit': 3.0757470975, 'production': 3.765} 1730912400 3.765
2024-11-07 09:35:16 info: {'from': '2024-11-06T18:00:00.000+01:00', 'profit': 3.0757470975, 'production': 3.765} 1730912400 3.0757470975
2024-11-07 09:35:16 info: {'from': '2024-11-06T19:00:00.000+01:00', 'profit': 2.048436104, 'production': 3.658} 1730916000 3.658
2024-11-07 09:35:16 info: {'from': '2024-11-06T19:00:00.000+01:00', 'profit': 2.048436104, 'production': 3.658} 1730916000 2.048436104
2024-11-07 09:35:16 info: {'from': '2024-11-06T20:00:00.000+01:00', 'profit': 0, 'production': 0} 1730919600 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T20:00:00.000+01:00', 'profit': 0, 'production': 0} 1730919600 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T21:00:00.000+01:00', 'profit': 0, 'production': 0} 1730923200 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T21:00:00.000+01:00', 'profit': 0, 'production': 0} 1730923200 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T22:00:00.000+01:00', 'profit': 0, 'production': 0} 1730926800 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T22:00:00.000+01:00', 'profit': 0, 'production': 0} 1730926800 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T23:00:00.000+01:00', 'profit': 0, 'production': 0} 1730930400 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-06T23:00:00.000+01:00', 'profit': 0, 'production': 0} 1730930400 0.0
2024-11-07 09:35:16 info: {'from': '2024-11-01T00:00:00.000+01:00', 'cost': 0.04341722, 'consumption': 0.14} 1730415600 0.14
2024-11-07 09:35:16 info: {'from': '2024-11-01T00:00:00.000+01:00', 'cost': 0.04341722, 'consumption': 0.14} 1730415600 0.04341722
2024-11-07 09:35:16 info: {'from': '2024-11-01T01:00:00.000+01:00', 'cost': 0.039476129, 'consumption': 0.133} 1730419200 0.133
2024-11-07 09:35:16 info: {'from': '2024-11-01T01:00:00.000+01:00', 'cost': 0.039476129, 'consumption': 0.133} 1730419200 0.039476129
2024-11-07 09:35:16 info: {'from': '2024-11-01T02:00:00.000+01:00', 'cost': 0.0368899839, 'consumption': 0.133} 1730422800 0.133
2024-11-07 09:35:16 info: {'from': '2024-11-01T02:00:00.000+01:00', 'cost': 0.0368899839, 'consumption': 0.133} 1730422800 0.0368899839
2024-11-07 09:35:16 info: {'from': '2024-11-01T03:00:00.000+01:00', 'cost': 0.0359439091, 'consumption': 0.137} 1730426400 0.137
2024-11-07 09:35:16 info: {'from': '2024-11-01T03:00:00.000+01:00', 'cost': 0.0359439091, 'consumption': 0.137} 1730426400 0.0359439091
2024-11-07 09:35:16 info: {'from': '2024-11-01T04:00:00.000+01:00', 'cost': 0.9829313032, 'consumption': 3.958} 1730430000 3.958
2024-11-07 09:35:16 info: {'from': '2024-11-01T04:00:00.000+01:00', 'cost': 0.9829313032, 'consumption': 3.958} 1730430000 0.9829313032
2024-11-07 09:35:16 info: {'from': '2024-11-01T05:00:00.000+01:00', 'cost': 0.184110696, 'consumption': 0.72} 1730433600 0.72
2024-11-07 09:35:16 info: {'from': '2024-11-01T05:00:00.000+01:00', 'cost': 0.184110696, 'consumption': 0.72} 1730433600 0.184110696
```

corneel27, 2024-11-07:
Ophalen van Tibber data werkt alleen voor dagen van gisteren (vanaf ca 10:00 uur) en daarvoor.. Tibber haalt via de netbeheerder waarden op uit de slimme meter en dat doen ze 's morgens  een keer per dag.
DAO gebruikt de meetdat van HA om het restant (van vandaag) aan te vullen.
Werkt die rapportage van vandaag nog steeds niet?

corneel27, 2024-11-07:
> Klopt het dat DAO alleen de laatste twee uur uitvraagt vanuit HA?

Nee, want dat zou de hele dag moeten zijn, tenzij jij de eerste uren al binnen had gekregen van Tibber. Maar dat kan ook niet 🤔
Zitten er op een of andere manier toch al data in jouw day_ahead database?

> Welke DB engine gebruik je zelf?

Ik gebruik zowel voor HA als day_ahead MariaDB

corneel27, 2024-11-07:
Gaat het energy dashboard van HA wel goed bij jou?
Heb je daar dezelfde sensoren opgegeven?
Je gebruikt sensoren met de term "power" erin. Dat duidt eerder op vermogen. Zijn jouw sensoren oplopende tellerstanden?

thoutenbos, 2024-11-07:
Ja ik gebruik deze sensoren ook in het Energy Dashboard, dat werkt prima. Het zijn Home Wizard producten. Ondanks de wat ongelukkige naamgeving zijn het inderdaad oplopende tellerstanden. In mijn day_ahead database zit zeker al data van vandaag ja. Deze data komt vanuit Tibber en zie ik daar ook in de logs binnenkomen. 

```
MariaDB [day_ahead]> select variabel.name, FROM_UNIXTIME(`values`.time), `values`.*  from `variabel`, `values` where `variabel`.id = `values`.variabel and variabel.id in (1,2) and time >=
 1730934000
+-----------+------------------------------+-------+----------+------------+-------+
| name      | FROM_UNIXTIME(`values`.time) | id    | variabel | time       | value |
+-----------+------------------------------+-------+----------+------------+-------+
| Verbruik  | 2024-11-07 00:00:00          | 12011 |        1 | 1730934000 | 0.132 |
| Verbruik  | 2024-11-07 01:00:00          | 12013 |        1 | 1730937600 | 2.025 |
| Verbruik  | 2024-11-07 02:00:00          | 12015 |        1 | 1730941200 | 2.618 |
| Verbruik  | 2024-11-07 03:00:00          | 12017 |        1 | 1730944800 | 2.618 |
| Verbruik  | 2024-11-07 04:00:00          | 12019 |        1 | 1730948400 | 2.588 |
| Verbruik  | 2024-11-07 05:00:00          | 12021 |        1 | 1730952000 | 2.732 |
| Verbruik  | 2024-11-07 06:00:00          | 12023 |        1 | 1730955600 | 2.877 |
| Verbruik  | 2024-11-07 07:00:00          | 12025 |        1 | 1730959200 | 0.123 |
| Verbruik  | 2024-11-07 08:00:00          | 12027 |        1 | 1730962800 | 0.091 |
| Verbruik  | 2024-11-07 09:00:00          | 12047 |        1 | 1730966400 | 0.286 |
| Verbruik  | 2024-11-07 10:00:00          | 12049 |        1 | 1730970000 | 0.165 |
| Verbruik  | 2024-11-07 11:00:00          | 12075 |        1 | 1730973600 | 0.126 |
| Verbruik  | 2024-11-07 12:00:00          | 12077 |        1 | 1730977200 | 0.342 |
| Verbruik  | 2024-11-07 13:00:00          | 12079 |        1 | 1730980800 | 0.286 |
| Verbruik  | 2024-11-07 14:00:00          | 12115 |        1 | 1730984400 | 0.242 |
| Productie | 2024-11-07 15:00:00          | 12111 |        2 | 1730988000 | 0.584 |
| Verbruik  | 2024-11-07 15:00:00          | 12117 |        1 | 1730988000 | 0.001 |
| Productie | 2024-11-07 16:00:00          | 12113 |        2 | 1730991600 | 3.333 |
| Verbruik  | 2024-11-07 16:00:00          | 12119 |        1 | 1730991600 |     0 |
+-----------+------------------------------+-------+----------+------------+-------+
19 rows in set (0.005 sec)
```

corneel27, 2024-11-07:
Wat ik nu zie is eigenlijk onmogelijk.

Nou gaat er een licht op:
Heb jij misschien de Tibber-pulse draaien en haalt de pulse de waarden op uit je slimme meter en die krijg jij binnen bij het ophalen van je Tibber-data.
Alleen zitten daarin geen 0-waarden voor records  dat er geen verbruik of  productie is.
Wil jij dit bevestigen, dan ga ik voor jou (en waarschijnlijk ook anderen) een oplossing bedenken.
Voor de korte termijn is dan voor jou de oplossing dat je het ophalen van Tibberdata uitzet (die regel in de settings weghalen). Dan gaat het programma vanaf morgen werken met de data uit de HA database.

thoutenbos, 2024-11-07:
Aaaaah wat goed! Die heb ik inderdaad!
Toch nog een doorbraak :-)

ik heb inmiddels zelf wel een workaround gevonden waarmee ik het maandoverzicht alsnog op kan vragen. Maar het zou top zijn mocht je de Pulse ondersteuning er ooit in kunnen maken. Als je nog meer data nodig hebt van het gedrag met de Pulse dan ben ik daar ook zeker voor beschikbaar.

Bedankt voor de volharding in dit issue, en ook zeer veel dank voor het beschikbaar stellen van dit fijne programma! Echt top!

thoutenbos, 2024-11-07:
Gelijk even gedaan (einde dag zodat er lekker veel data in zit)

```
2024-11-07 22:18:29 info: Tibber data present tot en met: 2024-10-22 23:00:00
2024-11-07 22:18:29 info: {'from': '2024-11-01T00:00:00.000+01:00', 'profit': 0, 'production': 0} 1730415600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T00:00:00.000+01:00', 'profit': 0, 'production': 0} 1730415600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T01:00:00.000+01:00', 'profit': 0, 'production': 0} 1730419200 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T01:00:00.000+01:00', 'profit': 0, 'production': 0} 1730419200 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T02:00:00.000+01:00', 'profit': 0, 'production': 0} 1730422800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T02:00:00.000+01:00', 'profit': 0, 'production': 0} 1730422800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T03:00:00.000+01:00', 'profit': 0, 'production': 0} 1730426400 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T03:00:00.000+01:00', 'profit': 0, 'production': 0} 1730426400 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T04:00:00.000+01:00', 'profit': 0, 'production': 0} 1730430000 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T04:00:00.000+01:00', 'profit': 0, 'production': 0} 1730430000 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T05:00:00.000+01:00', 'profit': 0, 'production': 0} 1730433600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T05:00:00.000+01:00', 'profit': 0, 'production': 0} 1730433600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T06:00:00.000+01:00', 'profit': 0.000264627, 'production': 0.001} 1730437200 0.001
2024-11-07 22:18:29 info: {'from': '2024-11-01T06:00:00.000+01:00', 'profit': 0.000264627, 'production': 0.001} 1730437200 0.000264627
2024-11-07 22:18:29 info: {'from': '2024-11-01T07:00:00.000+01:00', 'profit': 0, 'production': 0} 1730440800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T07:00:00.000+01:00', 'profit': 0, 'production': 0} 1730440800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T08:00:00.000+01:00', 'profit': 0, 'production': 0} 1730444400 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T08:00:00.000+01:00', 'profit': 0, 'production': 0} 1730444400 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T09:00:00.000+01:00', 'profit': 0, 'production': 0} 1730448000 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T09:00:00.000+01:00', 'profit': 0, 'production': 0} 1730448000 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T10:00:00.000+01:00', 'profit': 0.000282293, 'production': 0.001} 1730451600 0.001
2024-11-07 22:18:29 info: {'from': '2024-11-01T10:00:00.000+01:00', 'profit': 0.000282293, 'production': 0.001} 1730451600 0.000282293
2024-11-07 22:18:29 info: {'from': '2024-11-01T11:00:00.000+01:00', 'profit': 0.000554906, 'production': 0.002} 1730455200 0.002
2024-11-07 22:18:29 info: {'from': '2024-11-01T11:00:00.000+01:00', 'profit': 0.000554906, 'production': 0.002} 1730455200 0.000554906
2024-11-07 22:18:29 info: {'from': '2024-11-01T12:00:00.000+01:00', 'profit': 0.029588493, 'production': 0.111} 1730458800 0.111
2024-11-07 22:18:29 info: {'from': '2024-11-01T12:00:00.000+01:00', 'profit': 0.029588493, 'production': 0.111} 1730458800 0.029588493
2024-11-07 22:18:29 info: {'from': '2024-11-01T13:00:00.000+01:00', 'profit': 0.062704378, 'production': 0.226} 1730462400 0.226
2024-11-07 22:18:29 info: {'from': '2024-11-01T13:00:00.000+01:00', 'profit': 0.062704378, 'production': 0.226} 1730462400 0.062704378
2024-11-07 22:18:29 info: {'from': '2024-11-01T14:00:00.000+01:00', 'profit': 0.01158212, 'production': 0.04} 1730466000 0.04
2024-11-07 22:18:29 info: {'from': '2024-11-01T14:00:00.000+01:00', 'profit': 0.01158212, 'production': 0.04} 1730466000 0.01158212
2024-11-07 22:18:29 info: {'from': '2024-11-01T15:00:00.000+01:00', 'profit': 0, 'production': 0} 1730469600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T15:00:00.000+01:00', 'profit': 0, 'production': 0} 1730469600 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T16:00:00.000+01:00', 'profit': 0, 'production': 0} 1730473200 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T16:00:00.000+01:00', 'profit': 0, 'production': 0} 1730473200 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T17:00:00.000+01:00', 'profit': 0, 'production': 0} 1730476800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T17:00:00.000+01:00', 'profit': 0, 'production': 0} 1730476800 0.0
2024-11-07 22:18:29 info: {'from': '2024-11-01T18:00:00.000+01:00', 'profit': 0.0016019795, 'production': 0.005} 1730480400 0.005
2024-11-07 22:18:29 info: {'from': '2024-11-01T18:00:00.000+01:00', 'profit': 0.0016019795, 'production': 0.005} 1730480400 0.0016019795
2024-11-07 22:18:29 info: {'from': '2024-11-01T19:00:00.000+01:00', 'profit': 0.003603864, 'production': 0.012} 1730484000 0.012
2024-11-07 22:18:29 info: {'from': '2024-11-01T19:00:00.000+01:00', 'profit': 0.003603864, 'production': 0.012} 1730484000 0.003603864
2024-11-07 22:18:29 info: {'from': '2024-11-01T20:00:00.000+01:00', 'profit': 0.0041160812, 'production': 0.014} 1730487600 0.014
2024-11-07 22:18:29 info: {'from': '2024-11-01T20:00:00.000+01:00', 'profit': 0.0041160812, 'production': 0.014} 1730487600 0.0041160812
2024-11-07 22:18:29 info: {'from': '2024-11-01T21:00:00.000+01:00', 'profit': 0.003402036, 'production': 0.012} 1730491200 0.012
2024-11-07 22:18:29 info: {'from': '2024-11-01T21:00:00.000+01:00', 'profit': 0.003402036, 'production': 0.012} 1730491200 0.003402036
[... cut ...]
2024-11-07 22:18:30 info: {'from': '2024-11-07T19:00:00.000+01:00', 'cost': 0.0130893444, 'consumption': 0.036} 1731002400 0.036
2024-11-07 22:18:30 info: {'from': '2024-11-07T19:00:00.000+01:00', 'cost': 0.0130893444, 'consumption': 0.036} 1731002400 0.0130893444
2024-11-07 22:18:30 info: {'from': '2024-11-07T20:00:00.000+01:00', 'cost': 0.0464479796, 'consumption': 0.148} 1731006000 0.148
2024-11-07 22:18:30 info: {'from': '2024-11-07T20:00:00.000+01:00', 'cost': 0.0464479796, 'consumption': 0.148} 1731006000 0.0464479796
```

corneel27, 2024-11-07:
Nou alle data t/m 21:00 (dus tot 22:00 uur) zijn present.
Wat zit er nu voor vandaag aan data in de database van de variabelen met id 1, 2, 7 en 8 ?

thoutenbos, 2024-11-07:
Alleen voor de productie is er geen regel bij Tibber als er geen productie was, ook goed te zien in de db

```
MariaDB [day_ahead]> select variabel.name, FROM_UNIXTIME(`values`.time), `values`.*  from `variabel`, `values` where `variabel`.id = `values`.variabel and variabel.id in (1,2,7,8) and time >= 1730934000 order by `variabel`, `time`;
+-----------+------------------------------+-------+----------+------------+------------+
| name      | FROM_UNIXTIME(`values`.time) | id    | variabel | time       | value      |
+-----------+------------------------------+-------+----------+------------+------------+
| Verbruik  | 2024-11-07 00:00:00          | 12011 |        1 | 1730934000 |      0.132 |
| Verbruik  | 2024-11-07 01:00:00          | 12013 |        1 | 1730937600 |      2.025 |
| Verbruik  | 2024-11-07 02:00:00          | 12015 |        1 | 1730941200 |      2.618 |
| Verbruik  | 2024-11-07 03:00:00          | 12017 |        1 | 1730944800 |      2.618 |
| Verbruik  | 2024-11-07 04:00:00          | 12019 |        1 | 1730948400 |      2.588 |
| Verbruik  | 2024-11-07 05:00:00          | 12021 |        1 | 1730952000 |      2.732 |
| Verbruik  | 2024-11-07 06:00:00          | 12023 |        1 | 1730955600 |      2.877 |
| Verbruik  | 2024-11-07 07:00:00          | 12025 |        1 | 1730959200 |      0.123 |
| Verbruik  | 2024-11-07 08:00:00          | 12027 |        1 | 1730962800 |      0.091 |
| Verbruik  | 2024-11-07 09:00:00          | 12047 |        1 | 1730966400 |      0.286 |
| Verbruik  | 2024-11-07 10:00:00          | 12049 |        1 | 1730970000 |      0.165 |
| Verbruik  | 2024-11-07 11:00:00          | 12075 |        1 | 1730973600 |      0.126 |
| Verbruik  | 2024-11-07 12:00:00          | 12077 |        1 | 1730977200 |      0.342 |
| Verbruik  | 2024-11-07 13:00:00          | 12079 |        1 | 1730980800 |      0.286 |
| Verbruik  | 2024-11-07 14:00:00          | 12115 |        1 | 1730984400 |      0.242 |
| Verbruik  | 2024-11-07 15:00:00          | 12117 |        1 | 1730988000 |      0.001 |
| Verbruik  | 2024-11-07 16:00:00          | 12119 |        1 | 1730991600 |          0 |
| Verbruik  | 2024-11-07 17:00:00          | 12127 |        1 | 1730995200 |          0 |
| Verbruik  | 2024-11-07 18:00:00          | 12129 |        1 | 1730998800 |          0 |
| Verbruik  | 2024-11-07 19:00:00          | 12131 |        1 | 1731002400 |      0.036 |
| Verbruik  | 2024-11-07 20:00:00          | 12133 |        1 | 1731006000 |      0.148 |
| Verbruik  | 2024-11-07 21:00:00          | 12135 |        1 | 1731009600 |      0.157 |
| Productie | 2024-11-07 15:00:00          | 12111 |        2 | 1730988000 |      0.584 |
| Productie | 2024-11-07 16:00:00          | 12113 |        2 | 1730991600 |      3.333 |
| Productie | 2024-11-07 17:00:00          | 12121 |        2 | 1730995200 |       3.74 |
| Productie | 2024-11-07 18:00:00          | 12123 |        2 | 1730998800 |      3.742 |
| Productie | 2024-11-07 19:00:00          | 12125 |        2 | 1731002400 |      0.262 |
| cost      | 2024-11-07 00:00:00          | 12012 |        7 | 1730934000 |  0.0379239 |
| cost      | 2024-11-07 01:00:00          | 12014 |        7 | 1730937600 |   0.564562 |
| cost      | 2024-11-07 02:00:00          | 12016 |        7 | 1730941200 |    0.72143 |
| cost      | 2024-11-07 03:00:00          | 12018 |        7 | 1730944800 |   0.717629 |
| cost      | 2024-11-07 04:00:00          | 12020 |        7 | 1730948400 |   0.713226 |
| cost      | 2024-11-07 05:00:00          | 12022 |        7 | 1730952000 |   0.736845 |
| cost      | 2024-11-07 06:00:00          | 12024 |        7 | 1730955600 |   0.796492 |
| cost      | 2024-11-07 07:00:00          | 12026 |        7 | 1730959200 |  0.0446847 |
| cost      | 2024-11-07 08:00:00          | 12028 |        7 | 1730962800 |  0.0320431 |
| cost      | 2024-11-07 09:00:00          | 12048 |        7 | 1730966400 |  0.0958413 |
| cost      | 2024-11-07 10:00:00          | 12050 |        7 | 1730970000 |  0.0540293 |
| cost      | 2024-11-07 11:00:00          | 12076 |        7 | 1730973600 |  0.0392173 |
| cost      | 2024-11-07 12:00:00          | 12078 |        7 | 1730977200 |   0.104026 |
| cost      | 2024-11-07 13:00:00          | 12080 |        7 | 1730980800 |  0.0887713 |
| cost      | 2024-11-07 14:00:00          | 12116 |        7 | 1730984400 |  0.0753426 |
| cost      | 2024-11-07 15:00:00          | 12118 |        7 | 1730988000 | 0.00037291 |
| cost      | 2024-11-07 16:00:00          | 12120 |        7 | 1730991600 |          0 |
| cost      | 2024-11-07 17:00:00          | 12128 |        7 | 1730995200 |          0 |
| cost      | 2024-11-07 18:00:00          | 12130 |        7 | 1730998800 |          0 |
| cost      | 2024-11-07 19:00:00          | 12132 |        7 | 1731002400 |  0.0130893 |
| cost      | 2024-11-07 20:00:00          | 12134 |        7 | 1731006000 |   0.046448 |
| cost      | 2024-11-07 21:00:00          | 12136 |        7 | 1731009600 |  0.0462045 |
| profit    | 2024-11-07 15:00:00          | 12112 |        8 | 1730988000 |   0.217779 |
| profit    | 2024-11-07 16:00:00          | 12114 |        8 | 1730991600 |    1.50114 |
| profit    | 2024-11-07 17:00:00          | 12122 |        8 | 1730995200 |    2.26469 |
| profit    | 2024-11-07 18:00:00          | 12124 |        8 | 1730998800 |      1.843 |
| profit    | 2024-11-07 19:00:00          | 12126 |        8 | 1731002400 |  0.0952613 |
+-----------+------------------------------+-------+----------+------------+------------+
54 rows in set (0.004 sec)
```

corneel27, 2024-11-07:
Ik ga ermee aan de slag, wordt vervolgd.

corneel27, 2024-11-07:
Ik zie nu dat er gaten zitten in de data van "vandaag". Alle nul-records worden niet meegestuurd:
```
2024-11-07 22:22:07 info: {'from': '2024-11-06T22:00:00.000+01:00', 'profit': 0, 'production': 0} 1730926800 0.0
2024-11-07 22:22:07 info: {'from': '2024-11-06T22:00:00.000+01:00', 'profit': 0, 'production': 0} 1730926800 0.0
2024-11-07 22:22:07 info: {'from': '2024-11-06T23:00:00.000+01:00', 'profit': 0, 'production': 0} 1730930400 0.0
2024-11-07 22:22:07 info: {'from': '2024-11-06T23:00:00.000+01:00', 'profit': 0, 'production': 0} 1730930400 0.0
2024-11-07 22:22:07 info: {'from': '2024-11-07T15:00:00.000+01:00', 'profit': 0.2177793816, 'production': 0.584} 1730988000 0.584
2024-11-07 22:22:07 info: {'from': '2024-11-07T15:00:00.000+01:00', 'profit': 0.2177793816, 'production': 0.584} 1730988000 0.2177793816
2024-11-07 22:22:07 info: {'from': '2024-11-07T16:00:00.000+01:00', 'profit': 1.5011372046, 'production': 3.333} 1730991600 3.333
2024-11-07 22:22:07 info: {'from': '2024-11-07T16:00:00.000+01:00', 'profit': 1.5011372046, 'production': 3.333} 1730991600 1.5011372046
2024-11-07 22:22:07 info: {'from': '2024-11-07T17:00:00.000+01:00', 'profit': 2.264691176, 'production': 3.74} 1730995200 3.74
2024-11-07 22:22:07 info: {'from': '2024-11-07T17:00:00.000+01:00', 'profit': 2.264691176, 'production': 3.74} 1730995200 2.264691176
2024-11-07 22:22:07 info: {'from': '2024-11-07T18:00:00.000+01:00', 'profit': 1.8430038528, 'production': 3.742} 1730998800 3.742
2024-11-07 22:22:07 info: {'from': '2024-11-07T18:00:00.000+01:00', 'profit': 1.8430038528, 'production': 3.742} 1730998800 1.8430038528
2024-11-07 22:22:07 info: {'from': '2024-11-07T19:00:00.000+01:00', 'profit': 0.0952613398, 'production': 0.262} 1731002400 0.262
2024-11-07 22:22:07 info: {'from': '2024-11-07T19:00:00.000+01:00', 'profit': 0.0952613398, 'production': 0.262} 1731002400 0.0952613398
```
Tot gisteren 23:00 uur is alles aanwezig van production (ook nul-waarden).
Vandaag pas vanaf 15:00 uur.

Ik denk dat de simpelste oplossing is dat ik data van "vandaag" negeer en niet opsla.
Dan valt DAO voor "vandaag" eenvoudig terug op de aanwezige data in de HA-database (zoals bij alle mensen zonder een Tibber-pulse). 
Ben jij hiermee akkoord?

thoutenbos, 2024-11-07:
Jazeker, klinkt goed

corneel27, 2024-11-08:
Inmiddels staat versie 2024.11.1 op github met de volgende changelog:

> Fixed an error when getting Tibber-data when using a Tibber pulse: 
> only data before today are stored

Je kunt de DAO addon updaten. Je kunt dit versnellen door in de add-on shop van HA te klikken op"zoek naar updates" in het puntjes-menu.

thoutenbos, 2024-11-08:
Bedankt! Heb de update geïnstalleerd. Morgen laat ik weten of het werkt (heb vandaag al data binnen gehad)

corneel27, 2024-11-08:
Je zou met een query alle consumption data vanaf 1-11 kunnen verwijderen en dan via het run menu opnieuw data ophalen. Dan moet het goed werken.

thoutenbos, 2024-11-09:
Werkt weer fantastisch!
Heel veel dank Cees 😃

closed 2024-11-09 by corneel27