corneel27/day-ahead #143: Bad gateway 502

syntax-error1978, 2024-11-05:
Hallo,
Kan ik een klein beetje hulp krijgen met installeren van de addon?

Als ik probeer de addon te starten krijg ik bij open de web ui de melding "bad gateway"

wat heb ik geprobeerd:
home assistant herstarten
Addon herbouwen
Addon verwijderen en opnieuw installeren 

in het verleden heeft de addon wel gedraaid 

alvast bedankt!

corneel27, 2024-11-05:
Je kunt inderdaad het beste beginnen van scratch en opnieuw installeren.
Maar de add-on alleen "verwijderen" via de knop op het add-on scherm is niet voldoende.
De oude instellingen blijven bewaardnen die zitten jou in de weg.
Er zijn twee mogelijken:

1. Je verwijderd voor je alles opnieuw installeert ook je instellingen. Dat gaat alleen als je toegang tot HA hebt met samba (dat is ook een add-on). Dan ga je met je file-explorer naar de map "addon_configs" en je verwijdert de submap `325af97a_day_ahead_opt` en je volgt de tutorial in DOCS.md.
2. Je kijkt in de logging van de add-on (op de informatie pagina rechtsboven "logboek"), waar het fout gaat. Misschien kun je het in de instellingen corrigeren. Of je kopieert de instellingen van het bestand "options_start.json" naar options.json in de map dao_data.

Succes!

syntax-error1978, 2024-11-09:
Bovenstaande werkte! 
had ik zelf ook wel kunnen bedenken, soms moet je nog iets zelf doen ;)

afin, het draait weer, nu alles proberen in te stellen.

ik loop wel tegen 1 dingetje aan: 1 omvormer met 2 oriëntaties, 3.5kw oost en 4.3kw west. ik heb ze nu als 2 systemen gedaan met dezelfde automatisering om ze uit te laten zetten maar gaat dit goed? niet dat er veel negatieve uren zijn...

corneel27, 2024-11-09:
Ja dat gaat zeker werken (als je geen typefouten hebt gemaakt 😏).

closed 2024-11-09 by corneel27