corneel27/day-ahead #153: Optimistic charging

thoutenbos, 2024-11-23:
Hoi Cees,

Weer een vraagje. Ik wil er wederom bij zetten dat ik echt een enorm blije gebruiker ben van je programma. Zie mijn vragen dus graag ook louter als suggesties om van 99% naar 100% te gaan. Echt super dat je dit allemaal deelt en ondersteunt!!!

Komende nacht/dag zijn de prijzen (ex vat/markup) lange tijd 0 cent. Omdat de prijzen morgen gedurende de hele termijn niet veel hoger liggen wordt de accu echter niet volledig benut.

[image]
[image]

Ideaal zou zijn als de accu bij 0-tarief sowieso volgeladen wordt tot 100% met de optimistische verwachting dat we dit ergens in de komende dagen wel weer kwijt kunnen voor een mooie prijs. Kan me zo voorstellen dat dit wel lastig is, omdat het ook buiten de periode van bekende prijzen valt. Misschien heb jij nog een goed idee?

Groeten,
Thijs

corneel27, 2024-11-23:
Dag Tijs,
Ik kijk om 13:00 uur altijd naar de prijzen en de berekening van morgen.
Aan de hand daarvan "plan" ik o.a. het laden van de auto.
Toen ik de prijzen voor morgen zag heb ik de auto op 100% gezet morgen om 13:00 uur klaar et voila bij de berekening van 14:00 vandaag:
[image]
Dus wat jij wil kan al, maar jij wilt dat geautomatiseerd?
Ik ga erover nadenken.
Dank voor de tip!

Zo lijkt het me ook (zeker in de winter) handig om de accu bij lage prijzen meer te laden.
Ook daar ga ikover nadenken.

Groeten van Cees

corneel27, 2024-11-23:
Oh, maar dat kan ook. Je kunt bij de instellingen twee entities opgeven, waarmee in je HA het minimum en maximum van je SoC aan het einde de plannings periode. Bijvoorbeeld 
```
      "entity min soc end opt": "input_number.min_soc_einde_opt",
```
Zie Docs.md

thoutenbos, 2024-11-23:
Tek gek!! Die had in gemist idd

closed 2024-11-23 by thoutenbos