corneel27/day-ahead #189: versie 2025.1.0 loopt vast

sjampeter, 2025-01-04:
sinds versie 2025.1.0 loopt het script vast bij mij.
ook bij de nieuwere versie 2025.1.1 is dit het geval.
ik kom van versie 2024.12.3

bij meer mensen het geval?

onderstaand een deel van het log :

\`input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: Fout bij schrijven naar input_text.dao_notificatie, waarde 2025-01-04 12:00:36 fout: File: /root/dao/prog/da_scheduler.py, line 65, in <module>
maximum recursion depth exceeded
[2025-01-04 12:13:15 +0100] [10] [CRITICAL] WORKER TIMEOUT (pid:17)
[2025-01-04 12:13:15 +0100] [17] [ERROR] Error handling request /
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 134, in handle
    self.handle_request(listener, req, client, addr)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 177, in handle_request
    respiter = self.wsgi(environ, resp.start_response)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1498, in __call__
    return self.wsgi_app(environ, start_response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 115, in menu
    return run_process()
           ^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 232, in run_process
    proc = run(cmd, stdout=PIPE, stderr=PIPE)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 550, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 2059, in _communicate
    ready = selector.select(timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/selectors.py", line 415, in select
    fd_event_list = self._selector.poll(timeout)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/base.py", line 204, in handle_abort
    sys.exit(1)
SystemExit: 1
[2025-01-04 12:13:15 +0100] [17] [INFO] Worker exiting (pid: 17)
[2025-01-04 12:13:16 +0100] [26] [INFO] Booting worker with pid: 26
[2025-01-04 12:16:02 +0100] [10] [CRITICAL] WORKER TIMEOUT (pid:26)
[2025-01-04 12:16:02 +0100] [26] [ERROR] Error handling request /
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 134, in handle
    self.handle_request(listener, req, client, addr)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 177, in handle_request
    respiter = self.wsgi(environ, resp.start_response)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1498, in __call__
    return self.wsgi_app(environ, start_response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 115, in menu
    return run_process()
           ^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 232, in run_process
    proc = run(cmd, stdout=PIPE, stderr=PIPE)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 550, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 2059, in _communicate
    ready = selector.select(timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/selectors.py", line 415, in select
    fd_event_list = self._selector.poll(timeout)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/base.py", line 204, in handle_abort
    sys.exit(1)
SystemExit: 1
[2025-01-04 12:16:02 +0100] [26] [INFO] Worker exiting (pid: 26)
[2025-01-04 12:16:03 +0100] [29] [INFO] Booting worker with pid: 29`

corneel27, 2025-01-04:
Aan dat deel van de software is helemaal niets veranderd.
Hij probeert een logfile weg te schrijven naar de directory dao_data\log\
Kun je die map via samba bereiken?
Is er genoeg schijfruimte?
Heb je een backup gemaakt tijdens het installeren van 2025.1.0?
Misschien kun je die backup terugzetten en kijken of de fout er dan nog is.
Ik hoor het graag.

sjampeter, 2025-01-04:
nu terug gegaan naar versie 2024-11-1. dat werkt nu weer even. ik kijk het even aan voor ik weer update naar 2025.1.1. 
het is vreemd daar de schijfruimte voldoende is.

corneel27, 2025-01-04:
Fijn dat het voorlopig weer werkt.
Ontstaat deze fout als je een commando geeft via het run menu en zoja welk commando?
Gigenn de gewone taken via de scheduler wel goed?

sjampeter, 2025-01-04:
morgen doe ik nog een poging om naar de nieuwste versie te gaan.
vooralsnog kan ik er de vinger niet op leggen wat en wanneer hier nu iets fout gaat.
wordt vervolgd

corneel27, 2025-01-04:
Prima als je het nog eens probeert.
Ik begrijp dat het niets met schijfruimte of toegankelijkheid van de data-map te maken heeft (anders deed versie2024.11.1. het ook niet).
Zou je dan mijn volgende voorgaande vragen willen  beantwoorden:

> Ontstaat deze fout als je een commando geeft via het run menu en zoja welk commando?
Gingen de gewone taken via de scheduler wel goed?

Ik wil het graag begrijpen maar vooral oplossen!

sjampeter, 2025-01-05:
update:

geheugen is in orde. bedankt voor de tip betreffende systemmonitor.
deze had ik nog niet.
systeem draait in proxmox op een thin client.
geheugen uitgebreidt met 25 gb voor de zekerheid.

bij update naar meest recente versie gaat het wederom mis.
weerdata en prijzen zijn prima op te halen.
zodra ik een optimalisatieberekening wil laten uitvoeren, gaat het fout.

onderstaand nog een stukje recente log.

`[2025-01-05 11:40:08 +0100] [24] [INFO] Worker exiting (pid: 24)
[2025-01-05 11:40:09 +0100] [33] [INFO] Booting worker with pid: 33
[2025-01-05 11:44:28 +0100] [10] [CRITICAL] WORKER TIMEOUT (pid:16)
[2025-01-05 11:44:28 +0100] [16] [ERROR] Error handling request /
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 134, in handle
    self.handle_request(listener, req, client, addr)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 177, in handle_request
    respiter = self.wsgi(environ, resp.start_response)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1498, in __call__
    return self.wsgi_app(environ, start_response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 115, in menu
    return run_process()
           ^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 232, in run_process
    proc = run(cmd, stdout=PIPE, stderr=PIPE)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 550, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 2059, in _communicate
    ready = selector.select(timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/selectors.py", line 415, in select
    fd_event_list = self._selector.poll(timeout)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/base.py", line 204, in handle_abort
    sys.exit(1)
SystemExit: 1
[2025-01-05 11:44:28 +0100] [16] [INFO] Worker exiting (pid: 16)
[2025-01-05 11:44:29 +0100] [36] [INFO] Booting worker with pid: 36
[2025-01-05 11:48:25 +0100] [10] [CRITICAL] WORKER TIMEOUT (pid:33)
[2025-01-05 11:48:25 +0100] [33] [ERROR] Error handling request /
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 134, in handle
    self.handle_request(listener, req, client, addr)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/sync.py", line 177, in handle_request
    respiter = self.wsgi(environ, resp.start_response)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1498, in __call__
    return self.wsgi_app(environ, start_response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
[... 49 lines cut ...]
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 115, in menu
    return run_process()
           ^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 232, in run_process
    proc = run(cmd, stdout=PIPE, stderr=PIPE)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 550, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 1207, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/subprocess.py", line 2059, in _communicate
    ready = selector.select(timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/selectors.py", line 415, in select
    fd_event_list = self._selector.poll(timeout)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/workers/base.py", line 204, in handle_abort
    sys.exit(1)
SystemExit: 1
[2025-01-05 11:55:59 +0100] [36] [INFO] Worker exiting (pid: 36)
[2025-01-05 11:55:59 +0100] [48] [INFO] Booting worker with pid: 48`

sjampeter, 2025-01-05:
terug naar 2024.11.1 en draait wederom probleemloos.
zeer vreemd

corneel27, 2025-01-05:
Dit vind ik wel heel bijzonder, maar wel fijn voor je.
Heb je niets aan de instellingen van de Proxmox veranderd of iets in een andere applicatie gewijzigd?

sjampeter, 2025-01-05:
dat is het gekke inderdaad. helemaal niets veranderd.
bij de update naar 2025.1.1 ging het mis en krijg ik het niet meer goed.
zelfs heel de integratie al eens verwijderd en opnieuw geinstalleerd.

ik zal morgen eens versie voor versie proberen om te kijken waar het mis gaat

corneel27, 2025-01-07:
Dan moet je een "local addon" installeren:
- download de zip-file
- mbv Samba en je "explorer" kopieer je de map **dao** in de zipfile naar de map **addons** van je HA-machine
- Ga in HA naar Instellingen/Add-ons 
- ga naar de add-winkel
- klik via het 3-puntjes menu op "updates zoeken"
- verlaat de winkel, wacht even, klik op F5 (verversen)
- dan krijg je de addon in beeld en kun je hem gewoon installeren
- kopieer je instellingen en secrets via samba en addon-configs of via het dashboard.

Let op: draai nooit voor langere tijd twee keer DAO naast elkaar, omdat er dan schrijffouten naar entities in HA of naar de database kunnen optreden.

sjampeter, 2025-01-08:
goede nieuws : met samba share is dit inderdaad kinderspel. super.
slechte nieuws : welke versie ik ook probeer op deze manier ( ook oudere versies als degene die nu draait) krijg ik niet aan de praat. 
werkwijze. de huidige dao zet ik stop. dan installeer ik de nieuwe (oude) versie en probeer deze te starten. hier gaat het al mis. zie onderstaande log:
ik vermoed dat er hier iets anders aan de hand is. 
zou een totale verwijdering van het dao programma een mogelijke oplossing kunnen zijn?
uiteraard even de instellingen en secrets bewaren in een kladblokje.

=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data doesn't exist, made
ln: failed to create symbolic link 'app/static/data': File exists
Setting up watches.
Watches established.
Failed to read config file: gunicorn_config.py
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/app/base.py", line 110, in get_config_from_filename
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/dao/webserver/gunicorn_config.py", line 5, in <module>
    port = get_config(app_datapath + "options.json", ["dashboard", "port"], 5000)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 117, in get_config
    config = Config(file_name=file_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 22, in __init__
    self.options = self.parse(file_name)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 14, in parse
    with open(file_name, "r") as file_json:
         ^^^^^^^^^^^^^^^^^^^^
NotADirectoryError: [Errno 20] Not a directory: 'app/static/data/options.json'
=> directory dao_data made, files copied
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data doesn't exist, made
ln: failed to create symbolic link 'app/static/data': File exists
Setting up watches.
Watches established.
Failed to read config file: gunicorn_config.py
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/app/base.py", line 110, in get_config_from_filename
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/dao/webserver/gunicorn_config.py", line 5, in <module>
    port = get_config(app_datapath + "options.json", ["dashboard", "port"], 5000)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 44, in get_config
    config = Config(file_name=file_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 17, in __init__
    self.options = self.parse(file_name)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../../dao/prog/da_config.py", line 9, in parse
    with open(file_name, "r") as file_json:
         ^^^^^^^^^^^^^^^^^^^^
NotADirectoryError: [Errno 20] Not a directory: 'app/static/data/options.json'

corneel27, 2025-01-08:
> => /root/dao/webserver/app/static/data doesn't exist, made
ln: failed to create symbolic link 'app/static/data': File exists

Dit vind ik heel bijzonder:
Enerzijds: bestand niet, dus ik ga het maken (zo is het ook bedoeld)
Anderzijds: kan symbolic link bestand niet maken want bestand bestaat al.

Ik heb ook anderen gehad die problemen hadden met de combinatie promox  en add-on DAO.
Maar versie 2024.11.1 draait wel?

Inderdaad: alleen verwijderen via Instellingen en Add-ons is geen totale verwijdering.
Met name de bestanden in `addon_configs` blijven bestaan. Dat is een keuze van HA. Ik vermoed voor "spijtoptanten".
Dus als je na verwijdering van de add-ons via HA en daarna die map via samba verwijdert. Dan begin jemet een schone lei.
In ieder geval het proberen waard.
Succes!

sjampeter, 2025-01-09:
thanks. ik weet niet wat er fout gaat, maar best is met schone lei wederom beginnen denk ik.
heel bijzonder allemaal, maar het houd je van de straat zeg maar.

komend weekend pak ik dit op, voor nu draait het maar als ik hierin begin te rommelen wil ik voldoende tijd hebben om te repareren en eventuele bijkomende problemen (gaat altijd zo haha) op te kunnen lossen.

keep you posted !

sjampeter, 2025-01-12:
nieuwe install is helaas geen oplossing. 
ik heb dao verwijderd, tevens gekeken via samba naar de map "addon_configs/94e386ba-day-ahead-opt/" . 
deze werd automatisch al verwijderd door hass, dus hier kon ik niks aan veranderen.
het enige wat ik nog kan verzinnen is om nogmaals helemaal opnieuw te installeren en daarbovenop de database te verwijderen. 
echter weet ik niet of dit uberhaubt kan en wenselijk is.
via Phpmyadmin kun je volgens mij de gehele database verwijderen van day ahead.
wat denk jij?

[image]

corneel27, 2025-01-12:
Je kunt de database verwijderen en ook alle config data. Je krijgt dan een "lege" database.
Na installatie van de addon  zul je de database weer moeten vullen met prijzen en meteodata.
Voor de test kun je ook de day-ahead database alleen op Sqlite zetten. Dan begin je ook opnieuw.

Maar ik twijfel of het daaraan ligt.
Ik denk dat we het moeten zoeken in de proxmox instellingen.
Ik ga ook nog even zoeken...

corneel27, 2025-01-12:
Heb je dit gelezen;
https://community.home-assistant.io/t/add-ons-not-allowing-configuration-and-not-showing-as-started-when-it-actually-starts/457407

corneel27, 2025-01-12:
Ik denk dat deze topic voor jou nog meer interessant is.

corneel27, 2025-01-12:
En wellicht ten overvloede:
https://www.wundertech.net/how-to-set-up-home-assistant-on-proxmox/

sjampeter, 2025-01-15:
blijft bijzonder dit. mijn zwager draait exact dezelfde configuratie op proxmox. hij kan zonder problemen naar 2025.1.2.  lijkt mij dan ook bijzonder dat dit probleem veroorzaakt wordt door proxmox. ik zal nog eens in je aangegeven lectuur duiken

sjampeter, 2025-01-15:
probleem zit echt in de database. neem ik de da en ha database over zoals ingesteld in verse versie 2025.2.1, dus met sqlite, draait alles prima. moet trouwens alle 2 de databases op sqlite zetten.

sjampeter, 2025-01-15:
[image]
deze is misschien ook wel interessant. lijkt de mariaDB niet te mogen bereiken terwijl naam en ww niet veranderd is. deze komen feitelijk ook overeen. the plot thickens.

corneel27, 2025-01-15:
Een paar vragen:
- van welke addon komt die logging hierboven?
- hoe heb jij mariadb geïnstalleerd aparte addon in HA of als extra server op proxmox
- hoe zien jouw db configuraties in dao eruit?

sjampeter, 2025-01-17:
excuus voor de wat late reactie.

die logging komt van mariadb.
mariadb draait als aparte addon in HA.

[image]

sjampeter, 2025-01-17:
draai nu op sqlite op versie 2024.11.1. 
draait opzich prima
updaten naar 2025-1-2 en wederom fout (met database sqlite)
zie onderstaande log van dao

[image]

backup terug gezet van 2024.11.1 (ook op basis sqlite) en draait als een zonnetje!!
heel vreemd dit allemaal

sjampeter, 2025-01-18:
voor de duidelijkheid, zelfs met onderstaande instellingen draait ie wel op 2024.11.1 maar niet op welke andere versie ook.

[image]

er kunnen toch geen fouten sluipen in 2 databases? dat er 1 corrupt gaat kan ik nog begrijpen, echter de 2e die het zelfde gedrag vertoont?
vreemd

corneel27, 2025-01-18:
Ik heb intussen gekeken naar de verschillen op db-gebied tussen die twee versies: 2024.11.1 en 2025.1.2
Maar het enige wat echt veranderd is het wegwerken van fouten met de database postgresql.
Bij de onderliggende python-modules (zie requirements.txt) is een wijziging doorgevoerd dir misschien de fout kan veroorzaken.
Daar staat nu (versie 2025.1.2) de regel: `sqlalchemy~=2.0.35`
In versie 2024.10.4 stond daar; `sqlalchemy`, dus **zonder** versienummer erachter.
Je zou eens kunnen proberen door versie 2025.1.2 lokaal te installeren en voordat je tot installatie overgaateerst die regel aan te passen.

sjampeter, 2025-01-19:
installatie gedaan met aanpassing regel sqlalchemy. krijg heel de versie niet eens meer aan gang. begint zolangzamerhand een hoofpijndossier te worden, amai.

onderstaande log uit dao 2025.1.2 met aangepast sqlalchemy : 

> NotADirectoryError: [Errno 20] Not a directory: 'app/static/data/options.json'
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data doesn't exist, made
ln: failed to create symbolic link 'app/static/data': File exists
Setting up watches.
Watches established.
Failed to read config file: gunicorn_config.py
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/gunicorn/app/base.py", line 110, in get_config_from_filename
    spec.loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/root/dao/webserver/gunicorn_config.py", line 5, in <module>
    port = get_config(app_datapath + "options.json", ["dashboard", "port"], 5000)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 124, in get_config
    config = Config(file_name=file_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 23, in __init__
    self.options = self.parse(file_name)
                   ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_config.py", line 15, in parse
    with open(file_name, "r") as file_json:
         ^^^^^^^^^^^^^^^^^^^^
NotADirectoryError: [Errno 20] Not a directory: 'app/static/data/options.json'

sjampeter, 2025-01-24:
waarachtig enige witte rook. spul draait nu op de nieuwste versie met sqlite als database. 
wat zou het voordeel nog zijn om terug te gaan naar de mariadb? 
hier draait hass overigens wel op..

waar het aan gelegen heeft? geen idee
of het morgen ook nog draait? geen idee.

keep you posted

corneel27, 2025-01-26:
Hopelijk blijft het nu goed gaan!

sjampeter, 2025-01-30:
updaten naar de volgende versie zonder problemen, echter wel met sqlite database.

heb de day ahaed database verwijderd en nogmaals geprobeerd terug te migreren naar mariadb.

[image]
deze dus.
krijg het niet voor elkaar.
steeds onderstaande melding in de log van mariadb 
lijkt of ik geen toegang meer krijg oid.

[image]

vreemd allemaal hoor.

corneel27, 2025-02-02:
Als het met Home Assistant wel goed gaat kan ik nog maar een oorzaak bedenken: heeft de user "day_ahead" wel genoeg rechten op de database "day_ahead"?

sjampeter, 2025-02-03:
hier had ik inderdaad ook al wat onderzoek naar gedaan, echter zonder resultaat. ik heb een nieuwe ha_api token gemaakt, echter zonder resultaat. overigens was de oude volgens mij nog steeds actief maar goed. het proberen waard.
verders hebben we onderstaande screenshot nog.
de database had ik al eens verwijderd en opnieuw aangemaakt,  echter zonder resultaat.

[image]

corneel27, 2025-02-03:
Ik zie nu dat ie naar 172.30.33.9 wil connecten.
Volgens kan dat alleen met de user "service" (een soort van interne ha tunnel).
Je zou eens kunnen proberen met de volgende gegevens in te vullen in options.json.
Bijvoorbeeld:
```
  "database da": {
    "engine": "mysql",
    "server": "192.168.178.94",
    "port": 3306,
    "database": "day_ahead",
    "username": "day_ahead",
    "password": "!secret db_da_password"
  },
```
En dan het ip-adres van je ha machine bij "server"

sjampeter, 2025-02-03:
hmm. ik draai eerst ff backup voor ik probeer.
jij geeft aan bij engine : "mysql"
dit moet toch mariadb zijn oid?

sjampeter, 2025-02-03:
[image]
lijkt dan toch iets in mariadb te zijn? kan ik daar per ongeluk iets veranderd hebben wat deze "tunnel" veroorzaakt?

corneel27, 2025-02-03:
Zowel mysql als mariadb gebruiken de mysql engine van sqlalchemy.
In het configuratie-scherm van mariadb heb ik ook alleen homeassistant staan met username homeassistant.
Dat staat waarschijnlijk exact hetzelfde als bij jou (afgezien van password).
Heb je al geprobeerd met die uitgebreide setting?

sjampeter, 2025-02-04:
aha, dat had ik zo even niet door. duidelijk.
ik heb bovenstaande instellingen voor database da overgenomen met het juiste ip adres.
tot nu toe draait ie wel, echter krijg ik in het log nog van MariaDB wederom nu dezelfde foutmelding.

[image]

als ik de database ha wil omzetten terug naar mariadb loopt de boel wel weer vast.
onderstaand de instelling welke nu werkt (en volgens mij weer de database in mariadb gevuld wordt, correct me if i'm wrong)

[image]

corneel27, 2025-02-04:
Je zou bij de ha-db dezelfde truc kunnen uithalen als bij de da-db:
```
  "database ha": {
    "engine" : "mysql",
    "server": "192.168.178.36",
    "port": 3306,
    "database": "homeassistant",
    "username": "homeassistant",
    "password": "!secret db_ha_password"
  },
```

sjampeter, 2025-02-05:
aha, daar gaat er wel iets mis. 
met deze input loopt ie weer vast.
nu de oorzaak nog.
er is een foutmelding bijgekomen in het log van mariaDB.

[image]

closed 2025-06-16 by sjampeter