corneel27/day-ahead #194: Entsoe prices no longer correctly retrieved

soeter01, 2025-01-19:
The day ahead prices from entsoe are no longer correctly retrieved. Tomorrow is empty.. no error messages. Entsoe integration suffers from same problem.

Something to do with entsoe api perhaps?

corneel27, 2025-01-20:
I confirm your experience with entsoe.
I have upgraded the used python package but still the same error.
I keep investigate the error.
In the meantime you can use "nordpool" to get the current prices

corneel27, 2025-01-20:
The problem lies in the entsoe platform not in the python-software.
They moved to a new platform (https://newtransparency.entsoe.eu/market/energyPrices). On that site there also no prices for the region NL (but for DE they are present).

closed 2025-01-21 by corneel27

corneel27, 2025-01-26:
Ik heb versie 2025.1.3 in de addon-branche gezet en een PR naar main gedaan.
En wil je de PR reviewen (ik heb het ook van @tonvanboven  gevraagd)?

soeter01, 2025-01-27:
Hallo Cees, doe ik graag. Uiterlijk morgen middag.

Zag nog wel vreemd gedrag bij inplannen machine:
- ingepland door dao om om 3.00 te starten vandaag. Maar bij de berekening
om 3.00 wordt de planning doorgeschoven naar een later tijdstip niet
overeenkomstig met de eis. Zie bijgaande foto’s van de berekeningen. Zegt
je dat wat?

Groet Ronald