corneel27/day-ahead #195: "entity set power feedin" geinverteerde waarde

hvdort, 2025-01-25:
De waarde van "entity set power feedin" is juist tegengesteld aan wat mijn omvormer via modbus wil hebben. Ik ben nog niet zo goed thuis in HA. Wat is de beste methode om de waarde te vermenigvuldigen met -1 om zo de gewenste waarde te krijgen?

corneel27, 2025-01-25:
Waarschijnlijk heb je een (of meer) Sessy('s).
Zie voor instellingen de discussion van @mgroen63:
https://github.com/corneel27/day-ahead/discussions/154

Hij gebruikt onderstaande automatisering:
```
alias: DAO Set Power Setpoint dvwg f1
description: ""
triggers:
  - trigger: state
    entity_id:
      - input_number.dao_sessy_dvw9_power_setpoint
conditions: []
actions:
  - action: number.set_value
    data_template:
      entity_id: number.sessy_dvw9_power_setpoint
      value: |
        {{ (states('input_number.dao_sessy_dvw9_power_setpoint') | float) * -1}}
mode: single
```
Kom je hiermee verder?

hvdort, 2025-01-27:
Nee, ik heb sinds kort een BlueSun hybride omvormer van 15kW met daarbij 20kWh aan accucapaciteit (bruto). 
De koppeling via modbus doe ik met een lilygo_t_can module met esphome erop.

Ik heb intussen nog een andere oplossing gevonden (denk ik). Dat is bij de instellingen van de modbus in esphome:
number:
- platform: modbus_controller
  modbus_controller_id: Bluesun_inverter
  name: "ChgPowerSet"
  register_type: holding
  address: 2502    ## address of the register inside the Modbus slave device
  unit_of_measurement: "W"
  value_type: S_DWORD
  multiply: -1  

De "multiply: -1" lijkt te doen wat het moet doen. Echter, er is ook nog een ander probleem:

Ik krijg bij het schrijven naar deze variabele een foutmelding:
2025-01-27 08:17:37 fout: Fout bij schrijven naar number.lilygo_t_can_chgpowerset, waarde -1080
2025-01-27 08:17:37 fout: File: /root/dao/webserver/../prog/day_ahead.py, line 3422, in <module>
2025-01-27 08:17:38 fout: File: /root/dao/webserver/../prog/day_ahead.py, line 3398, in main
2025-01-27 08:17:38 fout: File: /root/dao/prog/da_base.py, line 474, in run_task_function
2025-01-27 08:17:38 fout: File: /root/dao/webserver/../prog/day_ahead.py, line 2839, in calc_optimum
2025-01-27 08:17:38 fout: File: /root/dao/prog/da_base.py, line 360, in set_entity_value
2025-01-27 08:17:38 fout: File: /root/dao/prog/da_base.py, line 155, in set_value
2025-01-27 08:17:38 fout: Onverwachte fout: 
Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 2628, in calc_optimum
    self.set_entity_value(
  File "/root/dao/prog/da_base.py", line 360, in set_entity_value
    self.set_value(entity_id, value)
  File "/root/dao/prog/da_base.py", line 148, in set_value
    raise ValueError
ValueError

De waarde, in dit geval -1080, wordt wel degelijk netjes geschreven in de betreffende entiteit:
Logboek: lilygo t-can ChgPowerSet gewijzigd in -1.080,0 getriggerd door actie Nummer: Stel in

Maar dus toch een foutmelding/crash. Kan het zijn dat het een "type" probleem is? De entiteit wordt automatisch aangemaakt door esphome en is van het type S_DWORD (signed 32 bit integer) terwijl een handmatig aangemaakte entity volgens mij een float is?

De oplossing via een automatisering lost waarschijnlijk ook dit probleem op, maar het is natuurlijk een minder fraaie oplossing ;-)

corneel27, 2025-01-27:
Modbus staat meestal niet toe om negatieve waarden te schrijven.
Dan moet het getal worden omgezet door er (in jouw geval) 2**31 bij op te tellen.
Onderstaand als voorbeeld een automation die ik gebruik om naar een 16 bit modbus register van Victron te schrijven:
```
alias: set setpoint voeding op de gx
description: ""
mode: single
triggers:
  - entity_id:
      - input_number.feedin_grid
    trigger: state
conditions: []
actions:
  - data:
      hub: victron
      unit: 100
      address: 2700
      value: >-
        {% set new_value = states.input_number.feedin_grid.state| int(0) %} 
        {% if new_value < 0 %} {% set new_value = new_value + 65536 %} {% endif %}
        {{ new_value }}
    action: modbus.write_register
```
Kun je hier wat mee?

corneel27, 2025-01-27:
Wat ook nog belangrijk is:
Als je een number-helper aanmaakt in HA dan moet je daarbij de grenzen opgeven waarbinnen zijn waarden moeten liggen.
Is dat bij jou van toepassing en staan deze grenzen ruim genoeg?

hvdort, 2025-01-27:
In dit geval kan ik wel een negatieve waarde schrijven via modbus. Ik zei al: de waarde wordt daadwerkelijk geschreven (getuige de logfile), ik is zie die ook daadwerkelijk geschreven worden in de omvormer. Helaas komt DAO toch met een foutmelding.
De route via een losse helper is inderdaad ook mogelijk, maar dat leek me minder strak. Ik kan namelijk ook via de GUI zonder meer de betreffende waarde schrijven via modbus, alles zonder automation of aparte helpers. Door er in de modbus-definitie geen sensor maar een number van te maken.
In volg dus dezelfde stappen als jij, alleen was ik (door gebrek aan kennis over HA) nog niet toe aan het toepassen van automations. Ik kon tot nu toe zonder ;-)

corneel27, 2025-01-29:
Ziet er erg goed uit.
Er komt binnenkort op tweakers een nieuw topic over Day Ahead Optimizer.
Mag deze info daar gepubliceerd worden?

hvdort, 2025-01-30:
Ja natuurlijk. Altijd fijn om elkaar te helpen!

closed 2025-02-08 by hvdort