corneel27/day-ahead #199: Logboek wordt niet geladen / blijft leeg

wiljums, 2025-02-02:
Zelfs na een nieuwe installatie wordt het logboek van de addon niet geladen: 
[https://i.imgur.com/0ggcrQ2.png](url)
Vernieuwen of downloaden van de logs heeft geen effect.

DAO 2025.1.3
RPI 4B
Core 2025.1.4
Supervisor 2024.12.3

wiljums, 2025-02-02:
Via Systeem>Logboeken>Day Ahead Optimizer wordt het logboek wel getoond:

Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 90, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 41, in _assert_api_running
    raise ClientError("Home Assistant API is not running.")
hassapi.exceptions.ClientError: Home Assistant API is not running.
[2025-02-02 11:35:14 +0100] [14] [ERROR] Can't connect to ('0.0.0.0', 5000)
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-02 11:46:23 +0100] [12] [INFO] Starting gunicorn 23.0.0
[2025-02-02 11:46:23 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-02 11:46:24 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-02 11:46:25 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-02 11:46:26 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-02 11:46:27 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-02 11:46:28 +0100] [12] [ERROR] Can't connect to ('0.0.0.0', 5000)
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 90, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 41, in _assert_api_running
    raise ClientError("Home Assistant API is not running.")
hassapi.exceptions.ClientError: Home Assistant API is not running.

corneel27, 2025-02-02:
Als ik deze foutmelding ziedan staat er een fout in je instellingen.
Kun je deze kopieren en hier delen?
Dat kan (als het dashboard werkt) via "config"/"options" of als je samba hebt geinstalleerd via addon_configs /xxxx_day_ahead_opt/dao_ data/options.json

wiljums, 2025-02-02:
Inmiddels een aantal keer de addon opnieuw geïnstalleerd (inclusief herstart systeem), waarbij automatisch de map in addon_configs wordt verwijderd. Options.json is dus de default:

{
  "homeassistant": { },
  "database ha": {
    "engine": "sqlite",
    "database": "home-assistant_v2.db",
    "db_path": "/homeassistant"
  },
  "database da": {
    "engine": "sqlite",
    "db_path": "../data"
  },
  "meteoserver-key": "!secret meteoserver-key",
  "prices": {
    "source day ahead": "nordpool",
    "regular high": 0.50,
    "regular low": 0.40,
    "switch to low": 23,
    "energy taxes delivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.10880,
      "2025-01-01": 0.10154
    },
    "energy taxes redelivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.10880,
      "2025-01-01": 0.10154
    },
    "cost supplier delivery": {
      "2022-01-01": 0.002,
      "2023-03-01": 0.018,
      "2024-04-01": 0.0175,
      "2024-08-01": 0.020496
    },
    "cost supplier redelivery": {
      "2022-01-01": 0.002,
      "2023-03-01": 0.018,
      "2024-04-01": 0.0175,
      "2024-08-01": 0.020496
    },
    "vat": {
      "2022-01-01": 21,
      "2022-07-01": 9,
      "2023-01-01": 21
    },
    "last invoice": "2022-09-01",
    "tax refund": "True"
  },
  "logging level" : "info",
  "use_calc_baseload": "False",
  "baseload calc periode": 56,
  "baseload": [
    0.14,
    0.38,
    0.26,
    0.42,
    0.15,
    0.12,
    0.13,
    0.15,
    0.23,
    0.26,
    0.31,
    0.32,
    0.31,
    0.23,
    0.26,
    0.21,
    0.21,
    0.54,
    0.26,
    0.26,
    0.22,
    0.19,
    0.18,
    0.16
  ],
  "graphical backend": "",
  "graphics": {
    "style": "Solarize_Light2",
    "show" : "true",
    "prices delivery": "True",
    "prices redelivery": "True",
    "average delivery": "True"
  },
  "strategy": "minimize cost",
  "notifications": {
  },
  "grid": {
    "max_power": 17
  },
  "history": {
    "save days": 7
  },
  "dashboard": {
    "port": 5000
  },
  "boiler": {
    "boiler present": "False",
    "entity actual temp.": "sensor.boiler_gemeten",
    "entity setpoint": "sensor.boiler_ingesteld",
    "entity hysterese": "sensor.hysterese_hot_water",
    "cop": 2.9,
    "cooling rate": 0.4,
    "volume": 180,
    "heating allowed below": 44,
    "elec. power": 1500,
    "activate service": "press",
    "activate entity": "input_button.hw_trigger"
  },
  "heating": {
    "heater present": "False",
    "degree days factor": 3.6,
    "stages": [
      {
        "max_power": 225,
        "cop": 7.1
      },
      {
        "max_power": 300,
        "cop": 7.0
      },
      {
        "max_power": 400,
        "cop": 6.5
      },
      {
        "max_power": 500,
        "cop": 6.0
      },
      {
        "max_power": 600,
        "cop": 5.5
      },
      {
        "max_power": 750,
        "cop": 5.0
      },
      {
        "max_power": 1000,
        "cop": 4.5
      },
      {
        "max_power": 1250,
        "cop": 4.0
      }
    ],
    "entity adjust heating curve": "input_number.stooklijn_verschuiving_day_ahead",
    "adjustment factor": 0.04
  },
  "battery": [ ],
  "solar": [ ],
  "electric vehicle": [ ],
  "machines" : [ ],
  "tibber": {
    "api_token": "!secret tibber_api_token"
  },
  "report": {
    "entities grid consumption": [
      "sensor.grid_consumption_low",
      "sensor.grid_consumption_high"
    ],
    "entities grid production": [
      "sensor.grid_production_low",
      "sensor.grid_production_high"
    ],
    "entities solar production ac": [
      "sensor.solaredge_woning_ac_energy_kwh"
    ],
    "entities solar production dc": [],
    "entities ev consumption" : ["sensor.laadpunt_total_energy"],
    "entities wp consumption" : [],
    "entities boiler consumption": [],
    "entities battery consumption": ["sensor.ess_grid_consumption"],
    "entities battery production": ["sensor.ess_grid_production"]
  },
  "scheduler": {
    "active": "false",
    "0430": "get_meteo_data",
    "1030": "get_meteo_data",
    "1630": "get_meteo_data",
    "2230": "get_meteo_data",
    "1255": "get_day_ahead_prices",
    "1355": "get_day_ahead_prices",
    "1455": "get_day_ahead_prices",
    "1554": "get_day_ahead_prices",
    "1655": "get_day_ahead_prices",
    "xx00": "calc_optimum",
    "2359": "clean_data"
  }
}

corneel27, 2025-02-02:
Met jouw options.json is niks mis!
Maar toch wijst de foutmelding in de logging erop dat DAO geen contact krijgt met Home Assistant.
Draai jij Home Assistant gewoon op de RPi4 met HAOS als operating system of draait HA in een aparte container met Linux als operating system?

wiljums, 2025-02-02:
Ik draai HA Supervised op Debian Bookworm. Na de 1e installatie deed dao het wel, maar nu krijg ik na iedere herinstallatie meteen de error 502: bad gateway. 
Kan het zijn dat de addon niet volledig deinstalleert en er nog iets in de weg zit van een eerdere installatie?

corneel27, 2025-02-02:
Als je HA supervised draait dan kun je het beste de gegevens van HA invullen. Bijvoorbeeld:
```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.94",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
Bij ip-adres het adres waarop jij met een browser baar ha gaat
Het token genereer je in ha en zet je het beste in secrets.json (zie DOCS.md)

wiljums, 2025-02-02:
Ok, ga ik proberen. Aangezien de web ui van dao niet eens opstart, zal ik het direct in de options.json invoeren. Kijken of het dan wel lukt.

wiljums, 2025-02-03:
Ik blijf deze error krijgen:
https://i.imgur.com/gOomg1C.png 

En dit in het logboek:
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-03 09:32:58 +0100] [12] [INFO] Starting gunicorn 23.0.0
[2025-02-03 09:32:58 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 09:32:59 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 09:33:00 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 09:33:01 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 09:33:02 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 90, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 41, in _assert_api_running
    raise ClientError("Home Assistant API is not running.")
hassapi.exceptions.ClientError: Home Assistant API is not running.
[2025-02-03 09:33:03 +0100] [12] [ERROR] Can't connect to ('0.0.0.0', 5000)

Aanpassen van de poort naar bijv. 5025 heeft blijkbaar geen effect, althans in de log staat nog steeds 5000.

corneel27, 2025-02-03:
Het lijkt erop dat er twee fouten zijn:
1. er kan geen verbinding worden gemaakt met Home Assistant
2. en de poort (5000) is niet bereikbaar.
Laten we eerst proberen de eerste fout te tackelen.

**Geen verbinding met HA**
DAO maakt gebruikt van de api van Home Assistant (zie ook https://developers.home-assistant.io/docs/api/rest/)
Je kunt in een terminal checken of je contact krijgt met deze api:
``
curl   -H "Authorization: Bearer eyJ0e....PQk"   -H "Content-Type: application/json" http://192.168.178.36:8123/api/states/sun.sun
``
Je moet
-  `eyJ0e....PQk` vervangen door je HA-token dat je hebt ingevuld bij secrets (zonder aanhalingstekens). Dat is een hele lange string van tekens.
- `192.168.78.36` vervangen door het ipadres van jouw ha-machine.

Als het goed is krijg je dan dit resultaat:
``
{"entity_id":"sun.sun","state":"above_horizon","attributes":{"next_dawn":"2025-02-04T06:33:02.472988+00:00","next_dusk":"2025-02-03T17:05:57.283476+00:00","next_midnight":"2025-02-03T23:50:01+00:00","next_noon":"2025-02-04T11:49:57+00:00","next_rising":"2025-02-04T07:09:39.899660+00:00","next_setting":"2025-02-03T16:29:08.961933+00:00","elevation":9.15,"azimuth":228.21,"rising":false,"friendly_name":"Sun"},"last_changed":"2025-02-03T07:11:19.222495+00:00","last_reported":"2025-02-03T15:10:36.967888+00:00","last_updated":"2025-02-03T15:10:21.676852+00:00","context":{"id":"01JK65SKFCCSTYG9AP8HDXQ50Y","parent_id":null,"user_id":null}}
``

Wat is bij jou het resultaat?

wiljums, 2025-02-03:
Bij deze:
{"entity_id":"sun.sun","state":"below_horizon","attributes":{"next_dawn":"2025-02-04T06:33:38.483784+00:00","next_dusk":"2025-02-03T17:07:46.848056+00:00","next_midnight":"2025-02-03T23:51:13+00:00","next_noon":"2025-02-04T11:51:09+00:00","                      next_rising":"2025-02-04T07:09:56.602177+00:00","next_setting":"2025-02-04T16:33:07.753280+00:00","elevation":-2.51,"azimuth":246.78,"rising":false,"friendly_name":"Sun"},"last_changed":"2025-02-03T16:35:18.146644+00:00","last_reported":"2025-02-03T16:43:26.480133+00:00","last_updated":"2025-02-03T16:43:18.147691+00:00","context":{"id":"01JK6B3S83DD4GE42C2VTC8REG","parent_id":null,"user_id":null}}

corneel27, 2025-02-03:
OK, volgende stap.
Omdat je ha supervised draait is het belangrijk dat de goede poort open staat.
Als je met ssh inlogt op je debian-os kun je kijken welke poorten open staan met dit commando:
`netstat -lntu`
Staat poort 5000 daartussen:
```
tcp        0      0 127.0.0.1:11984         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:5000            0.0.0.0:*               LISTEN     
tcp        0      0 192.168.178.36:40000    0.0.0.0:*               LISTEN     
```
Met de state **LISTEN**?

wiljums, 2025-02-03:
Daar staat poort 5000 niet tussen (en dus ook niet met de status LISTEN)... Hoe krijg ik de poort open? Of een andere poort gebruiken?

corneel27, 2025-02-03:
Debian werkt met "uncomplicated firewall".
Dus je kunt dit commando gebruiken:
`sudo ufw allow 5000`
Zorg er wel voor dat je bij de configuratie van de addon DAO poort 5000 niet redirect naar een andere poort.
Dus in options.json staat 
```
  "dashboard": {
    "port": 5000
  },
```
En bij de configuratie staat:

[image]

wiljums, 2025-02-03:
Response: sudo: ufw: command not found

options.json en poort configuratie kloppen

wiljums, 2025-02-03:
Na: sudo apt install ufw

is de response:
sudo ufw allow 5000
Rules updated
Rules updated (v6)

wiljums, 2025-02-03:
Nog geen verschil:

=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-03 23:07:42 +0100] [12] [INFO] Starting gunicorn 23.0.0
[2025-02-03 23:07:42 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 23:07:43 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 23:07:44 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 23:07:45 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
[2025-02-03 23:07:46 +0100] [12] [ERROR] connection to ('0.0.0.0', 5000) failed: [Errno 13] Permission denied
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 90, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 41, in _assert_api_running
    raise ClientError("Home Assistant API is not running.")
hassapi.exceptions.ClientError: Home Assistant API is not running.
[2025-02-03 23:07:47 +0100] [12] [ERROR] Can't connect to ('0.0.0.0', 5000)

corneel27, 2025-02-03:
Heb je de debian-machine al een keer opnieuw opgestart met de nieuwe instellingen?

corneel27, 2025-02-03:
En dan ben ik benieuwd of je ook een fout krijgt met dit adres in de adresbalk (dus buiten ha om):
`http://192.168.178.36:5000/`
waarbij je `192.168.178.36` vervangt door het ip-adres van de debian-machine.

wiljums, 2025-02-03:
na herstart helaas hetzelfde resultaat, ip:5000/  is via browser niet te bereiken (deze site is niet bereikbaar)

corneel27, 2025-02-04:
Ik vrees dat ik je nu niet meer kan helpen.
Ik heb geen ervaring met HA-supervised.
Ik heb ook nog gezocht op internet en via chatgpt:
https://chatgpt.com/share/67a1e5fb-f050-800b-8ed3-d095a732b334
Misschien staat daar nog iets tussen wat jou kan helpen.

ThijsCJ, 2025-10-08:
2025-10-08 09:44:12.523 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host X.X.X.X:5000 ssl:default [Connect call failed (' X.X.X.X', 5000)]

Ik heb hetzelfde probleem en hij lijkt niet via ingress connectie te kunnen maken.