corneel27/day-ahead #200: fout:root:Check your settings for Home Assitant database

robbinonline, 2025-02-12:
Ik krijg de volgende foutmelding. Heb de add-on geïnstalleerd en maak gebruik van de standaard database van HA (sqlite). 
Volgens de handleiding zou de volgende configuratie voldoende moeten zijn om te kunnen starten:

`
{
  "homeassistant": { },

  "database ha": {
    "engine": "sqlite",
    "database": "homeassistant_v2.db",
    "db_path": "/homeassistant"
  },

 "database da": {
    "engine": "sqlite",
    "db_path": "../data"
  },

`

Maar de foutmelding die ik krijg blijft:
fout:root:Check your settings for Home Assitant database

corneel27, 2025-02-12:
Je hebt een klein type-foutje gemaakt:
```
"database ha": {
    "engine": "sqlite",
    "database": "home-assistant_v2.db",
    "db_path": "/homeassistant"
  },
'''
Zoek de verschillen ....

robbinonline, 2025-02-12:
Wat een rookie mistake. Maar voor mijn verdediging de code komt rechtstreeks uit de handleiding:

https://github.com/corneel27/day-ahead/blob/main/dao/DOCS.md gelijk onder het de kop: 
Toegang tot de database van homeassistant

alleen met zelfs met deze aanpassing werkt het nog steeds niet. Blijf dezelfde error houden.

robbinonline, 2025-02-12:
Ja in de config map staat de database: home-assistant_v2.db
En ook de add-on meerdere keren herstart, zelfs de add-on vanavond nog even verwijderd en geherinstalleerd.

 Ook als ik options.json op een andere manier benaderen zie ik volgens mij de juiste configuratie. 

```
  "database ha": {
    "engine": "sqlite",
    "database": "home-assistant_v2.db",
    "db_path": "/homeassistant"
  },
```

corneel27, 2025-02-12:
Ik heb op mijn linux laptop VirtualBox geinstalleerd.
En in een van die boxen heb ik home-assistant geinstalleerd met sqlite en DAO als addon.
Ik heb die box weer even opgestart en daar staat toch echt:
```
  "database ha": {
    "engine": "sqlite",
    "database": "home-assistant_v2.db",
    "db_path": "/homeassistant"
  },
```

[image]

Je hebt in HA geen regel met `recorder:`staan?

Op wat voor machine draait jij ha (met proxmox heb ik al vaker gedonder gehad)?
Draait je met HAOS of draai je "supervised"?

robbinonline, 2025-02-12:
Laatste versie van HAOS 2025.2.(2) als virtual machine op een Synology NAS.
Ja heb wel een recorder regel staan:  purge_keep_days: 3 om de grootte van de database te berperken.

corneel27, 2025-02-13:
Er is een addon waarmee je via een webpagina kunt browsen in je sqlite database.
Die werkt ongeveer hetzelfde als DAO.
Zou je die addon eens willen testen?
Het gaat om deze addon:
https://github.com/hassio-addons/addon-sqlite-web
Ik ben benieuwd of je daarmee wel toegang krijgt tot de database.

robbinonline, 2025-02-13:
Ja die werkt prima

[image]

robbinonline, 2025-02-13:
Filename komt ook overeen met de instellingen. Heb ngo een paar varianten geprobeerd zoals

```"database": "/homeassistant/home-assistant_v2.db"
of
 "db_path": "/homeassistant/"
```

Maar nog steeds de error:
fout:root:Check your settings for Home Assitant database

corneel27, 2025-02-13:
Er moet ergens iets fout gaan, maar heb nog geen idee waar.
Kun je in de settings het logging level op "debug" zetten en dan de logging (op de logpagina van de addon, dus via Instellingen/Add-ons enz) ) hier delen?

corneel27, 2025-02-13:
Ik heb ook nog even in de code van de  addon-sqlite-web gekeken, maar daar staat ook:
`database="/homeassistant/home-assistant_v2.db"`
Ik heb wel een extra check ingebouwd of de database bestaat.
Die zet "/homeassistant" eerst om naar een absoluut pad.
Misschien dat het daar bij jou fout gaat, maar dat moet blijken uit de logging met logging-level op **debug**.

robbinonline, 2025-02-13:
```
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-13 15:30:37 +0100] [9] [INFO] Starting gunicorn 23.0.0
[2025-02-13 15:30:37 +0100] [9] [INFO] Listening at: http://0.0.0.0:5000 (9)
[2025-02-13 15:30:37 +0100] [9] [INFO] Using worker: sync
[2025-02-13 15:30:37 +0100] [15] [INFO] Booting worker with pid: 15
[2025-02-13 15:30:37 +0100] [16] [INFO] Booting worker with pid: 16
fout:root:Check your settings for Home Assitant database
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-13 15:32:44 +0100] [10] [INFO] Starting gunicorn 23.0.0
[2025-02-13 15:32:44 +0100] [10] [INFO] Listening at: http://0.0.0.0:5000 (10)
[2025-02-13 15:32:44 +0100] [10] [INFO] Using worker: sync
[2025-02-13 15:32:44 +0100] [16] [INFO] Booting worker with pid: 16
[2025-02-13 15:32:44 +0100] [17] [INFO] Booting worker with pid: 17
fout:root:Check your settings for Home Assitant database
../data/options.json MODIFY 
Setting up watches.
Watches established.
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 90, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.BadGateway: 502 status code returned from http://supervisor/core/api/
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-13 15:37:41 +0100] [10] [INFO] Starting gunicorn 23.0.0
[2025-02-13 15:37:41 +0100] [10] [INFO] Listening at: http://0.0.0.0:5000 (10)
[2025-02-13 15:37:41 +0100] [10] [INFO] Using worker: sync
[2025-02-13 15:37:41 +0100] [16] [INFO] Booting worker with pid: 16
[2025-02-13 15:37:41 +0100] [17] [INFO] Booting worker with pid: 17
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-02-13 15:39:06 +0100] [9] [INFO] Starting gunicorn 23.0.0
[2025-02-13 15:39:06 +0100] [9] [INFO] Listening at: http://0.0.0.0:5000 (9)
[2025-02-13 15:39:06 +0100] [9] [INFO] Using worker: sync
[2025-02-13 15:39:06 +0100] [15] [INFO] Booting worker with pid: 15
[2025-02-13 15:39:06 +0100] [16] [INFO] Booting worker with pid: 16
debug:root:Dialect: sqlite, database: day_ahead.db, db_path: ../data
debug:root:db_url: sqlite://///root/dao/data/day_ahead.db
debug:root:Dialect: sqlite, database: home-assistant_v2.db, db_path: /homeassistant
debug:root:db_url: sqlite://///homeassistant/home-assistant_v2.db
fout:root:Check your settings for Home Assitant database
debug:root:Connection status Pool size: 5  Connections in pool: 0 Current Overflow: -5 Current Checked out connections: 0 at line 140 in /root/dao/prog/da_base.py
```

corneel27, 2025-02-13:
Kun jij met ssh een commandline krijgen op je ha-machine?
Dan kun je zo een shell krijgen tot de dao-addon.
Dat werkt als volgt:
- log in met ssh, dan krijg je uiteindelijk een schrm dat er als volgt uitziet:
[image]
- Je vraagt alle draaiende addons op met `docker ps`. Dan zie je daar de addon van dao tussen staan:
[image]
- je kopieert het laatste stuk van die regel `addon_325af97a_day_ahead_opt` in de volgende opdracht: 
`docker exec -it addon_325af97a_day_ahead_opt bash` Met name de code `325af97a` zal bij jou anders zijn.
- Dan krijg je een prompt in de dao-addon: `root@325af97a-day-ahead-opt:~/dao/prog# `
- Geef dan het commando `python3`, daarmee krijg je een interactieve python shell
- geeft daar het commando: `import sqlalchemy_utils`
- vervolgens: `db_url = "sqlite://///homeassistant/home-assistant_v2.db"`
- vervolgens: `sqlalchemy_utils.database_exists(db_url)`
- Je zou dan dit scherm moeten hebben dat eindigt met `True`:
[image]

Ik ben benieuwd.

Je eindigt / sluit af met:
- python shell verlaten : `exit()`
- addon-shel verlaten: `exit` 
- ha-shell verlaten: `exit` 
[image]

corneel27, 2025-02-14:
Ik ben juist blij dat jij het geduld hebt om samen met mij dit op te lossen.
Daar hebben anderen met eenzelfde setup ook weer baat bij.

Ik heb nog een paar check's.
Zou je die willen uitproberen?

Ga weer naar de python-shell van de dao-addon:
Geef het commando `import os.path`
en daarna: `os.path.isfile("/homeassistant/home-assistant_v2.db")`
Dan zou je weer `True` moeten krijgen:
[image]

En dan nog een tweede check, ook weer in de python-shell, de volgende commando's
- `from sqlalchemy import create_engine, select`
- `db_name = "home-assistant_v2.db"`
- `db_path = "/homeassistant"`
- `engine = create_engine(f"sqlite:///{db_path}/{db_name}")`
- `connection = engine.connect()`
- connection.scalar(select(1))`
Daar moet **1** uitkomen.
[image]
Ik heb zelf een volgorde fout gemaakt maar hierboven staan de commando's in de goede volgorde.

Ik ben weer benieuwd!

robbinonline, 2025-02-14:
Beide geen succes

```
root@94e386ba-day-ahead-opt:~/dao/prog# python3
Python 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os.path
>>> os.path.isfile("/homeassistant/home-assistant_v2.db")
False
>>> 
```

```
root@94e386ba-day-ahead-opt:~/dao/prog# python3
Python 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from sqlalchemy import create_engine, select
>>> db_name = "home-assistant_v2.db"
>>> db_path = "/homeassistant"
>>> engine = create_engine(f"sqlite:///{db_path}/{db_name}")
>>> connection = engine.connect()
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 146, in __init__
    self._dbapi_connection = engine.raw_connection()
                             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 3298, in raw_connection
    return self.pool.connect()
           ^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 449, in connect
    return _ConnectionFairy._checkout(self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 1263, in _checkout
    fairy = _ConnectionRecord.checkout(pool)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 712, in checkout
    rec = pool._do_get()
          ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py", line 179, in _do_get
    with util.safe_reraise():
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py", line 177, in _do_get
    return self._create_connection()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 390, in _create_connection
    return _ConnectionRecord(self)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 674, in __init__
    self.__connect()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 900, in __connect
    with util.safe_reraise():
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 896, in __connect
    self.dbapi_connection = connection = pool._invoke_creator(self)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/create.py", line 646, in connect
    return dialect.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 622, in connect
    return self.loaded_dbapi.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlite3.OperationalError: unable to open database file

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 3274, in connect
    return self._connection_cls(self)
[... 24 lines cut ...]
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/impl.py", line 177, in _do_get
    return self._create_connection()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 390, in _create_connection
    return _ConnectionRecord(self)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 674, in __init__
    self.__connect()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 900, in __connect
    with util.safe_reraise():
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/util/langhelpers.py", line 146, in __exit__
    raise exc_value.with_traceback(exc_tb)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/pool/base.py", line 896, in __connect
    self.dbapi_connection = connection = pool._invoke_creator(self)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/create.py", line 646, in connect
    return dialect.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 622, in connect
    return self.loaded_dbapi.connect(*cargs, **cparams)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) unable to open database file
(Background on this error at: https://sqlalche.me/e/20/e3q8)
>>> 
```

robbinonline, 2025-02-14:
Het werkt, ik weet alleen niet of dit de juiste manier van oplossen is. Vond het vreemd dat de docker root geen toegang had tot de homeassistant map. Dus heb de permissions gerest en nu heeft de addon wel toegang tot de database. 

```
sudo chmod 755 /homeassistant
sudo chown root:root /homeassistant
```

Ga nog even kijken of het ook blijft werken na een herstart ..

corneel27, 2025-02-14:
Als je inlogt op de dao-addon.
Je gaat naar "/" door `cd /`
En je geeft `ls -la`
Dan krijg je o.a. homeassitant te zien:
[image]

robbinonline, 2025-02-14:
Ook na een volledige systeem herstart blijft het werken. Leek erop dat de permission een beetje in de war waren.

corneel27, 2025-02-14:
Dus probleem is hiermee opgelost?
De addon draait nu zonder opstartproblemen?

robbinonline, 2025-02-14:
Ja daar lijkt het voor nu wel op, ga vanmiddag verder met de rest van de configuratie. Even zien of er dan niets geks meer opkomt maar anders kan ik hierna deze thread sluiten.

corneel27, 2025-02-14:
Blijven nog twee vragen:

1. is dit altijd zo bij een nieuwe installatie bij een setup zoals die van jou?
2. en zo ja hoe los je dat op?

Vraag 1 zou jij kunnen beantwoorden door tijdelijk een tweede dao addon vanaf een andere branche te installeren.
Zou je dat willen proberen?
Dat gaat hetzelfde als de installatie van de eerste addon alleen geef je dan als repository op:
`https://github.com/corneel27/day-ahead/#addon`
Dat is de zogenaamde addon-branche, die gebruik ik (samen met anderen) om een nieuwe develop te testen en te reviewen.
Als die nu wel direct goed gaat is het probleem toch eenmalig en hoef ik niks meer te doen.
Na afloop van deze test kun je die tweede addon weer verwijderen.

robbinonline, 2025-02-14:
Ik heb het nog nooit eerder gehad maar ik draai deze home assistant al bijna 8 jaar dus het kan zomaar zijn dat er qua permissies in de loop der jaren iets is mis gegaan. Denk doordat ik nu op root niveau en niet op add-on de permissies heb gereset het niet meer te reproduceren valt. 

De tweede dao addon werkt nu ook direct

corneel27, 2025-02-14:
Fijn dat het nu allemaal goed werkt!

closed 2025-02-14 by corneel27