corneel27/day-ahead #204: Fout bij gebruik van postgresql voor dao database

balk77, 2025-03-20:
Hi, allereerst bedankt voor de ontwikkeling van deze mooie tool. Ik moet nog veel er in ontdekken.

Ik gebruik postgresql als recorder voor HA. DAO lijkt dat goed te snappen met
```
"homeassistant": { },
  "database ha": {
     "engine" : "postgresql",
     "server": "192.168.1.111",
     "port": 5432,
     "database": "homeassistant"
     "username": "homeassistant",
     "password": "!secret db_ha_password"
   }
```
Als ik de dao database ook in postgres zet, met 
```
"database da": {
   "engine" : "postgresql",
   "server": "192.168.1.111",
   "port": 5432,
   "database": "day_ahead",
   "username": "day_ahead",
   "password": "!secret db_da_password"
  }
```
dan krijg ik bij het ophalen van de day ahead prijzen
[image]

de volgende foutmelding. Wanneer ik sqlite gebruik krijg ik die melding niet. Heb jij enig idee hoe dit kan komen?
```
2025-03-20 06:58:12 info: Day Ahead Optimalisering versie: 2025.2.0
2025-03-20 06:58:12 info: Day Ahead Optimalisering gestart op: 20-03-2025 06:58:12
2025-03-20 06:58:12 info: Day Ahead Optimalisatie gestart: 20-03-2025 06:58:12 taak: get_day_ahead_prices
2025-03-20 06:58:12 info: Day ahead prijzen van Nordpool:
 [ { 'end': datetime.datetime(2025, 3, 20, 0, 0, tzinfo=tzutc()),
    'start': datetime.datetime(2025, 3, 19, 23, 0, tzinfo=tzutc()),
    'value': 108.74},
  <snip, veel data>,
  { 'end': datetime.datetime(2025, 3, 20, 22, 0, tzinfo=tzutc()),
    'start': datetime.datetime(2025, 3, 20, 21, 0, tzinfo=tzutc()),
    'value': 110.71},
  { 'end': datetime.datetime(2025, 3, 20, 23, 0, tzinfo=tzutc()),
    'start': datetime.datetime(2025, 3, 20, 22, 0, tzinfo=tzutc()),
    'value': 106.64}]
2025-03-20 06:58:12 fout: Er is een fout opgetreden, zie de fout-tracering
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.InvalidTextRepresentation: invalid input syntax for type bigint: "1742425200.0"
LINE 3: WHERE values.variabel = 3 AND values.time = '1742425200.0'
                                                    ^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/dao/prog/da_base.py", line 478, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 269, in get_day_ahead_prices
    self.prices.get_prices(
  File "/root/dao/prog/da_prices.py", line 177, in get_prices
    self.db_da.savedata(df_db)
  File "/root/dao/prog/db_manager.py", line 262, in savedata
    value_result = connection.execute(select_value).first()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1416, in execute
    return meth(
           ^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 523, in _execute_on_connection
    return connection._execute_clauseelement(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1638, in _execute_clauseelement
    ret = self._execute_context(
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1843, in _execute_context
    return self._exec_single_context(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1983, in _exec_single_context
    self._handle_dbapi_exception(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.DataError: (psycopg2.errors.InvalidTextRepresentation) invalid input syntax for type bigint: "1742425200.0"
LINE 3: WHERE values.variabel = 3 AND values.time = '1742425200.0'
                                                    ^

[SQL: SELECT values.id 
FROM values 
WHERE values.variabel = %(variabel_1)s AND values.time = %(time_1)s]
[parameters: {'variabel_1': 3, 'time_1': '1742425200.0'}]
(Background on this error at: https://sqlalche.me/e/20/9h9h)
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.InvalidTextRepresentation: invalid input syntax for type bigint: "1742425200.0"
LINE 3: WHERE values.variabel = 3 AND values.time = '1742425200.0'
                                                    ^

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3448, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3430, in main
    da_calc.run_task_function("prices")
  File "/root/dao/prog/da_base.py", line 478, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 269, in get_day_ahead_prices
    self.prices.get_prices(
  File "/root/dao/prog/da_prices.py", line 177, in get_prices
    self.db_da.savedata(df_db)
  File "/root/dao/prog/db_manager.py", line 262, in savedata
    value_result = connection.execute(select_value).first()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1416, in execute
    return meth(
           ^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 523, in _execute_on_connection
    return connection._execute_clauseelement(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1638, in _execute_clauseelement
    ret = self._execute_context(
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1843, in _execute_context
    return self._exec_single_context(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1983, in _exec_single_context
    self._handle_dbapi_exception(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.DataError: (psycopg2.errors.InvalidTextRepresentation) invalid input syntax for type bigint: "1742425200.0"
LINE 3: WHERE values.variabel = 3 AND values.time = '1742425200.0'
                                                    ^

[SQL: SELECT values.id 
FROM values 
WHERE values.variabel = %(variabel_1)s AND values.time = %(time_1)s]
[parameters: {'variabel_1': 3, 'time_1': '1742425200.0'}]
(Background on this error at: https://sqlalche.me/e/20/9h9h)
```

corneel27, 2025-03-20:
Dank voor de foutmelding.
Ik ga hem met de eerstvolgende versie repareren! Die komt deze week nog uit..

Voor de korte termijn kun je de fout ontlopen door gebruik te maken van prijsinfo bij Entsoe en ze daar op te halen.
Je moet dan wel een api-key bij Entsoe ophalen (zie DOCS.md) en in je secrets.json zetten.

corneel27, 2025-03-20:
Ik verwacht dat morgen versie 2025.3.0 wordt gereleased.
Hij staat nu in de developers branche en wacht op review door een van de twee testers.

balk77, 2025-03-21:
Ik kan bevestigen dat 2025.3.0 (beta) de bug verholpen heeft

closed 2025-03-21 by balk77