corneel27/day-ahead #208: Internal Server Error

balk77, 2025-03-22:
Ik gebruik (nu) een sqlite database voor het DA deel. Ik heb net enkele dagen DA draaien. Als ik onder Report een periode kies die ouder is dan vorige week, dan krijg ik deze foutmelding. Ik vermoed omdat de database leeg is. Als ik "Dit jaar" of "Deze maand" kies dan geen foutmelding. "Vorige maand" wel.

[image]

De log geeft me dit

```
TypeError: can't multiply sequence by non-int of type 'float'
[2025-03-22 17:11:00,625] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 224, in menu
    return reports()
           ^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 436, in reports
    report_df = report.get_grid_data(active_period, _tot=tot)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 1420, in get_grid_data
    df_ha = self.recalc_df_ha(df_ha, interval)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 802, in recalc_df_ha
    col_3 = row.consumption * row.da_cons
            ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
TypeError: can't multiply sequence by non-int of type 'float'
```

corneel27, 2025-03-25:
Ja dat zal wel komen door de nog lege database.
Heb je Tibber dan kun je de ontbrekende data van je verbruik en je productie ophalen door eerst de begindatum van je contract voldoende vroeg te zetten (bijv 1 jan 2024) en dan via het run-commando tibber-data ophalen.
Daarna kun je de begindatum van je contract op de correcte datum zetten.

corneel27, 2025-03-25:
Het kan ook zijn dat er wel verbruiksdata zijn , maar geen prijsdata.
Aanvullen van prijsinfo gaat moeilijk met Nordpool, dat kan per keer maar 1 dag aanvullen.
Je kunt eventueel prijsdata aanvullen als bron van prijsinfo Entsoe, Tibber of easyenergy te kiezen.
Als Entsoe kiest moet je wel een api-key ophalen en in je secrets zetten (zei DOCS.md) .
Daarna kun je in het run-menu de ontbrekende periode invullen en de data ophalen

balk77, 2025-03-26:
Hi, wellicht is het eenvoudiger om de foutmelding weg te poetsen met een try except. Met de mededeling dat er geen data is voor de gevraagde periode? En de suggestie om Tibber, Entsoe oid te gebruiken?

corneel27, 2025-03-27:
Ik heb de fout kunnen reproduceren, komt door ontbrekende prijs-info.
Ik ga in een volgende versie een adekwate foutmelding geven ipv een "internal server error"

hfvdv, 2025-04-03:
Heeft onderstaande te maken met dezelfde issue of moet ik hier een nieuwe issue voor aanmaken?

geen aanpassingen geweest er werkt eigenlijk foutloos sinds mid vorig haar. 
Laatste HA versie ( maar probleem trad ook met de voorlaatste )
DAY Ahead versie : 2025.3.1 

=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-04-03 12:28:13 +0200] [13] [INFO] Starting gunicorn 23.0.0
[2025-04-03 12:28:13 +0200] [13] [INFO] Listening at: http://0.0.0.0:5000 (13)
[2025-04-03 12:28:13 +0200] [13] [INFO] Using worker: sync
[2025-04-03 12:28:14 +0200] [23] [INFO] Booting worker with pid: 23
[2025-04-03 12:28:14 +0200] [24] [INFO] Booting worker with pid: 24
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 94, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/

corneel27, 2025-04-04:
Voorlopige oplossing: specificeer je verbinding met Home Assiatant in de settings:
 ```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.36",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
waarbij :
- `ip-adress` het ip adres van je ha machine is
- `token`: het api-token wat je in HA genereert

[image]

Zie ook: https://github.com/corneel27/day-ahead/issues/214

corneel27, 2025-04-04:
Om in de toekomst dit soort ervaringen met elkaar uit te wisselen heb ik op tweakers een topic geopend:
https://gathering.tweakers.net/forum/list_messages/2293338

Er komt in juni nog een belangrijke update van DAO aan: overgang van uur naar kwartier waarden van de day-ahead prijzen.
Ook de ervaringen daarmee kunnen we in het forum delen.

Ook ervaringen en tips van andere gebruikers zijn in dit topic welkom.

Ik zoek nog mensen die mij bij het bijhouden en up to date houden van dit topic willen helpen.

closed 2025-04-04 by corneel27