corneel27/day-ahead #212: Fout in self.calc_weekday_baseload(weekday)

balk77, 2025-03-27:
Hi, wanneer ik de baseload probeer te berekenen dan krijg ik deze error:
```
 2025-03-27 18:05:56 info: Day Ahead Optimalisering versie: 2025.3.1
2025-03-27 18:05:56 info: Day Ahead Optimalisering gestart op: 27-03-2025 18:05:56
2025-03-27 18:05:56 info: Day Ahead Optimalisatie gestart: 27-03-2025 18:05:56 taak: calc_baseloads
2025-03-27 18:05:56 fout: Er is een fout opgetreden, zie de fout-tracering
Traceback (most recent call last):
  File "/root/dao/prog/da_base.py", line 478, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 424, in calc_baseloads
    report.calc_save_baseloads()
  File "/root/dao/prog/da_report.py", line 2713, in calc_save_baseloads
    baseload = self.calc_weekday_baseload(weekday)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2639, in calc_weekday_baseload
    ev_consumption = self.get_sensor_week_sum(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2598, in get_sensor_week_sum
    for sensor in sensor_list:
TypeError: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3448, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3442, in main
    da_calc.run_task_function("calc_baseloads")
  File "/root/dao/prog/da_base.py", line 478, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 424, in calc_baseloads
    report.calc_save_baseloads()
  File "/root/dao/prog/da_report.py", line 2713, in calc_save_baseloads
    baseload = self.calc_weekday_baseload(weekday)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2639, in calc_weekday_baseload
    ev_consumption = self.get_sensor_week_sum(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2598, in get_sensor_week_sum
    for sensor in sensor_list:
TypeError: 'NoneType' object is not iterable
 
```
Ik heb deze fout met sqlite (9 dagen data, `  "baseload calc periode": 7,`) en ook met postgresql (vrijwel geen data). Ik zag ook fouten van een zelfde soort bij de optimalisatie op het hele uur (maar nu ben ik die foutmelding kwijt). `use_calc_baseload=False`

corneel27, 2025-03-27:
Kun je wel een report\balance oproepen van bijvoorbeeld "deze week"?

balk77, 2025-03-28:
"deze week" gaat goed, "dit jaar" gaat fout, met dit in de Home Assistang logs:
```
[2025-03-28 17:02:27,234] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.GroupingError: column "t2.start_ts" must appear in the GROUP BY clause or be used in an aggregate function
LINE 1: ...rt_ts), 'YYYY-MM') AS maand, to_char(to_timestamp(t2.start_t...
                                                             ^
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 248, in menu
    return reports(current_menu)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 572, in reports
    report_df = report.get_energy_balance_data(active_period, _tot=tot)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 1583, in get_energy_balance_data
    ha_result = self.get_sensor_data(
                ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 985, in get_sensor_data
    df_raw = pd.read_sql(query, connection)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 734, in read_sql
    return pandas_sql.read_query(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1836, in read_query
    result = self.execute(sql, params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1660, in execute
    return self.con.execute(sql, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1416, in execute
    return meth(
           ^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 523, in _execute_on_connection
    return connection._execute_clauseelement(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1638, in _execute_clauseelement
    ret = self._execute_context(
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1843, in _execute_context
    return self._exec_single_context(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1983, in _exec_single_context
    self._handle_dbapi_exception(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 942, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.ProgrammingError: (psycopg2.errors.GroupingError) column "t2.start_ts" must appear in the GROUP BY clause or be used in an aggregate function
LINE 1: ...rt_ts), 'YYYY-MM') AS maand, to_char(to_timestamp(t2.start_t...
                                                             ^
[SQL: SELECT to_char(to_timestamp(t2.start_ts), %(to_char_1)s) AS maand, to_char(to_timestamp(t2.start_ts), %(to_char_2)s) AS tijd, max(to_char(to_timestamp(t2.start_ts), %(to_char_3)s)) AS tot, sum(CASE WHEN (t2.state > t1.state) THEN t2.state - t1.state ELSE %(param_1)s END) AS cons 
FROM statistics AS t1 JOIN statistics AS t2 ON t2.start_ts = t1.start_ts + %(start_ts_1)s JOIN statistics_meta AS v1 ON v1.id = t1.metadata_id AND v1.id = t2.metadata_id 
WHERE v1.statistic_id = %(statistic_id_1)s AND t1.state IS NOT NULL AND t2.state IS NOT NULL AND t1.start_ts >= EXTRACT(epoch FROM to_timestamp(%(to_timestamp_1)s, %(to_timestamp_2)s)) - %(param_2)s AND t1.start_ts < EXTRACT(epoch FROM to_timestamp(%(to_timestamp_3)s, %(to_timestamp_4)s)) - %(param_3)s GROUP BY maand]
[parameters: {'to_char_1': 'YYYY-MM', 'to_char_2': 'YYYY-MM-01', 'to_char_3': 'YYYY-MM-DD HH24:MI:SS', 'param_1': 0, 'start_ts_1': 3600, 'statistic_id_1': 'sensor.dsmr_reading_electricity_delivered_1', 'to_timestamp_1': '2025-01-01 00:00:00', 'to_timestamp_2': 'YYYY-MM-DD hh24:mi:ss', 'param_2': 3600, 'to_timestamp_3': '2025-03-28 17:00:00', 'to_timestamp_4': 'YYYY-MM-DD hh24:mi:ss', 'param_3': 3600}]
(Background on this error at: https://sqlalche.me/e/20/f405)
```

corneel27, 2025-04-29:
Fixed with version 2025.4.1

balk77, 2025-05-08:
Ik draai nu 2025.4.2 en de issue is er nog steeds. Ik kan geen baseload genereren en reports gaan ook niet goed. `Report -> grid -> this year` werkt wel, `Report -> balans -> this year` geeft de volgende fout. Ik gebruik postgresql

```
[2025-05-08 21:53:28,630] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 945, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.UndefinedFunction: function datetime(double precision, unknown, unknown) does not exist
LINE 1: ...t_ts), 'YYYY-MM-DD') AS dag, strftime('%Y-%m-%d', datetime(t...
                                                             ^
HINT:  No function matches the given name and argument types. You might need to add explicit type casts.
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 248, in menu
    return reports(current_menu)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 576, in reports
    report_df = report.get_energy_balance_data(active_period, _tot=tot)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 1598, in get_energy_balance_data
    ha_result = self.get_sensor_data(
                ^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 992, in get_sensor_data
    df_raw = pd.read_sql(query, connection)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 734, in read_sql
    return pandas_sql.read_query(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1836, in read_query
    result = self.execute(sql, params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1660, in execute
    return self.con.execute(sql, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1416, in execute
    return meth(
           ^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 523, in _execute_on_connection
    return connection._execute_clauseelement(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1638, in _execute_clauseelement
    ret = self._execute_context(
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1843, in _execute_context
    return self._exec_single_context(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1983, in _exec_single_context
    self._handle_dbapi_exception(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 945, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.ProgrammingError: (psycopg2.errors.UndefinedFunction) function datetime(double precision, unknown, unknown) does not exist
LINE 1: ...t_ts), 'YYYY-MM-DD') AS dag, strftime('%Y-%m-%d', datetime(t...
                                                             ^
HINT:  No function matches the given name and argument types. You might need to add explicit type casts.
[SQL: SELECT to_char(to_timestamp(t2.start_ts), %(to_char_1)s) AS dag, strftime(%(strftime_1)s, datetime(t2.start_ts, %(datetime_1)s, %(datetime_2)s)) AS tijd, max(to_char(to_timestamp(t2.start_ts), %(to_char_2)s)) AS tot, sum(CASE WHEN (t2.state > t1.state) THEN t2.state - t1.state ELSE %(param_1)s END) AS cons 
FROM statistics AS t1 JOIN statistics AS t2 ON t2.start_ts = t1.start_ts + %(start_ts_1)s JOIN statistics_meta AS v1 ON v1.id = t1.metadata_id AND v1.id = t2.metadata_id 
[... cut ...]
[parameters: {'to_char_1': 'YYYY-MM-DD', 'strftime_1': '%Y-%m-%d', 'datetime_1': 'unixepoch', 'datetime_2': 'localtime', 'to_char_2': 'YYYY-MM-DD HH24:MI:SS', 'param_1': 0, 'start_ts_1': 3600, 'statistic_id_1': 'sensor.dsmr_reading_electricity_delivered_1', 'to_timestamp_1': '2025-05-05 00:00:00', 'to_timestamp_2': 'YYYY-MM-DD hh24:mi:ss', 'param_2': 3600, 'to_timestamp_3': '2025-05-08 21:00:00', 'to_timestamp_4': 'YYYY-MM-DD hh24:mi:ss', 'param_3': 3600}]
(Background on this error at: https://sqlalche.me/e/20/f405)
```

corneel27, 2025-05-09:
Ik heb dankzij jouw melding een andere fout in de postgresql-module gerepareerd, maar jouw fout kan ik niet nadoen.
Ik vrees dat het toch ook komt doordat de inhoud van je databasenatuurlijk anders is dan de mijne.
Zou je jouw HA-database kunnen exporteren naar sqlite3 en dan dat bestand met mij willen delen?

corneel27, 2025-05-09:
Om alles goed te kunnen testen heb ik ook nog nodig: een export van de DA-database en en copy van je settings.
Misschien alles inpakken in een zip en "we transfer" via een DM op GoT?

balk77, 2025-05-17:
Ik krijg nu deze fout bij 2025.5.0.rc5a:
```
2025-05-17 07:21:23 info: Day Ahead Optimalisering versie: 2025.5.0.rc5a
2025-05-17 07:21:23 info: Day Ahead Optimalisering gestart op: 17-05-2025 07:21:23
2025-05-17 07:21:23 info: Day Ahead Optimalisatie gestart: 17-05-2025 07:21:23 taak: calc_baseloads
2025-05-17 07:21:23 fout: Er is een fout opgetreden, zie de fout-tracering
Traceback (most recent call last):
  File "/root/dao/prog/da_base.py", line 480, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 425, in calc_baseloads
    report.calc_save_baseloads()
  File "/root/dao/prog/da_report.py", line 2785, in calc_save_baseloads
    baseload = self.calc_weekday_baseload(weekday)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2711, in calc_weekday_baseload
    ev_consumption = self.get_sensor_week_sum(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2670, in get_sensor_week_sum
    for sensor in sensor_list:
TypeError: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3518, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3512, in main
    da_calc.run_task_function("calc_baseloads")
  File "/root/dao/prog/da_base.py", line 480, in run_task_function
    getattr(self, run_task["function"])()
  File "/root/dao/prog/da_base.py", line 425, in calc_baseloads
    report.calc_save_baseloads()
  File "/root/dao/prog/da_report.py", line 2785, in calc_save_baseloads
    baseload = self.calc_weekday_baseload(weekday)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2711, in calc_weekday_baseload
    ev_consumption = self.get_sensor_week_sum(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2670, in get_sensor_week_sum
    for sensor in sensor_list:
TypeError: 'NoneType' object is not iterable
```

corneel27, 2025-05-18:
Ik heb mij HA-database omgezet naar progresql, maar ik krijg jouw fout niet nagedaan.
Waarschijnlijk zit er een fout in jouw report-definitie bij ev:
**`ev_consumption = self.get_sensor_week_sum(`**
Daar moet zoiets staan als:
```
"entities ev consumption" : [
      "sensor.fritz_dect_200_laadpunt_energie"
    ],
```
of als je geen ev hebt:
```
"entities ev consumption" : [],
```

corneel27, 2025-06-11:
Dank voor je melding.
Ik ga de code aanpassen.
Daar is een simpele oplossing voor:
als bij het ophaen van de list de regel niet in de settings staan dan niet **None** teruggeven maar een empty list: **[]**

corneel27, 2025-07-06:
Volgens mij is dit issue opgelost met versies 2025.6.2 en 2025.7.0.