corneel27/day-ahead #213: Supervisor status code 500

thoutenbos, 2025-03-31:
Hoi Cees,

Sinds vanmiddag lukt het opeens niet meer om de optimaliseringsberekening te maken. Bij een request van DAO naar supervisor ontstaat er een error 500 in supervisor. Ik heb al een algehele herstart geprobeerd maar helaas zonder resultaat.

Heb jij een idee wat er mis gaat?

Groeten,
Thijs

DAO logs:

```Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3448, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3411, in main
    da_calc = DaCalc("../data/options.json")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../prog/day_ahead.py", line 30, in __init__
    super().__init__(file_name=file_name)
  File "/root/dao/prog/da_base.py", line 94, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/
 ```

Supervisor log:

```
2025-03-31 20:41:56.863 ERROR (MainThread) [aiohttp.server] Error handling request from 172.30.33.1
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 198, in block_bad_requests
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 212, in system_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 229, in token_validation
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/middleware/security.py", line 293, in core_proxy
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/supervisor/supervisor/api/proxy.py", line 120, in api
    path = request.match_info["path"]
           ~~~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'path'```

DAO versie 2025.3.1 en supervisor versie 2025.03.4

dabit20, 2025-03-31:
Even verder vanaf #214.

Volgens de documentatie:

[image]

In een terminal:

```
~ curl -X GET -H "Authorization: Bearer ${SUPERVISOR_TOKEN}" -H "Content-Type: application/json" http://supervisor/core/api/config
{"allowlist_external_dirs":["/media","/tmp","/config/cam_snapshots","/config/www"],"allowlist_external_urls":[],"components":["mqtt.climate","ffmpeg","cloud","shelly.cover","persistent_notification","hue","shelly.event","shelly.sensor","bluetooth","trace","blueprint","text","dhcp","matter.update","input_datetime","hue.switch","shelly.select","matter.switch","esphome.button","websocket_api","template","sun.sensor","met","mobile_app","light","matter.button","min_max.sensor","ssdp","number","esphome.light","dsmr.sensor","zwave_js.button","nodered.binary_sensor","file_upload","mobile_app.binary_sensor","diagnostics","rfxtrx.event","template.binary_sensor","schedule","zwave_js.light","esphome.switch","hue.event","media_player","hacs.switch","esphome.number","homekit_controller","matter.vacuum","image_upload","zwave_js.select","script","usb","shelly","integration.sensor","rfxtrx.sensor","camera","webhook","hacs","homekit_controller.climate","cover","mqtt.binary_sensor","template.sensor","sensor","rfxtrx.switch","matter.fan","automation","rfxtrx.light","nodered.sensor","energy","met.weather","entsoe.sensor","mqtt.select","sun","matter.cover","weather","mobile_app.notify","mqtt","homeassistant_alerts","notify","homekit_controller.select","matter.valve","http","update","matter.light","group.binary_sensor","lock","fan","hassio","matter.binary_sensor","esphome.climate","hue.scene","rfxtrx.cover","vacuum","zwave_js.update","backup","network","recorder","logger","intent","dsmr","shelly.light","hue.sensor","analytics","cloud.tts","zwave_js.binary_sensor","hacs.update","esphome.camera","matter","device_tracker","input_number","zwave_js.sensor","timer","matter.number","shelly.binary_sensor","input_boolean","bluetooth_adapters","config","nodered","stream","matter.sensor","frontend","input_text","rfxtrx.binary_sensor","hassio.sensor","shelly.climate","modbus.sensor","zwave_js.number","device_automation","siren","matter.climate","threshold.binary_sensor","esphome.binary_sensor","media_source","my","matter.event","influxdb","api","rest","hardware","tado","homeassistant","mqtt.sensor","shelly.update","matter.lock","homekit_controller.switch","mobile_app.sensor","system_health","threshold","forecast_solar","system_log","esphome.update","stt","default_config","scene","homekit_controller.media_player","select","zwave_js","mqtt.switch","event","application_credentials","conversation","shelly.number","zone","repairs","shelly.button","hassio.binary_sensor","history","hassio.update","wake_word","mobile_app.device_tracker","forecast_solar.sensor","tag","binary_sensor","assist_pipeline","google_translate.tts","modbus","group","mqtt.button","esphome.sensor","homekit_controller.button","tts","auth","switch","counter","hue.binary_sensor","person","zeroconf","climate","valve","min_max","esphome","rest.sensor","onboarding","shelly.switch","energy.sensor","search","mqtt.number","hue.light","input_select","shelly.text","homekit_controller.sensor","rfxtrx","rfxtrx.siren","integration","lovelace","go2rtc","logbook","input_button","homeassistant.scene","rest_command","entsoe","matter.select","button","shelly.valve"],"config_dir":"/config","config_source":"storage","country":"NL","currency":"EUR","debug":false,"elevation":0,"external_url":"https://hass.icecoldcomputing.com","internal_url":null,"language":"en","latitude":51.77404080851815,"location_name":"Thuis","longitude":5.525264620498978,"radius":100,"recovery_mode":false,"safe_mode":false,"state":"RUNNING","time_zone":"Europe/Amsterdam","unit_system":{"length":"km","accumulated_precipitation":"mm","area":"m²","mass":"g","pressure":"Pa","temperature":"°C","volume":"L","wind_speed":"m/s"},"version":"2025.3.4","whitelist_external_dirs":["/media","/tmp","/config/cam_snapshots","/config/www"]}#                                                                                                                                          

➜  ~ curl -X GET -H "Authorization: Bearer ${SUPERVISOR_TOKEN}" -H "Content-Type: application/json" http://supervisor/core/api       
404: Not Found#                                                                                                                                                                                                                                                                                                                                                                                        

➜  ~ curl -X GET -H "Authorization: Bearer ${SUPERVISOR_TOKEN}" -H "Content-Type: application/json" http://supervisor/core/api/      
500 Internal Server Error

Server got itself in trouble#                                                                                                                                                                                                                                                                                                                                                                          
➜  ~ 
``` 

Lijkt toch niet te kloppen met wat ze beloven. Maar hoe dat dan opeens zomaar omvalt zonder interventie van, euh, systeembeheer...

tonvanboven, 2025-03-31:
Dag Cees, hier hetzelfde probleem. Bij het opstarten van de Add-on, versie 2025.3.1, staat er dit in de log van de Addn-on:

`=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-03-31 22:22:18 +0200] [14] [INFO] Starting gunicorn 23.0.0
[2025-03-31 22:22:18 +0200] [14] [INFO] Listening at: http://0.0.0.0:5000 (14)
[2025-03-31 22:22:18 +0200] [14] [INFO] Using worker: sync
[2025-03-31 22:22:18 +0200] [28] [INFO] Booting worker with pid: 28
[2025-03-31 22:22:19 +0200] [29] [INFO] Booting worker with pid: 29
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 94, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/`

corneel27, 2025-03-31:
Voorlopige oplossing: specificeer je verbinding met Home Assiatant in de settings:
 ```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.36",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
waarbij :
- `ip-adress` het ip adres van je ha machine is
- `token`: het api-token wat je in HA genereert

[image]

tonvanboven, 2025-03-31:
Top Cees, dat werkt! Dank voor je snelle reactie.

dabit20, 2025-03-31:
Hier werkt het ook, bedankt! 
Ik had het al geprobeerd met de waarde van ${SUPERVISOR_TOKEN} in de shell als token, maar dat werkte niet.

corneel27, 2025-03-31:
Om in de toekomst dit soort ervaringen met elkaar uit te wisselen heb ik op tweakers een topic geopend:
https://gathering.tweakers.net/forum/list_messages/2293338

Er komt in juni nog een belangrijke update van DAO aan: overgang van uur naar kwartier waarden van de day-ahead prijzen.
Ook de ervaringen daarmee kunnen we in het forum delen.

Ook ervaringen en tips van andere gebruikers zijn in dit topic welkom.

Ik zoek nog mensen die mij bij het bijhouden en up to date houden van dit topic willen helpen.

closed 2025-03-31 by corneel27

thoutenbos, 2025-04-01:
Bedankt Cees voor de snelle response!

Met de aanpassing werkt het weer perfect. Blijft gek dat het opeens instort, bij mij ook zonder wijziging verder (voor zover ik weet). Ben ook wel actief op Tweakers dus zal het topic in de gaten houden en ook wat proberen te ondersteunen.