corneel27/day-ahead #214: hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/

dabit20, 2025-03-31:
Zucht. 
Tot 17.00 vanavond draaide alles prima; dat was de laatste voorspelling. Sindsdien krijg ik InternalServerError's.

Stukje log:
```
=> directory dao_data exist
=> /root/dao/data doesn't exist, made
=> /root/dao/webserver/app/static/data exist
Setting up watches.
Watches established.
[2025-03-31 20:32:35 +0200] [12] [INFO] Starting gunicorn 23.0.0
[2025-03-31 20:32:35 +0200] [12] [INFO] Listening at: http://0.0.0.0:5000 (12)
[2025-03-31 20:32:35 +0200] [12] [INFO] Using worker: sync
[2025-03-31 20:32:35 +0200] [22] [INFO] Booting worker with pid: 22
[2025-03-31 20:32:35 +0200] [23] [INFO] Booting worker with pid: 23
Traceback (most recent call last):
  File "/root/dao/prog/da_scheduler.py", line 65, in <module>
    main()
  File "/root/dao/prog/da_scheduler.py", line 60, in main
    da_sched = DaScheduler("../data/options.json")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_scheduler.py", line 10, in __init__
    super().__init__(file_name)
  File "/root/dao/prog/da_base.py", line 94, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/
```

Dit was met homeassistant 2025.3.0 en DAO 2025.2.0. 
Inmiddels een rondje updaten verder naar het meeste recente HaOS (15.1), Core (2025.3.4) en DAO (2025.3.1). Maakt helaas geen verschil.

Ik kan bij de webUI van DAO. Probeer ik daar bij Run een actie uit te voeren, dan resulteert dat in dezelfde 'InternalServerError: 500 status code returned from http://supervisor/core/api/' foutmelding als in de logs. Baseloads berekenen, dayahead/meteoprognoses ophalen, optimalisatieberekening met/zonder debug, allemaal resulteren ze in die fout. 

Addon herstarten, HA herstarten, VM waar HA op draait herstarten, geen verschil.

Ik heb geen idee waar ik moet zoeken, dat is de reden achter het creeeren van dit issue.

De NodeRED addon draait overigens prima; mij lijkt dat die dezelfde API gebruikt? Heb ik nooit wat voor in moeten stellen in ieder geval.

sjampeter, 2025-03-31:
hier hetzelfde issue sinds 15 uur, en bij mijn zwager sinds 20 uur. bijzonder weer.

corneel27, 2025-03-31:
Voorlopige oplossing: specificeer je verbinding met Home Assiatant in de settings:
 ```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.36",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
waarbij :
- `ip-adress` het ip adres van je ha machine is
- `token`: het api-token wat je in HA genereert

[image]

closed 2025-03-31 by corneel27