corneel27/day-ahead #215: DAO no longer works with version 1.3 and 3.1

soeter01, 2025-04-01:
As of today with version 1.3 and 3.1 following erros occurs with automated (scheduled) or manual run commands. Nothing seem to work anymore after months of no issues whatsoever. Nothing obvious was changed. Any ideas?

Logging van bewerking "Optimaliseringsberekening met debug":
Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3448, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3411, in main
    da_calc = DaCalc("../data/options.json")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../prog/day_ahead.py", line 30, in __init__
    super().__init__(file_name=file_name)
  File "/root/dao/prog/da_base.py", line 94, in __init__
    super().__init__(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in __init__
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/

hvdort, 2025-04-01:
I have exact the same problem since today

pvhpsb, 2025-04-01:
Hier ook hetzelfde probleem.

corneel27, 2025-04-02:
Voorlopige oplossing: specificeer je verbinding met Home Assiatant in de settings:
 ```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.36",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
waarbij :
- `ip-adress` het ip adres van je ha machine is
- `token`: het api-token wat je in HA genereert

[image]

Zie ook: https://github.com/corneel27/day-ahead/issues/214

corneel27, 2025-04-02:
Om in de toekomst dit soort ervaringen met elkaar uit te wisselen heb ik op tweakers een topic geopend:
https://gathering.tweakers.net/forum/list_messages/2293338

Er komt in juni nog een belangrijke update van DAO aan: overgang van uur naar kwartier waarden van de day-ahead prijzen.
Ook de ervaringen daarmee kunnen we in het forum delen.

Ook ervaringen en tips van andere gebruikers zijn in dit topic welkom.

Ik zoek nog mensen die mij bij het bijhouden en up to date houden van dit topic willen helpen.

closed 2025-04-02 by corneel27