corneel27/day-ahead #216: Custom component

nlmaca, 2025-04-02:
Deze addon is niet voor iedereen te installeren. 
Ik draai o.a. HA, MySQL, PostgresSql etc in docker containers en heb zodoende de DAO addon niet beschikbaar en kan deze ook niet via 'repositories' installeren, omdat die functie er niet in zit. 
Normaal gesproken zijn er dan ook custom_components beschikbaar. Ik zie alleen voor DAO die oplossing niet?

https://github.com/corneel27/day-ahead/blob/main/dao/DOCS.md

Vraag: 
- Is er een custom component beschikbaar?
- Is de add on op een andere manier werkend te krijgen indien HA in een docker omgeving draait?

corneel27, 2025-04-04:
Kijk op het topic in tweakers over DAO.
Daar wordt dit uitgebreid besproken.

Om in de toekomst dit ervaringen met elkaar uit te wisselen heb ik op tweakers een topic geopend:
https://gathering.tweakers.net/forum/list_messages/2293338

Er komt in juni nog een belangrijke update van DAO aan: overgang van uur naar kwartier waarden van de day-ahead prijzen.
Ook de ervaringen daarmee kunnen we in het forum delen.

Ook ervaringen en tips van andere gebruikers zijn in dit topic welkom.

Ik zoek nog mensen die mij bij het bijhouden en up to date houden van dit topic willen helpen.

closed 2025-04-04 by corneel27

corneel27, 2025-04-04:
Ik bedoelde het topic op tweakers. 
Maar als jij tijd en zin hebt om mee te testen en nieuwe releases te reviewen (op github) dan ben je ook welkom.