corneel27/day-ahead #217: 500 error van HomeAssistant

georgeboot, 2025-04-03:
Een kennis van me, krijgt sinds gisteren opeens deze fout:

```

Logging van bewerking "Optimaliseringsberekening zonder debug":
Traceback (most recent call last):
  File "/root/dao/webserver/../prog/day_ahead.py", line 3448, in <module>
    main()
  File "/root/dao/webserver/../prog/day_ahead.py", line 3411, in main
    da_calc = DaCalc("../data/options.json")
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/../prog/day_ahead.py", line 30, in init
    super().init(file_name=file_name)
  File "/root/dao/prog/da_base.py", line 94, in init
    super().init(hassurl=self.hassurl, token=self.hasstoken)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 36, in init
    self._assert_api_running()
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 40, in _assert_api_running
    if not self._api_is_running():
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 46, in _api_is_running
    return self._get("/")["message"] == "API running."  # type: ignore
           ^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 54, in _get
    return self._process_response(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 90, in _process_response
    self._raise_error(response.status_code, response.url)
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/hassapi/client/base.py", line 95, in _raise_error
    raise error(f"{status_code} status code returned from {url}",)  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
hassapi.exceptions.InternalServerError: 500 status code returned from http://supervisor/core/api/
```

De ‘online check’ geeft dus een 500 terug. Home assistant heeft geen update gehad (daarna wel geprobeerd maar hielp niet). Draait nu 2025.4.0.

Herstart add on helpt ook niet. Probleem lijkt in home assistant te zitten.

Enig idee waarom HA dit doet? Is het te debuggen? Wellicht advies wat ik er mee kan doen?

balk77, 2025-04-03:
Ik krijg deze fout ook, ik weet niet sinds wanneer. Ik draai Home Assistant 2025.3.4

hfvdv, 2025-04-04:
Ik heb vanochtend een nieuwe install gedaan op een nieuwe home assistant installatie en loop daar ook tegen dezelfde foutmelding aan.
- Saving doet het wel
- Het lijkt erop dat ie de tibber prijzen niet op kan halen?

H

georgeboot, 2025-04-04:
In mijn geval sloopt hij ook de optimalisatieberekening, en dus ‘doet hij het niet’.

Is dat bij jullie ook zo? Begon hier 2 april ergens.

corneel27, 2025-04-04:
Voorlopige oplossing: specificeer je verbinding met Home Assiatant in de settings:
 ```
"homeassistant": {
    "protocol api": "http",
    "ip adress": "192.168.178.36",
    "ip port": 8123,
    "token": "!secret ha_api_token"
  },
```
waarbij :
- `ip-adress` het ip adres van je ha machine is
- `token`: het api-token wat je in HA genereert

[image]

Zie ook: https://github.com/corneel27/day-ahead/issues/214

corneel27, 2025-04-04:
Om in de toekomst dit soort ervaringen met elkaar uit te wisselen heb ik op tweakers een topic geopend:
https://gathering.tweakers.net/forum/list_messages/2293338

Er komt in juni nog een belangrijke update van DAO aan: overgang van uur naar kwartier waarden van de day-ahead prijzen.
Ook de ervaringen daarmee kunnen we in het forum delen.

Ook ervaringen en tips van andere gebruikers zijn in dit topic welkom.

Ik zoek nog mensen die mij bij het bijhouden en up to date houden van dit topic willen helpen.

closed 2025-04-04 by corneel27