corneel27/day-ahead #219: Internal Server Error when on Postgresql

balk77, 2025-04-11:
Hi, ik gebruik postgresql. Indien ik naar "Savings" ga krijg ik een internal server error (in de browser). Ik draai versie 2025.04 maar had dezelfde fout bij de vorige versie. Dit komt uit de log:
```
[2025-04-11 14:17:45,308] ERROR in app: Exception on / [POST]
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 945, in do_execute
    cursor.execute(statement, parameters)
psycopg2.errors.GroupingError: column "t1.time" must appear in the GROUP BY clause or be used in an aggregate function
LINE 1: ...t1.time), 'HH24:MI') AS uur, to_char(to_timestamp(t1.time), ...
                                                             ^
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 261, in menu
    return reports("savings")
           ^^^^^^^^^^^^^^^^^^
  File "/root/dao/webserver/app/routes.py", line 582, in reports
    report_df = calc_function(active_period, _tot=tot,
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 2172, in calc_saving_consumption
    df = self.get_columns(calc_dict, active_period, _tot)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 1799, in get_columns
    df_column = self.get_da_data(key, vanaf, tot, get_interval=get_interval,
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/prog/da_report.py", line 1770, in get_da_data
    code_result = pd.read_sql(query, connection)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 734, in read_sql
    return pandas_sql.read_query(
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1836, in read_query
    result = self.execute(sql, params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/pandas/io/sql.py", line 1660, in execute
    return self.con.execute(sql, *args)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1416, in execute
    return meth(
           ^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 523, in _execute_on_connection
    return connection._execute_clauseelement(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1638, in _execute_clauseelement
    ret = self._execute_context(
          ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1843, in _execute_context
    return self._exec_single_context(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1983, in _exec_single_context
    self._handle_dbapi_exception(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 2352, in _handle_dbapi_exception
    raise sqlalchemy_exception.with_traceback(exc_info[2]) from e
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/base.py", line 1964, in _exec_single_context
    self.dialect.do_execute(
  File "/root/dao/venv/day_ahead/lib/python3.11/site-packages/sqlalchemy/engine/default.py", line 945, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.ProgrammingError: (psycopg2.errors.GroupingError) column "t1.time" must appear in the GROUP BY clause or be used in an aggregate function
LINE 1: ...t1.time), 'HH24:MI') AS uur, to_char(to_timestamp(t1.time), ...
                                                             ^
[SQL: SELECT to_char(to_timestamp(t1.time), %(to_char_1)s) AS uur, to_char(to_timestamp(t1.time), %(to_char_2)s) AS tijd, min(to_char(to_timestamp(t1.time), %(to_char_3)s)) AS vanaf, max(to_char(to_timestamp(t1.time), %(to_char_4)s)) AS tot, sum(t1.value) AS cons_zonder 
FROM values AS t1, variabel AS v1 
WHERE v1.code = %(code_1)s AND t1.variabel = v1.id AND t1.time >= EXTRACT(epoch FROM to_timestamp(%(to_timestamp_1)s, %(to_timestamp_2)s)) AND t1.time < EXTRACT(epoch FROM to_timestamp(%(to_timestamp_3)s, %(to_timestamp_4)s)) GROUP BY uur]
[parameters: {'to_char_1': 'HH24:MI', 'to_char_2': 'YYYY-MM-DD HH24:MI', 'to_char_3': 'YYYY-MM-DD HH24:MI:SS', 'to_char_4': 'YYYY-MM-DD HH24:MI:SS', 'code_1': 'cons_zonder', 'to_timestamp_1': '2025-04-11 00:00:00', 'to_timestamp_2': 'YYYY-MM-DD hh24:mi:ss', 'to_timestamp_3': '2025-04-11 14:00:00', 'to_timestamp_4': 'YYYY-MM-DD hh24:mi:ss'}]
(Background on this error at: https://sqlalche.me/e/20/f405)
```

corneel27, 2025-04-13:
Dank voor het melden.
Ik heb mijn database geconverteerd naar postgresql en getest.
Ik kon de fout nadoen en heb het gerepareerd.
Er komt binnenkort een update met de fix.

closed 2025-04-13 by corneel27

corneel27, 2025-04-13:
Er staat een testversie in de addon branche met een fix voor de door jou gerapporteerde fout.