corneel27/day-ahead #220: MInimum power not respected.

CrazyHenk44, 2025-04-12:
Hi,

I've set my batteries to have a minimum power:

```
"battery": [{
    "name": "D4R7",
    "entity actual level": "sensor.sessy_d4r7_state_of_charge",
    "capacity": 5.2,
    "lower limit": 5,
    "optimal lower level": 10,
    "upper limit": 100,
    "charge stages" : 
    [
      { "power": 0, "efficiency": 0.70 },
      { "power": 220, "efficiency": 0.70 },
      [snip]
    ],
    "discharge stages": 
    [
      { "power": 0, "efficiency": 0.70 },
      { "power": 170, "efficiency": 0.70 },
         [snip
    ],
    "reduced hours":{
       "1": 900,
           [snip]
       "8": 900},
    "solar": [],
    "cycle cost": 0,
    "dc_to_bat efficiency" : 1,
    "bat_to_dc efficiency" : 1,
    "minimum power": 200,
    "entity from battery": "input_number.power_advies_d4r7",
    "entity calculated soc": "input_number.doel_soc_d4r7"
  },      [snip]
```

But i find my batteries charging and discharging at 60 watts now and then, since the inverter also costs 50 watt, the expected SOC wont be reached and it stays 'burning' at 60 watts for hours in a row.

Can you please make sure you never push "entity from battery" below 200 and above -200 ?

corneel27, 2025-04-12:
I think I don't have explained it well in DOCS.md
When DAO calculates that in a certain houre there should less charged than 200 W (in your case) but for instance 100W.
Than **two** entities are in play:
- **entity from battery** which will  "loaded" with 200
- **entity stop inverter**  (if available) this is  a input_datetime, which will be "loaded" with a date/time 30 minutes away from "now"
It is up to you to make in automation in Home Assistant that will be triggered with this input_datetime to stop the charging/discharging of the battery.

Does this solve your problem?

CrazyHenk44, 2025-04-13:
So, if **entity stop inverter** is not available the **minimum power** is not respected at all?

Because I get 60 in **entity from battery**

corneel27, 2025-04-13:
I must investigate this further.
Can you send me:
- copy of your settings
- a copy of a calculation log (log level "info") when 60 W is filled in in your **entity from battery**

CrazyHenk44, 2025-04-16:
I used "entity from battery" and it should have been "entity set power feedin". Thanks @corneel27 for pointing that out.

closed 2025-04-16 by CrazyHenk44