corneel27/day-ahead #225: Instellen van helper entity (boolean battery balanceren) in home assistant geeft een corrupte entity

Torch1969, 2025-04-17:
Home Assistant 2025.3.4 in docker
DAO 2025.4.1 in docker

Instellen van helper entity in home assistant geeft een corrupte entity

Ik heb in dao een batterij gedefinieerd en de bijbehorende entiteiten in home assistant
Na een berekening worden alle entiteiten netjes bijgewerkt, behalve die voor balanceren:
[image]

Na openen van de entiteit en aan/uit zetten of wijzigen van de instellingen is de entiteit weer ok (tot volgende uur me berekening en bijwerken van de entiteiten):
[image]

Ik heb in de logs gekeken (met debug aan) van dao, home assistant en portainer, maar nergens iets opvallends en zeker geen foutmelding.

Dit is mijn options.json:
`{
  "homeassistant": {
      "ip adress": "192.168.1.101",
      "ip port": 8123,
      "token": "!secret ha_token"
  },
  "database ha": {
    "engine": "sqlite",
    "database": "home-assistant_v2.db",
    "db_path": "/homeassistant"
  },
  "database da": {
    "engine": "sqlite",
    "db_path": "../data"
  },
  "meteoserver-key": "!secret meteoserver-key",
  "prices": {
    "source day ahead": "nordpool",
    "regular high": 0.10,
    "regular low": 0.10,
    "switch to low": 23,
    "energy taxes delivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.10880,
      "2025-01-01": 0.10154
    },
    "energy taxes redelivery": {
      "2022-01-01": 0.06729,
      "2023-01-01": 0.12599,
      "2024-01-01": 0.10880,
      "2025-01-01": 0.10154
    },
    "cost supplier delivery": {
      "2024-08-01": 0.02
    },
    "cost supplier redelivery": {
      "2024-08-01": 0.02
    },
    "vat": {
      "2022-01-01": 21,
      "2022-07-01": 9,
      "2023-01-01": 21
    },
    "last invoice": "2025-01-01",
    "tax refund": "True"
  },
  "logging level" : "info",
  "use_calc_baseload": "False",
  "baseload calc periode": 56,
  "baseload": [
    0.36,
    0.36,
    0.36,
    0.36,
    0.36,
    0.36,
    0.45,
    0.55,
    0.40,
    0.40,
    0.40,
    0.40,
    0.55,
    0.40,
    0.40,
    0.40,
    0.40,
    0.70,
    0.60,
    0.50,
    0.50,
    0.50,
    0.40,
    0.36
  ],
  "graphical backend": "",
  "graphics": {
    "style": "Solarize_Light2",
    "show" : "true",
    "prices delivery": "True",
    "prices redelivery": "True",
    "average delivery": "True"
  },
  "strategy": "minimize cost",
  "notifications": {
  },
  "grid": {
    "max_power": 8
  },
  "history": {
    "save days": 7
  },
  "dashboard": {
    "port": 5000
  },
  "boiler": {
    "boiler present": "False",
    "entity actual temp.": "sensor.boiler_gemeten",
    "entity setpoint": "sensor.boiler_ingesteld",
    "entity hysterese": "sensor.hysterese_hot_water",
    "cop": 2.9,
    "cooling rate": 0.4,
    "volume": 180,
    "heating allowed below": 44,
    "elec. power": 1500,
    "activate service": "press",
    "activate entity": "input_button.hw_trigger"
  },
  "heating": {
    "heater present": "False",
    "degree days factor": 3.6,
    "stages": [
      {
        "max_power": 225,
        "cop": 7.1
      },
      {
        "max_power": 300,
        "cop": 7.0
      },
      {
        "max_power": 400,
        "cop": 6.5
      },
      {
        "max_power": 500,
        "cop": 6.0
      },
      {
        "max_power": 600,
        "cop": 5.5
      },
      {
        "max_power": 750,
        "cop": 5.0
      },
      {
        "max_power": 1000,
        "cop": 4.5
      },
      {
        "max_power": 1250,
        "cop": 4.0
      }
    ],
    "entity adjust heating curve": "input_number.stooklijn_verschuiving_day_ahead",
    "adjustment factor": 0.04
  },
  "battery": [
    {
      "name": "Sessy_dllu",
      "entity actual level": "sensor.sessy_dllu_state_of_charge",
      "capacity": 5.5,
      "upper limit": 100,
      "lower limit": 0,
      "optimal lower level": 0,
      "entity min soc end opt": "input_number.dao_sessy_dllu_min_soc_end",
      "entity max soc end opt": "input_number.dao_sessy_dllu_max_soc_end",
      "charge stages": [
        {"power": 0.0,
         "efficiency": 1},
        {"power": 60.0,
         "efficiency": 0.7},
        {"power": 110.0,
         "efficiency": 0.758},
        {"power": 220.0,
         "efficiency": 0.850},
        {"power": 330.0,
         "efficiency": 0.892},
        {"power": 440.0,
         "efficiency": 0.912},
        {"power": 660.0,
         "efficiency": 0.933},
        {"power": 880.0,
         "efficiency": 0.942},
        {"power": 1100.0,
         "efficiency": 0.946},
        {"power": 1320.0,
         "efficiency": 0.942},
        {"power": 1540.0,
         "efficiency": 0.938},
        {"power": 1760.0,
         "efficiency": 0.929},
        {"power": 1980.0,
         "efficiency": 0.921},
        {"power": 2200.0,
         "efficiency": 0.908}
      ],
      "discharge stages": [
        {"power": 0.0,
         "efficiency": 1},
        {"power": 60.0,
         "efficiency": 0.7},
        {"power": 85.0,
         "efficiency": 0.735},
        {"power": 170.0,
         "efficiency": 0.829},
        {"power": 255.0,
         "efficiency": 0.882},
        {"power": 340.0,
         "efficiency": 0.921},
        {"power": 510.0,
         "efficiency": 0.943},
        {"power": 680.0,
         "efficiency": 0.957},
        {"power": 850.0,
         "efficiency": 0.957},
        {"power": 1020.0,
         "efficiency": 0.953},
        {"power": 1190.0,
         "efficiency": 0.943},
        {"power": 1360.0,
         "efficiency": 0.936},
        {"power": 1530.0,
         "efficiency": 0.929},
        {"power": 1700.0,
         "efficiency": 0.925}
      ],
      "reduced hours":
       {  "1": 2200,
          "2": 2200,
         "18": 2200,
         "19": 2200
       },
      "minimum power": 350,
      "dc_to_bat efficiency": 1,
      "bat_to_dc efficiency": 1,
      "cycle cost": 0.01,
      "entity set power feedin": "input_number.dao_sessy_dllu_set_power_feedin",
      "entity set operating mode": "input_select.dao_sessy_dllu_set_operating_mode",
      "entity stop inverter": "input_datetime.dao_sessy_dllu_stop_inverter",
      "entity balance switch": "input_boolean.dao_sessy_dllu_balance_switch",
      "entity calculated soc": "input_number.dao_sessy_dllu_calculated_soc",
      "solar": []
    }    
  ],
  "solar": [ 
    {
      "name": "Zon ZW",
      "tilt": 45,
      "orientation": 40,
      "capacity": 4.2,
      "yield": 0.01125,
      "entity pv switch": ""
    },
    {
      "name": "Zon NO",
      "tilt": 45,
      "orientation": -140,
      "capacity": 2.6,
      "yield": 0.00765,
      "entity pv switch": ""
    }  
  ],
  "electric vehicle": [ ],
  "machines" : [ ],
  "tibber": {
    "api_token": "!secret tibber_api_token"
  },
  "report": {
    "entities grid consumption": [
      "sensor.sessy_p9xk_tariff_1_consumed_energy",
      "sensor.sessy_p9xk_tariff_2_consumed_energy"
    ],
    "entities grid production": [
      "sensor.sessy_p9xk_tariff_1_produced_energy",
      "sensor.sessy_p9xk_tariff_2_produced_energy"
    ],
    "entities solar production ac": [
      "sensor.extra_total",
      "sensor.envoy_lifetime_energy_production"
    ],
    "entities solar production dc": [],
    "entities ev consumption" : ["sensor.laadpaal_thuis_energie"],
    "entities wp consumption" : ["sensor.ac_totaal_energie"],
    "entities boiler consumption": [],
    "entities battery consumption": ["sensor.sessy_dllu_charged_energy"],
    "entities battery production": ["sensor.sessy_dllu_discharged_energy"]
  },
  "scheduler": {
    "active": "true",
    "0545": "get_meteo_data",
    "1145": "get_meteo_data",
    "1745": "get_meteo_data",
    "2345": "get_meteo_data",
    "1255": "get_day_ahead_prices",
    "1355": "get_day_ahead_prices",
    "1455": "get_day_ahead_prices",
    "1554": "get_day_ahead_prices",
    "1655": "get_day_ahead_prices",
    "xx00": "calc_optimum",
    "2359": "clean_data"
  }
}
`

corneel27, 2025-04-17:
Je hebt een input_boolean opgegeven (die gebruik ik zelf ook) dus het zou goed moeten gaan.
Ik heb meer info nodig om te kijken waar het fout gaat:
- kun je mij een logging sturen van een berekening (met "logging level": "debug")
- kijk daarna ook in de logging van homeassistant. Bereikbaar via Instellingen\Systeem\Logboeken en dan Home Assistant Core rechtsboven:
[image]

Torch1969, 2025-04-18:
Hierbij de gevraagde info. Heb de berekening om 8:37:15 gestart

Het laatste stukje van de home assistant core log:
`2025-04-18 08:27:57.269 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for lilygo-t5-e-paper @ 10.10.20.86: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.10.20.86', port=6053))]: [Errno 113] Connect call failed ('10.10.20.86', 6053) (SocketAPIError)
2025-04-18 08:33:11.669 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for lilygo-t5-e-paper @ 10.10.20.86: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='10.10.20.86', port=6053))]: [Errno 113] Connect call failed ('10.10.20.86', 6053) (SocketAPIError)`
Zoals je kunt zien geen melding om 8:37:23

En de log van DAO zelf:

[calc_2025-04-18__08:37:23.log](https://github.com/user-attachments/files/19809155/calc_2025-04-18__08.37.23.log)

Ook daarin zie ik niets geks (neem aan dat het deel onderin relevant is waar de helpers in home assistant worden gevuld.)

corneel27, 2025-04-18:
Ik zie ook niks geks, dus ik blijf het antwoord nog even schuldig.
Ik kijk nu bij mij naar de regel van de helper in het overzicht van de helpers en zie nu hetzelfde icoon erbij staan in de kolom "Bewerkbaar", met de popup tekst "Niet beheerbaar".
Dat zie ik nu pas.
Maar de boolean werkt wel goed, want ik heb wel situaties gehad dat deze werd "aan" gezet en dan werd ook de automation getriggerd waarmee mijn grid-inkoop naar nul werd gestuurd.
Ik heb de input_boolean ook in een mijn gui in lijst met entiteiten staan en daar kan ik hem ook met de hand aan/uit zetten.
Ik zie nu ook dat veel helpers "Niet beheerbaar" zijn (met name die met een template-value).
Ik weet nog niet wat het betekent "Niet beheerbaar".
Jij wel?

Dus ik zie het voorlopig nog even aan als een  "feature" en niet als als een fout. 
Maar ik laat mij graag van het tegendeel overtuigen

corneel27, 2025-04-18:
OK, issue closed

closed 2025-04-18 by corneel27