corneel27/day-ahead #229: Ondersteuning voor meer dan 1 boiler

llevering, 2025-04-29:
Wij hebben 2 boilers i.p.v. (o.a. i.v.m. beperkte beschikbaarheid hoogte). Het zou mooi zijn als meer dan 1 boiler ook een optie zou zijn.

Daarnaast zou het mooi zijn als je kon instellen of ze gelijktijdig mogen draaien of niet. De boilers gebruiken dezelfde luchttoevoer, als het buiten koud (< 5c) dan slaat 1 van de boilers altijd af als je tegelijk laat lopen. Daarom plan ik ze zelf nooit tegelijk in, maar als het automatisch kan is dat mooier.

Voor nu is de workaround door 1 van de 2 boilers als machine in te plannen, dat werkt uiteraard ook.

ranjowd, 2025-04-29:
Zou die gelijktijdigheid ook met de warmtepomp combineert kunnen worden? in het geval je boiler door een warmtepomp wordt verwarmd die ook voor de verwarming dient? (maar dat dus niet gelijktijdig kan)

sjorszanders, 2025-05-12:
Ondersteuning voor meerdere boilers zou erg prettig zijn, zodat je hem bijvoorbeeld kunt inzetten voor een hottub of zwembad.