corneel27/day-ahead #230: Legionellarun boilers

llevering, 2025-04-29:
Het zou mooi als je kon aangeven dat de boilers ééns in de x dagen een legionella run moeten doen (eventueel met een soort van window van x, dagen om het te optimaliseren naar een gunstige dag).

ranjowd, 2025-04-29:
Ik heb ook al aan zoiets zitten denken. Op dit moment ben ik daar best flexibel in. De run moet tussen de 7 en 14 dagen draaien. Ik plan dit adhv de meerdaagse solar forecast. Ik zou evt in home assistant kunnen berekenen wanneer een legionella run weer nodig is, wetende dat dat een dag is met voldoende PV opbrengst.

corneel27, 2025-04-29:
Mijn voorkeur gaat uit naar een oplossing zoals gesuggereerd door @ranjowd .
Het is heel moeilijk (zo niet onmogelijk) om dat in DAO voor elkaar te krijgen.
In vier van de vijf weken zijn de prijzen op zondag het laagst, gevolgd door zaterdag, maar dat zegt natuurlijks niks over de pv-productie.

Als je HA dat laat bepalen zou je op die dag (door HA) het boiler-setpoint kunnen verhogen naar de door jou gewenste temperatuur (wel eerst de hysterese met een zelfde delta verhogen).
Na de run zet Home Assistant dan alles weer terug.

In principe is er in DAO ook een meerdaagse zon-voorspelling in de database.
Ik zou die eventueel wel beschikbaar kunnen maken via een api.

Torch1969, 2025-04-29:
> Mijn voorkeur gaat uit naar een oplossing zoals gesuggereerd door [@ranjowd](https://github.com/ranjowd) . Het is heel moeilijk (zo niet onmogelijk) om dat in DAO voor elkaar te krijgen. In vier van [...]

Ja graag, de zonnepanelen voorspelling (pv_ac) in ieder geval ook voor morgen al in de api teruggeven (en niet zoals nu pas als de prijzen bekend zijn).

corneel27, 2025-04-29:
Ik zet het op mijn to-do lijst voor een van de komende versies.

hemertje, 2026-10-03:
Een aanvulling vanuit een andere hoek, met meetgegevens. Ik gebruik een warmtepompboiler (Panasonic Aquarea, 300 L). De legionellarun draait bij mij via een scheduler in Node-RED (elke vrijdag 12:30, alleen als er meer dan 150 uur sinds de vorige is verstreken), dus los van DAO en niet op de goedkoopste of zonnigste dag.

Twee dingen die ik zie, die aansluiten bij de aanpak van @corneel27 (setpoint en hysterese in HA verhogen op een gekozen dag):
- Boven circa 52 °C schakelt de backup-heater (3 kW, COP 1) bij en een run naar 65 °C is bijna volledig weerstandsverwarming. Verhoogt HA alleen het setpoint, dan prijst DAO die stap alsof een warmtepomp met COP 3 het doet. Een tweetraps vermogen voor de boiler (vermogen en COP per stap) zou dat verhelpen. Ik heb dat samen met een paar metingen aan het boilermodel (vermogen, warmtecapaciteit per sensorgraad) in een apart issue gezet: https://github.com/corneel27/day-ahead/issues/873.
- Ik zag op Tweakers dat je de planningshorizon verlengt met prijsvoorspellingen. Daar past een wekelijks (of 7–14 daags) venster mooi in.

Ik meet de energie van mijn eerstvolgende volledige run (vrijdag 9 oktober) en deel de cijfers hier als dat nuttig is.