corneel27/day-ahead #237: [solar] Ondersteuning voor aftoppende omvormer

storeman, 2025-05-01:
Voor de optimalisatie zou het denk ik goed zijn als er op een solar-configuratie een `top_capacity` komt.

Achtergrond:
Ik heb een 10kW Growatt omvormer met twee strings. Per string is er een max van 7,5kW. Ik heb echter 9,6kW aan panelen liggen op 1 string, maar dit wordt dus nooit gehaald op de top-uren. In de uren voor en na de piek, is die 9,6kW natuurlijk wel weer relevant.

Bonus:
Als ik een tweede string met veel capaciteit zou leggen, dan zou er per string dus een max van 7,5kW zijn en op de omvormer zelf 10kW. In de huidige configuratie kan dit niet. Dan zou je twee PV's aan elkaar moeten linken, of op omvormer-niveau nog is moeten kunnen configureren.

corneel27, 2025-05-01:
Staat inmiddels op de to-do lijst.

corneel27, 2025-06-04:
Gerealiseerd in versie 2025.6.0

closed 2025-06-04 by corneel27