corneel27/day-ahead #247: Hoog vermogen warmtepomp terwijl niet ingepland

bvweerd, 2025-05-06:
Ik heb geen warmtevraag op de warmtepomp, en volgens de debug info zou ie niet worden ingepland. Toch geeft ie in de berekening een enorme warmtevraag in het eerste uur van de planning. Als ik de warmtevraag op 'ja' zet, wordt ie wel gewoon ingepland. Doe ik iets nog niet goed?

Debug info:
```
2025-05-06 14:22:40 info: Nog benodigde warmte: 7.4 kWh
2025-05-06 14:22:40 debug: Starting new HTTP connection (1): 192.168.2.2:8123
2025-05-06 14:22:40 debug: http://192.168.2.2:8123 "GET /api/states/input_boolean.dao_hp_heat_demand HTTP/1.1" 200 389
2025-05-06 14:22:40 info: Actuele warmtevraag: Nee
2025-05-06 14:22:40 debug: Implementatie on/off warmtepomp
2025-05-06 14:22:40 debug: Warmtepomp draait minimaal 1 uren
2025-05-06 14:22:40 info: Geen warmtevraag - warmtepomp wordt niet ingepland
```
Config:
```
  "heating": {
    "heater present": "True",
    "degree days factor": 1.55,
    "adjustment" : "on/off",
    "entity adjust heating curve": "input_number.dao_stooklijn_verschuiving_day_ahead",
    "entity avg outside temp": "input_number.dao_avg_outside_temperature",
    "entity hp cop": "sensor.dao_hp_cop",
    "entity hp power": "sensor.dao_hp_power",
    "entity hp switch" : "input_boolean.dao_hp_switch", 
    "entity hp heat demand": "input_boolean.dao_hp_heat_demand",
    "adjustment factor": 0.04,
    "stages": [
      {
        "max_power": 225,
        "cop": 7.1
      },
      {
        "max_power": 300,
        "cop": 7.0
      },
      {
        "max_power": 400,
        "cop": 6.5
      },
      {
        "max_power": 500,
        "cop": 6.0
      },
      {
        "max_power": 600,
        "cop": 5.5
      },
      {
        "max_power": 750,
        "cop": 5.0
      },
      {
        "max_power": 1000,
        "cop": 4.5
      },
      {
        "max_power": 1250,
        "cop": 4.0
      }
    ]
  },
```

[image]

corneel27, 2025-05-07:
Ik heb er nog even over nagedacht en vroeg mij nu weer af (wel te laat) of het wel klopt om dan over alle uren het verbruik op nul te zetten.
Zou dat niet alleen het eerste uur moeten zijn?

bvweerd, 2025-05-07:
Ik heb de "heat demand" switch geinterpreteerd als dat er niet mee gepland
hoeft te worden wanneer deze uit staat. Dus ook niet later in de planning.

Als je de switch aan zet, dan komt ie wel keurig in de planning, op de
gunstige tijdstippen.

corneel27, 2025-05-07:
Ik ga het toch aanpassen in de volgende push naar addon..
Bij de andere planningstrategieen van de heatpump wordt ook alleen het eerste uur geblokkeerd als heat_demand false is.
Dat wil ik graag uniform houden.
Dus het wordt:
```  
     model += c_hp[0] == 0
     model += hp_on[0] == 0
```
Akkoord?

bvweerd, 2025-05-08:
Prima. Is het een idee om de handleiding dan ook iets aan te passen van:

`entity hp heat demand : HA 'binary_sensor' entiteit die aangeeft of de thermostaat van de on/off warmtepomp is in- of uitgeschakeld. Als "on" zal de warmtepomp worden ingepland.`

naar:

`entity hp heat demand : HA 'binary_sensor' entiteit die aangeeft of de thermostaat van de on/off warmtepomp is in- of uitgeschakeld. Als "on" zal de warmtepomp direct worden ingepland, anders geoptimaliseerd.`

corneel27, 2025-05-08:
Goed idee om dat allemaal te verduidelijken, bijvoorbeeld:

**heater present**  als **true** zal de warmtepomp worden ingepland als **false** wordt er geen warmtepomp ingepland.

**entity hp enabled**:  een 'binary_sensor' of input_boolean die aangeeft of de aanwezige warmtepomp in bedrijf is (waarde is **on** en moet worden ingepland of dat hij uit staat (waarde is **off**) door bijvoorbeeld de zomer of een vakantie en niet mag worden ingepland.

**entity hp heat demand** : een 'binary_sensor' of input_boolean die aangeeft of er directe warmtevraag is in de woning (bijvoorbeeld via een thermostaat). aan de warmtepomp. Als "on" zal de inzet van de warmtepomp vanaf het eerste uur worden geoptimliseerd. Als "off" zal het eerste uur de warmtepomp altijd uit zijn en de inplanning worden berekend vanaf het volgende uur.

Zo akkoord?

bvweerd, 2025-05-08:
perfect!

PR #249 (2025-05-07): Fix heatpump calculation when heat demand is off
fixes #247
commit 6397cef5 (2025-05-07): Fix heatpump calculation when heat demand is off (#249)

fixes #247