corneel27/day-ahead #250: version in version.py

jsimonetti, 2025-05-08:
Hi,
thank you for the excellent program. I have one question regarding the `prog/version.py`

In the docker container on docker hub, the contents of the file is `__version__ = "addon"`, while the content of the file in git is `__version__ = "2025.2.0"`, even though version 2025.4.2 is the latest version.

How does this work, and does the version in the file really matter for anything? I am running the container without addon (next to home assistant core container), and (apart from an initial startup, trying to parse version 'addon' as an integer) everything seems to work fine.

Thank you for your time!

corneel27, 2025-05-08:
When all went right then during the building of the image the version-number in version.py is set right.
But I am testing with pre-builded images and that goes at the moment wrong.
So please don't use the pre builded images from docker hub of ghrc.io.

When you work with HA OS follow the instructions in DOCS.md
When you work with HA in a container follow the instructions in the TS of https://gathering.tweakers.net/forum/list_message/82368732.
Is that ok for you?

jsimonetti, 2025-05-08:
Thank you for the quick reply. I will use the instruction from GoT!

closed 2025-05-16 by jsimonetti