corneel27/day-ahead #280: Geen oplossing  voor: minimize cost, en postgres Current Overflow: -5

balk77, 2025-06-13:
Sinds 2025.6.1 (en wellicht ook .0) krijg ik de volgende foutmelding. In 2025.5.0 heb ik geen fout, met dezelfde config (behalve de breaking changes).
```
2025-06-13 15:49:17 debug: Starting new HTTP connection (1): 192.168.1.7:8123
2025-06-13 15:49:17 debug: http://192.168.1.7:8123 "GET /api/states/sensor.sessy_all_state_of_charge HTTP/1.1" 200 429
2025-06-13 15:49:17 info: Startwaarde SoC Sessys: 60.0%
2025-06-13 15:49:17 info: Boiler niet aanwezig of staat uit, boiler wordt niet ingepland
2025-06-13 15:49:17 info: Warmtepomp niet aanwezig of enabled - warmtepomp wordt niet ingepland
Welcome to the CBC MILP Solver 
Version: Trunk
Build Date: Oct 24 2021 

Starting solution of the Linear programming relaxation problem using Dual Simplex

Clp6003E Matrix has 6 large values, first at column 2147, row 1921 is nan
Clp3003W Analysis indicates model infeasible or unbounded
Clp6003E Matrix has 6 large values, first at column 2147, row 1921 is nan
Clp0004I Stopped due to errors - objective value 0
Clp0032I Errors objective 0 - 0 iterations time 0.002
2025-06-13 15:49:17 waarschuwing: Geen oplossing  voor: minimize cost
2025-06-13 15:49:17 debug: Connection status Pool size: 5  Connections in pool: 1 Current Overflow: -4 Current Checked out connections: 0 at line 541 in /root/dao/prog/da_base.py
2025-06-13 15:49:17 debug: Connection status Pool size: 5  Connections in pool: 1 Current Overflow: -4 Current Checked out connections: 0 at line 3425 in /root/dao/webserver/../prog/day_ahead.py
debug:root:Dialect: postgresql, database: day_ahead3, server: 192.168.1.111
debug:root:db_url: postgresql+psycopg2://day_ahead:G2DpVx66hfM2zK@192.168.1.111:5432/day_ahead3
debug:root:Dialect: postgresql, database: homeassistant3, server: 192.168.1.111
debug:root:db_url: postgresql+psycopg2://postgres:RMzK5tzHN2zebj@192.168.1.111:5432/homeassistant3
debug:root:Connection status Pool size: 5  Connections in pool: 0 Current Overflow: -5 Current Checked out connections: 0 at line 202 in /root/dao/prog/da_base.py
```
Hierin vallen 2 dingen op:
* waarschuwing: Geen oplossing  voor: minimize cost, en
* debug:root:Connection status Pool size: 5  Connections in pool: 0 Current Overflow: -5 Current Checked out connections: 0 at line 202 in /root/dao/prog/da_base.py
Ik heb mijn config door een JSON linter gehaald en die is correct, athans qua syntax. Zie ook hieronder.

<details>
<summary>Mijn config: (click to open)</summary>

```
{
    "homeassistant": {
        "protocol api": "http",
        "host": "192.168.1.7",
        "ip port": 8123,
        "token": "!secret ha_api_token"
    },
    "database ha": {
        "engine": "postgresql",
        "server": "192.168.1.111",
        "port": 5432,
        "database": "homeassistant3",
        "username": "postgres",
        "password": "!secret db_ha_password"
    },
    "database da": {
        "engine": "postgresql",
        "server": "192.168.1.111",
        "port": 5432,
        "database": "day_ahead3",
        "username": "day_ahead",
        "password": "!secret db_da_password"
    },
    "meteoserver-key": "!secret meteoserver-key",
    "prices": {
        "source day ahead": "nordpool",

        "energy taxes consumption": {
            "2022-01-01": 0.06729,
            "2023-01-01": 0.12599,
            "2024-01-01": 0.1088,
            "2025-01-01": 0.12286
        },
        "energy taxes production": {
[... 292 lines cut ...]
        "entities battery production": [
            "sensor.ess_grid_production"
        ],
        "entity co2-intensity": [
            "sensor.electricity_maps_co2_intensity"
        ]
    },
    "scheduler": {
        "1030": "get_meteo_data",
        "1255": "get_day_ahead_prices",
        "1355": "get_day_ahead_prices",
        "1455": "get_day_ahead_prices",
        "1554": "get_day_ahead_prices",
        "1630": "get_meteo_data",
        "1655": "get_day_ahead_prices",
        "2230": "get_meteo_data",
        "2359": "clean_data",
        "active": "True",
        "0430": "get_meteo_data",
        "xx00": "calc_optimum"
    }
}
```
</details>
Ik ga even graven

corneel27, 2025-06-13:
Ik kan op het eerste gezicht niks ontdekken waarom dao geen oplossing kan vinden.
Ik heb een deel van je instellingen (battery en solar) gekopieerd naar een test-instellingen set (verder alles uit) met dit als resultaat:

[image]

Dit is de logging:
```
2025-06-13 23:29:38 info: Day Ahead Optimalisering versie: 2025.6.0
2025-06-13 23:29:38 info: Day Ahead Optimalisering gestart op: 13-06-2025 23:29:38
2025-06-13 23:29:38 info: Day Ahead Optimalisatie gestart: 13-06-2025 23:29:38 taak: calc_optimum_met_debug
2025-06-13 23:29:38 info: Debug = True
2025-06-13 23:29:38 info: Baseload uit instellingen
2025-06-13 23:29:38 info: Start waarden: 
    uur                tijd       p_l       p_t  base     pv_ac  pv_dc
0    23 2025-06-13 23:00:00  0.261283  0.261283  0.16  0.000000      0
1     0 2025-06-14 00:00:00  0.263824  0.263824  0.14  0.000000      0
2     1 2025-06-14 01:00:00  0.242649  0.242649  0.38  0.000000      0
3     2 2025-06-14 02:00:00  0.257895  0.257895  0.26  0.000000      0
4     3 2025-06-14 03:00:00  0.248348  0.248348  0.42  0.000000      0
5     4 2025-06-14 04:00:00  0.263824  0.263824  0.15  0.000000      0
6     5 2025-06-14 05:00:00  0.265639  0.265639  0.12  0.015850      0
7     6 2025-06-14 06:00:00  0.249304  0.249304  0.13  0.131141      0
8     7 2025-06-14 07:00:00  0.251590  0.251590  0.15  0.283804      0
9     8 2025-06-14 08:00:00  0.176704  0.176704  0.23  0.506460      0
10    9 2025-06-14 09:00:00  0.147664  0.147664  0.26  0.722390      0
11   10 2025-06-14 10:00:00  0.146683  0.146683  0.31  0.916312      0
12   11 2025-06-14 11:00:00  0.141626  0.141626  0.32  1.023592      0
13   12 2025-06-14 12:00:00  0.130046  0.130046  0.31  1.024691      0
14   13 2025-06-14 13:00:00  0.124069  0.124069  0.23  0.913810      0
15   14 2025-06-14 14:00:00  0.118176  0.118176  0.26  0.345967      0
16   15 2025-06-14 15:00:00  0.139557  0.139557  0.21  0.175081      0
17   16 2025-06-14 16:00:00  0.147216  0.147216  0.21  0.475964      0
18   17 2025-06-14 17:00:00  0.158384  0.158384  0.54  1.001626      0
19   18 2025-06-14 18:00:00  0.235255  0.235255  0.26  0.374336      0
20   19 2025-06-14 19:00:00  0.284176  0.284176  0.26  0.391506      0
21   20 2025-06-14 20:00:00  0.295005  0.295005  0.22  0.174692      0
22   21 2025-06-14 21:00:00  0.301092  0.301092  0.19  0.061264      0
23   22 2025-06-14 22:00:00  0.292985  0.292985  0.18  0.000000      0
24   23 2025-06-14 23:00:00  0.282228  0.282228  0.16  0.000000      0
2025-06-13 23:29:40 info: Reduced hours applied for Sessys
2025-06-13 23:29:40 info: Startwaarde SoC Sessys: 39.0%
2025-06-13 23:29:40 info: Boiler niet aanwezig of staat uit, boiler wordt niet ingepland
2025-06-13 23:29:40 info: Warmtepomp niet aanwezig of enabled - warmtepomp wordt niet ingepland
2025-06-13 23:29:40 info: Strategie: minimale kosten
2025-06-13 23:29:40 info: Het programma heeft een optimale oplossing gevonden.
2025-06-13 23:29:40 info: Niet geoptimaliseerd, kosten met day ahead tarieven: -0.14 
2025-06-13 23:29:40 info: Geoptimaliseerd, kosten met day ahead tarieven: -0.48 
2025-06-13 23:29:40 info: Levering: 6.60   (kWh)
2025-06-13 23:29:40 info: In- en uitgaande energie per uur batterij Sessys
   uur   ac->    eff   ->dc pv->dc   dc->    eff  ->bat  o_eff    SoC
          kWh      %    kWh    kWh    kWh      %    kWh      %      %
    23  -0.51  94.30  -0.54   0.00  -0.54 100.00  -0.54  94.30  33.63
     0  -1.00  94.30  -1.06   0.00  -1.06 100.00  -1.06  94.30  23.03
     1   0.00     --   0.00   0.00  -0.00     --  -0.00     --  23.03
     2   0.00     --   0.00   0.00  -0.00     --  -0.00     --  23.03
     3   0.00     --   0.00   0.00  -0.00     --  -0.00     --  23.03
     4  -1.00  94.30  -1.06   0.00  -1.06 100.00  -1.06  94.30  12.42
     5  -1.00  94.30  -1.06   0.00  -1.06 100.00  -1.06  94.30   1.82
     6   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
     7   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
     8   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
     9   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
    10   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
    11   0.00     --   0.00   0.00  -0.00     --  -0.00     --   1.82
    12   2.33  94.47   2.20   0.00   2.20 100.00   2.20  94.47  23.82
    13   2.33  94.47   2.20   0.00   2.20 100.00   2.20  94.47  45.82
    14   2.33  94.47   2.20   0.00   2.20 100.00   2.20  94.47  67.82
    15   0.00     --   0.00   0.00  -0.00     --  -0.00     --  67.82
    16   0.00     --   0.00   0.00  -0.00     --  -0.00     --  67.82
    17   0.00     --   0.00   0.00  -0.00     --  -0.00     --  67.82
[... 24 lines cut ...]
 13.00    2.33     0.00   1.65   0.00   0.23   0.00   0.00   0.00   0.91   0.20   -0.00  20.00
 14.00    2.33     0.00   2.24   0.00   0.26   0.00   0.00   0.00   0.35   0.27   -0.00  20.00
 15.00    0.00     0.00   0.03   0.00   0.21   0.00   0.00   0.00   0.18   0.00   -0.00  20.00
 16.00    0.00     0.00   0.00   0.27   0.21   0.00   0.00   0.00   0.48   0.00   -0.04  20.00
 17.00    0.00     0.00   0.00   0.46   0.54   0.00   0.00   0.00   1.00   0.00   -0.07  20.00
 18.00    0.00     0.00   0.00   0.11   0.26   0.00   0.00   0.00   0.37   0.00   -0.03  20.00
 19.00    0.00     1.63   0.00   1.76   0.26   0.00   0.00   0.00   0.39   0.00   -0.50  20.00
 20.00    0.00     1.63   0.00   1.58   0.22   0.00   0.00   0.00   0.17   0.00   -0.47  20.00
[... cut ...]
2025-06-13 23:29:40 info: Winst: € 0.34
2025-06-13 23:29:40 info: Onderstaande settings worden NIET doorgezet naar HA
2025-06-13 23:29:40 info: Grid set point: -840.0 W
2025-06-13 23:29:40 info: Cycle cost Sessys: 0.84 euro
2025-06-13 23:29:40 info: Netto vermogen naar(+)/uit(-) batterij Sessys zou zijn: -1000 W
2025-06-13 23:29:40 info: Balanceren zou zijn: False
```
Ik zie dus niks geks.
Jij wel?

Verdere vragen:
- heb jij dezelfde prijzen, pv-opwek en meteo als ik?
- en de hamvraag: hoe ziet jou baseload eruit?

balk77, 2025-06-18:
Het deel `debug:root:Connection status Pool size: 5 Connections in pool: 0 Current Overflow: -5 Current Checked out connections: 0 at line 202 in /root/dao/prog/da_base.py` kan geen kwaad. 

Maar, het wordt wel gek. Ik heb DAO als addon draaien. Daarin gaat het fout. En ik heb een virtual machine met een venv waarop ik DAO draai (zonder docker dus). Met dezelfde settings, zelfde database gaat het bij de addon versie fout maar bij de native versie goed. Ik snap er geen sikkepit van. Ik ga nog wat testen, wellicht gooi ik mijn addon weg en start ik opnieuw.

balk77, 2025-06-19:
Ik gooi hem toch weer open. Ik had vannacht weer een `geen oplossing`. Ik vermoed dat het komt door een negatieve baseload. Ik ga eens graven en gooi dit issue toch nog even open.
```
025-06-17 23:00:27 info: baseload voor weekdag 0 :
2025-06-17 23:00:27 info: -0.83 0.003 0.008 0.007 0.009 0.012 0.1 -0.487 0.387 0.616 1.042 1.9 2.158 3.536 2.293 1.509 0.917 0.697 0.529 -1.174 -0.854 -0.936 -0.997 -0.4 
2025-06-17 23:00:27 info: baseload voor weekdag 1 :
2025-06-17 23:00:27 info: 0.008 0.006 0.007 0.01 0.025 0.127 0.248 0.432 0.49 0.602 0.899 1.273 2.282 1.216 2.042 1.782 0.865 0.735 0.626 -0.373 -0.473 -0.464 -0.331 -0.354 
2025-06-17 23:00:27 info: baseload voor weekdag 2 :
2025-06-17 23:00:27 info: -0.306 -0.302 -0.306 -0.032 -0.022 -0.088 -0.187 -0.766 -0.515 0.443 1.284 1.84 2.902 3.241 2.38 2.309 1.215 0.378 0.146 -0.389 -0.928 -0.792 -0.841 -0.419 
2025-06-17 23:00:27 info: baseload voor weekdag 3 :
2025-06-17 23:00:27 info: 0.074 0.065 0.073 0.066 0.071 0.078 -0.372 -0.221 0.627 0.766 0.942 1.302 0.948 1.519 2.324 1.448 0.466 0.708 0.38 -1.204 -1.075 -0.51 -0.662 0.064 
2025-06-17 23:00:27 info: baseload voor weekdag 4 :
2025-06-17 23:00:27 info: 0.077 0.074 0.126 0.124 0.128 0.126 -0.286 -0.471 0.188 0.756 1.316 1.698 1.993 2.858 2.149 1.916 0.899 0.556 0.354 0.052 -0.025 -0.35 -0.77 -0.394 
2025-06-17 23:00:27 info: baseload voor weekdag 5 :
2025-06-17 23:00:27 info: -0.4 0.085 -0.378 0.093 0.071 0.094 0.138 -0.744 0.389 0.931 0.939 1.158 1.632 2.164 1.487 1.322 0.94 2.164 0.669 -0.735 -1.07 -0.405 -0.291 -0.313 
2025-06-17 23:00:27 info: baseload voor weekdag 6 :
2025-06-17 23:00:27 info: 0.012 0.167 0.202 0.282 0.17 0.144 0.168 0.741 0.673 0.828 1.388 1.346 1.434 2.671 2.326 2.15 1.709 0.6 0.464 0.214 -0.484 -0.316 -0.734 -0.388 
```

corneel27, 2025-06-19:
De oplossing die jij hierboven geeft is DE oplossing:
geef beide sensoren op het is niet voor niks een "list".
De "opgetelde" sensor in HA gaat daar niet goed mee om.
Dus als je die voor DAO hebt gemaakt kun je hem weggooien.

Ik sluit issue weer, OK?

balk77, 2025-06-24:
Weer een poging, dit is de foutmelding die ik krijg. Maar als ik dezelfde config in een andere instance van DAO plak dan convergeert ie wel.
```

2025-06-24 21:57:37 debug: cycle cost: 0.05 eur/kWh
2025-06-24 21:57:37 debug: Starting new HTTP connection (1): 192.168.1.7:8123
2025-06-24 21:57:37 debug: http://192.168.1.7:8123 "GET /api/states/sensor.sessy_all_state_of_charge HTTP/1.1" 200 429
2025-06-24 21:57:37 info: Startwaarde SoC Sessys: 74.0%
2025-06-24 21:57:37 info: Boiler niet aanwezig of staat uit, boiler wordt niet ingepland
2025-06-24 21:57:37 info: Warmtepomp niet aanwezig of enabled - warmtepomp wordt niet ingepland
Welcome to the CBC MILP Solver 
Version: Trunk
Build Date: Oct 24 2021 

Starting solution of the Linear programming relaxation problem using Dual Simplex

Clp6003E Matrix has 6 large values, first at column 1757, row 1573 is nan
Clp3003W Analysis indicates model infeasible or unbounded
Clp6003E Matrix has 6 large values, first at column 1757, row 1573 is nan
Clp0004I Stopped due to errors - objective value 0
Clp0032I Errors objective 0 - 0 iterations time 0.002
2025-06-24 21:57:37 waarschuwing: Geen oplossing  voor: minimize cost
2025-06-24 21:57:37 debug: Connection status Pool size: 5  Connections in pool: 1 Current Overflow: -4 Current Checked out connections: 0 at line 541 in /root/dao/prog/da_base.py
2025-06-24 21:57:37 debug: Connection status Pool size: 5  Connections in pool: 1 Current Overflow: -4 Current Checked out connections: 0 at line 3425 in /root/dao/webserver/../prog/day_ahead.py
debug:root:Dialect: postgresql, database: day_ahead3, server: 192.168.1.111
debug:root:db_url: postgresql+psycopg2://day_ahead:abc@192.168.1.111:5432/day_ahead3
debug:root:Dialect: postgresql, database: homeassistant3, server: 192.168.1.111
debug:root:db_url: postgresql+psycopg2://postgres:abc@192.168.1.111:5432/homeassistant3
debug:root:Connection status Pool size: 5  Connections in pool: 0 Current Overflow: -5 Current Checked out connections: 0 at line 202 in /root/dao/prog/da_base.py

```

balk77, 2025-06-26:
Ik heb een manier gevonden om het model te exporteren `model.write("model2025.5.0.lp")`. Bestand 2025.5.0 geeft wel een oplossing, het andere niet.
[model2025.5.0.lp.txt](https://github.com/user-attachments/files/20932192/model2025.5.0.lp.txt)
[model2026.6.2.lp.txt](https://github.com/user-attachments/files/20932193/model2026.6.2.lp.txt)

balk77, 2025-06-27:
Bedankt voor je hulp! Ik ben intussen over naar Sqlite voor de DAO database, Home Assistant is wel postgresql. Ik heb nu de TESTING addon draaien met postgreql en de normale, als productie, met sqlite. Kan je iets met deze logs?
[runlog_debug_2025.6.2.rc5_postgresql.txt](https://github.com/user-attachments/files/20945137/runlog_debug_2025.6.2.rc5_postgresql.txt)
[runlog_debug_2025.6.2_sqlite.txt](https://github.com/user-attachments/files/20945136/runlog_debug_2025.6.2_sqlite.txt)

balk77, 2025-06-27:
Goed dat je het gevonden hebt. Ik gebruik dummy entities en een dummy database. Laat maar weten wanneer ik kan testen

corneel27, 2025-06-27:
Ik ben er diep ingedoken.
Het probleem zit niet in de programmacode, maar in een setting van je database.
Daar staat namelijk de timezone op "UTC".
En deze moet worden omgezet naar  'Europe/Amsterdam'.
Je moet dit omzetten met het commando:
`alter database day_ahead set timezone to 'Europe/Amsterdam';`
Daarna zal het wel werken.
Kun je uitvoeren en me bevestigen dat dit de oplossing was?

balk77, 2025-06-28:
ja! Dat is de oplossing! Testje gedaan:  `Optimaliseringsberekening met debug` voordat ik `alter database day_ahead set timezone to 'Europe/Amsterdam';` deed --> foutmelding, daarna geen foutmelding.

Is deze setting op een manier te automatiseren bij het opzetten van de database?

corneel27, 2025-07-06:
Met versie 2025.7.0 is dit issue opgelost.

closed 2025-07-06 by corneel27